borglab/gtsam #728: Python Wrapper of custom factor definition

carlin314, 2021-03-30:
## Feature

Python Wrapper of custom factor definition

## Motivation

When I do problem, I need to custom define factor(the whole project is based on python), I find gtsam python install is convenient. 

But I don't find python factor custom sample, if there is one, please tell me.

## Pitch

Python Wrapper of custom factor definition

## Alternatives

**repo**: [minisam](https://github.com/dongjing3309/minisam)

**paper**: [miniSAM: A Flexible Factor Graph Non-linear Least Squares Optimization Framework](https://arxiv.org/pdf/1909.00903.pdf)

ProfFan, 2021-03-30:
Custom factors can be implemented with gtsam-python-project. There isn't support for custom factor fully in Python yet. Maybe this can be implement as an optional feature by using https://pybind11.readthedocs.io/en/stable/advanced/pycpp/object.html#calling-python-methods ?

dellaert, 2021-03-30:
@ProfFan you give contradictory info :-) AFAIK we *cannot* make custom factors in python. But @mattking-smith was really interested in this as well. Matt?

closed 2021-06-07 by ProfFan