borglab/gtsam #737: OrientedPlane3Factor  'gtsam::IndeterminantLinearSystemException'

hunkyu, 2021-04-08:
## Description
terminate called after throwing an instance of'gtsam::IndeterminantLinearSystemException'
  what():
Indeterminant linear system detected while working near variable
8070450532247928988 (Symbol: p156).

It is relatively random when it runs for a period of time.

## Steps to reproduce

1. add(PriorFactor<OrientedPlane3>(gtsam::symbol('p',0), lidar_plane, plane_noise));
this->initial_estimate_->insert(gtsam::symbol('p',0),lidar_plane);
2. gtsam::OrientedPlane3 plane_from = cache_plane[0].coffes;
            auto plane_measured_from_pose = plane_from.transform(pose_to);
                    const auto x_p_noise = noiseModel::Isotropic::Sigma(3, 0.001);
            gtsam::OrientedPlane3Factor plane_factor(plane_measured_from_pose.planeCoefficients(),x_p_noise,gtsam::symbol('x',size),gtsam::symbol('p',size));
            this->initial_estimate_->insert(gtsam::symbol('p',size),cache_plane[0].coffes);
            this->gtSAMgraph_->add(plane_factor);

## Environment
ubuntu 18.04
gtsam 4.0.2

varunagrawal, 2022-04-02:
This usually means your system is under-constrained (aka rank-deficient), so it would be good to check your factor graph and make sure each variable is properly constrained.

varunagrawal, 2022-07-07:
@hunkyu any updates? Please reopen if this is still an issue.