borglab/gtsam #746: SO3 matrix log when theta is close to zero

yiping, 2021-04-19:
https://github.com/borglab/gtsam/blob/43e8f1e5aeaf11890262722c1e5e04a11dbf9d75/gtsam/geometry/SO3.cpp#L279-L281

Should line 281 be

`magnitude = 0.5 - tr_3 / 12.0;`

instead?

tmcg0, 2021-05-28:
This is probably worth someone from the GTSAM team getting their eyes on. In practice this wouldn't affect the computation much because abs(tr_3) is always <1e-7. But there does seem to be an inconsistency between the comment at line 280 and the code at 281. I'll try to track down what it should be and post here.

tmcg0, 2021-05-28:
original function for magnitude is:
`magnitude = f(theta) = theta/(2.0*sin(theta))` # See [Blanco](https://jinyongjeong.github.io/Download/SE3/jlblanco2010geometry3d_techrep.pdf).

taylor series expansion of f(theta) at theta=0 should be:
1/2 + theta^2/12 + 7*theta^4/720 + H.O.T.

I'm a little bit rusty and unsure what's happening in the code. Is it that for small theta we're assuming that tr(R) ~ theta? And that's why we do this variable swap between `tr_3` and `theta`?

dellaert, 2021-05-28:
OK, so, the angle `theta` is a function of the trace *t*, which ranges from -1 to 3. Here is [Wolfram alpha's plot](https://www.wolframalpha.com/input/?i=acos%28%28t+-+1.0%29+%2F+2.0%29+):

[image]

We are interested in getting the Taylor series of `theta/(2.0*sin(theta))` as you said, but in *t* (or *t-3*). For the life of me I can't remember how I got the Taylor of magnitude expansion in t. Maybe that's a fruitful direction to go? Both comment and code might be wrong.

ProfFan, 2021-05-28:
I remember doing this math when dealing with Cayley and stuff long long time ago. What I remember is that it's because the original form cannot be evaluated at tr_3 = 0, so one needs to not use the formula in first branch, but the original form to devise this taylor expansion.

tmcg0, 2021-05-28:
Glad I'm not the only rusty one here :)

So here's a snippet from Blanco:
[image]

We're calling magnitude the `theta/(2 sin(theta))` term. Let's call `t=tr(R)`
And `theta=acos((t-1)/2)`
So in full messy form, `magnitude = f(t) = acos((t-1)/2)/(2 sin(acos((t-1)/2)))` which I think is ugly but Wolfram Alpha doesn't:

[image]
[image]

So, to first order....
`magnitude = 2*pi/(3*sqrt(3)) + (-1/3-2*pi/(9*sqrt(3)))*t`

Can someone tell me what I'm doing wrong here? That constant term out front evaluates to ~1.2, very different from the 0.5 in the code currently.

ProfFan, 2021-05-29:
@tmcg0 
> What I remember is that it's because the original form cannot be evaluated at tr_3 = 0, so one needs to not use the formula in first branch, but the original form to devise this taylor expansion.

Wolfram Alpha may suffer from the same problem :) This limit is not easy to calculate, especially for CAS systems, if you don't give hints to them.

Could you put up a simple PR with more through documentation? Many thanks!

dellaert, 2021-05-29:
Awesome. Tim, if you can do a small PR with code correction that would be fine, any additional docs would be nice to have, but you could also just refer to this issue.

tmcg0, 2021-05-29:
@ProfFan @dellaert will do, probably early next week. I'll probably add an analytical derivation to this issue and then reference the issue in a comment in the code in case anyone wants to see it.

ProfFan, 2021-05-30:
Actually I found out that Maxima solved the Taylor series automagically...

[image]

Sympy and Wolfram Alpha both failed.

closed 2021-06-03 by ProfFan