borglab/gtsam #756: testShonanAveraging: Failure during make check

quantumhub, 2021-04-24:
## Description
A failure in automatic test during make check. Is it serious?

108/242 Test #108: testShonanAveraging ....................***Failed    0.02 sec
/home/kepler/Documents/GTSLAM/gtsam-release-4.1.0/gtsam/sfm/tests/testShonanAveraging.cpp:171: Failure: "expected -4.1072555088854512e-14 but was: -2.0463630789890885e-12" 
There were 1 failures
...
The following tests FAILED:
	108 - testShonanAveraging (Failed)
Errors while running CTest
CMakeFiles/check.dir/build.make:57: recipe for target 'CMakeFiles/check' failed
make[3]: *** [CMakeFiles/check] Error 8
CMakeFiles/Makefile2:978: recipe for target 'CMakeFiles/check.dir/all' failed
make[2]: *** [CMakeFiles/check.dir/all] Error 2
CMakeFiles/Makefile2:985: recipe for target 'CMakeFiles/check.dir/rule' failed
make[1]: *** [CMakeFiles/check.dir/rule] Error 2
Makefile:249: recipe for target 'check' failed
make: *** [check] Error 2

## Steps to reproduce
It appears during installation. I first had gtsam 4.0 installed. And xubuntu installed all dependencies automatically. Then, I try to compile gtsam 4.1:

sudo add-apt-repository ppa:borglab/gtsam-release-4.0
sudo apt update
sudo apt install libgtsam-dev libgtsam-unstable-dev
cd GTSLAM/
cd gtsam-release-4.1.0/
cd build/
cmake ..
make check

It is original code: gtsam-release-4.1.0.zip from github. No change.

## Expected behavior

Expecting 100% tests pass rather than: 
99% tests passed, 1 tests failed out of 242

## Environment

----OS info:
cat /etc/os-release
NAME="Ubuntu"
VERSION="18.04.5 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.5 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

----Python:
Python 2.7.17 (default, Feb 27 2021, 15:10:58) 

----No matlab

varunagrawal, 2021-06-10:
This is not really a bug since both expected and actual values are pretty close to zero, `1e-12` vs `1e-14`. This could be because you installed the GTSAM package first, it is using an older version of some dependency and so there is some numerical stability stuff going on in the back.

@quantumhub you can just change the tolerance value of the test to get it to pass. Anything less that `1e-10` is pretty safe IMO.

varunagrawal, 2021-06-10:
Closing since this is not really a bug in the code and may have to do with user's system set up.

closed 2021-06-10 by varunagrawal