borglab/gtsam #757: Manually updating the optimization variables inside ISAM2, is it possible?

lppllppl920, 2021-04-25:
Hi,

First of all, I want to thank the authors for creating such a great and easy-to-use optimization framework and I have enjoyed it a lot!

I am trying to develop a 3D dense SLAM with this framework and found that sometimes the optimization speed on closing a large-scale detected global loop is slow. There is a certain closed-form global loop closing algorithm that I want to use to manually modify all camera poses outside of the ISAM2 and then reinitialize the relevant variables inside ISAM2. Is it something that is possible under the current framework? I was thinking to add some functionality similar to ISAM2::addVariables but instead, update the existing items inside those std::map variables. Is there something that I need to pay attention to if this is actually something feasible to do?

Thank you so much!

pshea76, 2021-05-10:
Hey, 

I am by no means an authority on this page, but I was able to implement this by creating my own solver that inherits from ISAM2.  It was pretty complicated to figure out, but heres how I did it: 
```
// Reset the value of some subset of variables in ISAM2
void Solver::reInitialize(const gtsam::Values& new_variable_values){
    if( !new_variable_values.empty()){

        reinitialized_ = true; // Flag I used to keep track of when I had reinitialized something

        for(const gtsam::Values::ConstKeyValuePair& key_val : new_variable_values){

            // Mark this value to be re-eliminated at the next update
            extra_elimination_keys_.insert(key_val.key);    

            // Clear the cached deltas wrt the old linearization point
            deltaNewton_.at(key_val.key) = gtsam::Vector::Zero(key_val.value.dim());
            RgProd_.at(key_val.key) = gtsam::Vector::Zero(key_val.value.dim());
            delta_.at(key_val.key) = gtsam::Vector::Zero(key_val.value.dim());
            // Do not add this key to deltaReplacedMask_ or the above will be overwritten

            // Update the linearization point with our new estimate
            theta_.update(key_val.key, key_val.value);
        }
    }
}
```

Basically how it works is we update isam2's internal linearization point, set the delta w.r.t. the linearization point to zero and mark that variable for re-elimination.  When we modify the linearization point in isam, we need to be careful because the bayes tree is defined wrt the linearization point.  To avoid issues related to this, we need to mark the variable for elimination, and pass the "force_relinearize" flag to ISAM.  Also we should perform reinitializations right before starting the next solver iteration (i.e. don't go computing marginals with a messed up bayes tree).  If there is a better way to do this lmk, but this appears to do everything I expect so far.  

Best.
Pat

lppllppl920, 2021-05-16:
Thank you so much, Pat! This is very helpful!

closed 2021-05-16 by lppllppl920