borglab/gtsam #764: Custom factors & Python packaging

Skydes, 2021-05-07:
Hi there,

I have been playing with the Python API recently and I am really in awe of how neat it is! Coming from Ceres, everything seems so easy with GTSAM. Huge thanks to the team for pulling this :)

I want to optimize camera poses/calibration via reprojections while keeping the 3D points fixed (similar to https://github.com/borglab/gtsam/issues/697). There is no factor for this and I want to avoid writing a custom C++ factor, mostly because dependencies could then not anymore be handled by Python.

1. I hacked around using a prior on the points with very low`1e-8` noise, but this is hacky and inefficient, so I am looking for better :) Is there a way to freeze some optimization variables? Ceres has `SetParameterBlockConstant` but I couldn't find anything similar in GTSAM.
2. If not, any ETA on supporting custom factors in Python? This was mentioned in https://github.com/borglab/gtsam/issues/728.
3. We also use conda to handle dependencies, so I would be ok with a C++ factor if there was a GTSAM conda package.

ProfFan, 2021-05-07:
On (1), I think you can use zero covariance which will make the constraint an equality constraint :)
On (2), #748 is on my table now, I will start implementing it maybe tomorrow.
On (3), we currently do not use conda for distribution, but it should be fairly straightforward (contributions welcomed!)

Skydes, 2021-05-12:
Thanks for the quick reply.
1. For large graphs I observe that a zero covariance gives `inf` cost, but a `1e-8` sigma works fine.
2. That would be awesome! On our side I can think of many cases for which this would be useful.
3. I might look into it then.

ProfFan, 2021-06-07:
Python factor is now in develop, please try and test it hard so we can know what to change :)

closed 2021-06-07 by ProfFan

halajun, 2022-04-19:
> 

Hi @ProfFan. I saw an example file called CustomFactorExample.py, and from here seems python custom factor has been supported? If so, how can I get to know about the details of defining custom factor, except this example? Thanks!

halajun, 2022-04-19:
> > 
> 
> Hi @ProfFan. I saw an example file called CustomFactorExample.py, and from here seems python custom factor has been supported? If so, how can I get to know about the details of defining custom factor, except this example? Thanks!

No worries. I searched the library globally and found the source files (CustomFactor.h, CustomFactor.cpp) in "nonlinear" folder. Cheers!

ProfFan, 2022-04-19:
Hi @halajun , great to hear that you found the information you need! Please note though the current Python factor implementation will incur a performance cost, but is very versatile so you can use any Python package in the callback function. We have plans for a JIT-enabled fast Python CustomFactor (#875), so if you are interested, we can look into that together!

halajun, 2022-04-20:
> Hi @halajun , great to hear that you found the information you need! Please note though the current Python factor implementation will incur a performance cost, but is very versatile so you can use [...]

Hi @ProfFan. Thanks for your warm invitating, but I just need it to test my dummy method, not requiring for high performance at the moment. Anyway, Wish you good luck with the progress! 

Oh, by the way, would it easier to write custom factor in C++ library, and generate the python bindings without degrading performance?