borglab/gtsam #768: integrateMeasurement() data frame unclear

HMellor, 2021-05-25:
## Description

In [CombinedImuFactorsExample.cpp](https://github.com/borglab/gtsam/blob/ab92779b25b04b376fbbd1846bbbd21904c50e7a/examples/CombinedImuFactorsExample.cpp#L26) it is stated that the IMU data is provided aligned to NED (the navigation frame).

However, the documentation for [gtsam::PreintegratedCombinedMeasurements::integrateMeasurement()](https://gtsam.org/doxygen/4.0.0/a03435.html#a9c9d6f736444eab5d49fa9a9b305c738) states that the IMU data should be provided in body frame.

Only one of these can be correct and I am uncertain which is.

tmcg0, 2021-05-27:
Measurements should be provided in the body frame.  I think the example file comment is just misleading---perhaps the referenced comment line should end names in X, Y, Z instead of N, E, D.

mikesheffler, 2021-05-27:
Agreed. Saying 'N, E, D', may have been the author's way of describing a coordinate system that could perhaps be more accurately described as 'forward, right, down'. At least, I know I've made that specific whoopsie :)

tmcg0, 2021-05-28:
@HMellor you could always make a PR that cleans up that comment. :) clarity is always a plus.

PR #771 (2021-06-10): Correct ImuFactorExamples frame description
Fixes #768
closed 2021-06-10 by varunagrawal