borglab/gtsam #769: Compilation Failed on Mac OS X Big Sur with TBB 2021.2.0

zarathustr, 2021-05-25:
The current default TBB version installed using Brew on Mac OS X Big Sur is 2021.2.0. However, when using this library, the cmake system of GTSAM will report missing file of /usr/local/include/tbb/tbb_stddef.h. This file has been deprecated in TBB 2021. The correct configuration should be done by install TBB 2020 using:
`` brew install tbb@2020``

After that, in cmake/HandleTBB.cmake, we may need to add the specific path of TBB 2020 for recognition (Line 3):
``find_package(TBB 4.4 COMPONENTS tbb tbbmalloc PATHS /usr/local/opt/tbb@2020/lib)``
which correctly fixes the problem.

ProfFan, 2021-06-03:
Note you can use enviroment variables to do this, just add `TBB_PATH` or similar to your cmake command line.

closed 2021-06-03 by ProfFan

Gatsby23, 2022-11-05:
> TBB_PATH

Sorry for troubling you, but what version of gtsam have you used? I have met the same problem as you and change the CMakeLists.txt too, but it still can't find the tbb