borglab/gtsam #770: Cal3Bundler calibrate() jacobian unit test failure

akshay-krishnan, 2021-05-27:
## Description

The Jacobians of the calibrate() method of Cal3Bundler fail the numerical derivative unit test for certain calibration and input values. Specifically, they fail for the default value of Cal3Bundler i.e f = 1 with 0 distortion. 

## Steps to reproduce

I've added a couple of unit tests in a branch [here](https://github.com/borglab/gtsam/blob/f64ef4ed4239371ad7e6fea008a10e95587c8584/gtsam/geometry/tests/testCal3Bundler.cpp#L104). The unit test called DcalibrateDefault is the one that fails.  

TEST(Cal3Bundler, DcalibrateDefault) {
  Cal3Bundler trueK;
  Matrix Dcal, Dp;
  Point2 pn(0.5, 0.5);
  Point2 pi = trueK.uncalibrate(pn);
  Point2 actual = trueK.calibrate(pi, Dcal, Dp);
  CHECK(assert_equal(pn, actual, 1e-7));
  Matrix numerical1 = numericalDerivative21(calibrate_, trueK, pi);
  Matrix numerical2 = numericalDerivative22(calibrate_, trueK, pi);
  CHECK(assert_equal(numerical1, Dcal, 1e-5));
  CHECK(assert_equal(numerical2, Dp, 1e-5));
}

Failure: 

not equal:
expected = [
        -0.5, 0, 0;
        -0.5, 0, 0
]
actual = [
        -0.5, -0.25, -0.125;
        -0.5, -0.25, -0.125
]
actual - expected = [
        5.046e-11, -0.25, -0.125;
        5.046e-11, -0.25, -0.125
]

## Environment

Linux

## Additional information

The calibrate method uses the implicit function theorem to compute the derivatives, which internally uses the derivative of uncalibrate(). Since the unit test for uncalibrate() passes, I think it should be the implementation of the implicit function theorem that is causing the issue. 

The calibrate() method for Cal3Bundler uses an optimization to compute the point in calibrated coordinates. The distortion for this case is zero, the optimization converges in the first iteration itself, as expected.

dellaert, 2021-05-27:
@varunagrawal take a look if you have time?

varunagrawal, 2021-05-30:
From my understanding, this is because we ignore radial distortions in the initial value of `pn` in the `calibrate` method.
https://github.com/borglab/gtsam/blob/f64ef4ed4239371ad7e6fea008a10e95587c8584/gtsam/geometry/Cal3Bundler.cpp#L102
This causes the `pi` (the image coordinates) and `pn` (the intrinsic coordinates) to be identical leading to this issue.

When performing finite differences via the `traits<Point2>::Local`, we pass in the `.calibrate(pi)` result of `trueK` and the slightly modified `trueK`. However, since `.calibrate` for `Cal3Bundler` finds the value of `pn` on the first go (due to `(pn = inv(K)*pi) == pi`), it never gets a chance to add in the effects of the distortion. This leads to the finite difference being 0 and we see the issue.

I've opened PR #775 which fixes this issue. :slightly_smiling_face:

varunagrawal, 2021-05-30:
Fun fact: I am currently reading a lot of papers about camera calibration models for Skydio. This was a good issue for me to solve. :smile:

akshay-krishnan, 2021-06-01:
Nice catch Varun! So the "expected" Jacobian computed numerically is the one that is incorrect?

varunagrawal, 2021-06-01:
Yes

PR #775 (2021-06-04): Fix initial guess for Cal3Bundler `calibrate`
This PR fixes #770.

I simply incorporated the radial distortion from the iteration loop to the initial guess and the test passed. This makes sense since when we tweak the radial distortion parameters a bit for finite difference, we want to be able to see that effect in the output.
closed 2021-06-04 by varunagrawal