borglab/gtsam #772: Failing examples on 20.04

bryanhpchiang, 2021-05-29:
Ubuntu 20.04 - built gtsam from the develop branch earlier today. 

When running the examples (specifically the StereoVO one [here](https://github.com/borglab/gtsam/blob/develop/examples/StereoVOExample.cpp)), I'm getting an undefined symbol error:

```
./main: symbol lookup error: ./main: undefined symbol: _ZNK5gtsam6Factor9printKeysERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKSt8functionIFS6_mEE
```

I found out what the symbol represents, but not sure how to fix it:

```
(base) bryan@ito:~/work/play-gtsam$ c++filt _ZNK5gtsam6Factor9printKeysERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKSt8functionIFS6_mEE
gtsam::Factor::printKeys(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (unsigned long)> const&) const
```

Any suggestions would be appreciated!

Here's a minimal setup to reproduce (it builds but doesn't run):

Makefile:
```
all: main.cpp
	g++ main.cpp \
	-std=c++11 \
	-I /usr/include/eigen3 \
	-lboost_system -lboost_filesystem \
	-lgtsam \
	-ltbb \
	-o main

clear:
	rm -rf main
```

main.cpp: 
```
/* ----------------------------------------------------------------------------

 * GTSAM Copyright 2010, Georgia Tech Research Corporation,
 * Atlanta, Georgia 30332-0415
 * All Rights Reserved
 * Authors: Frank Dellaert, et al. (see THANKS for the full author list)

 * See LICENSE for the license information

 * -------------------------------------------------------------------------- */

/**
 * @file StereoVOExample.cpp
 * @brief A stereo visual odometry example
 * @date May 25, 2014
 * @author Stephen Camp
 */

/**
 * A 3D stereo visual odometry example
 *  - robot starts at origin
 *  -moves forward 1 meter
 *  -takes stereo readings on three landmarks
 */

#include <gtsam/geometry/Pose3.h>
#include <gtsam/geometry/Cal3_S2Stereo.h>
#include <gtsam/nonlinear/Values.h>
#include <gtsam/nonlinear/NonlinearEquality.h>
#include <gtsam/nonlinear/NonlinearFactorGraph.h>
#include <gtsam/nonlinear/LevenbergMarquardtOptimizer.h>
#include <gtsam/slam/StereoFactor.h>

using namespace std;
using namespace gtsam;

int main(int argc, char **argv)
{
    // create graph object, add first pose at origin with key '1'
    NonlinearFactorGraph graph;
    Pose3 first_pose;
    graph.emplace_shared<NonlinearEquality<Pose3>>(1, Pose3());

    // create factor noise model with 3 sigmas of value 1
    const auto model = noiseModel::Isotropic::Sigma(3, 1);
    // create stereo camera calibration object with .2m between cameras
    const Cal3_S2Stereo::shared_ptr K(
        new Cal3_S2Stereo(1000, 1000, 0, 320, 240, 0.2));

    //create and add stereo factors between first pose (key value 1) and the three landmarks
    graph.emplace_shared<GenericStereoFactor<Pose3, Point3>>(StereoPoint2(520, 480, 440), model, 1, 3, K);
    graph.emplace_shared<GenericStereoFactor<Pose3, Point3>>(StereoPoint2(120, 80, 440), model, 1, 4, K);
    graph.emplace_shared<GenericStereoFactor<Pose3, Point3>>(StereoPoint2(320, 280, 140), model, 1, 5, K);

    //create and add stereo factors between second pose and the three landmarks
    graph.emplace_shared<GenericStereoFactor<Pose3, Point3>>(StereoPoint2(570, 520, 490), model, 2, 3, K);
    graph.emplace_shared<GenericStereoFactor<Pose3, Point3>>(StereoPoint2(70, 20, 490), model, 2, 4, K);
    graph.emplace_shared<GenericStereoFactor<Pose3, Point3>>(StereoPoint2(320, 270, 115), model, 2, 5, K);

    // create Values object to contain initial estimates of camera poses and
    // landmark locations
    Values initial_estimate;

    // create and add iniital estimates
    initial_estimate.insert(1, first_pose);
    initial_estimate.insert(2, Pose3(Rot3(), Point3(0.1, -0.1, 1.1)));
    initial_estimate.insert(3, Point3(1, 1, 5));
    initial_estimate.insert(4, Point3(-1, 1, 5));
    initial_estimate.insert(5, Point3(0, -0.5, 5));

    // create Levenberg-Marquardt optimizer for resulting factor graph, optimize
    LevenbergMarquardtOptimizer optimizer(graph, initial_estimate);
    Values result = optimizer.optimize();

    result.print("Final result:\n");

    return 0;
}
```

bryanhpchiang, 2021-05-29:
Checking out 4.0.3 and building that from source worked instead.

bryanhpchiang, 2021-05-29:
Confirmed that Cython examples also work on 4.0.3.

ProfFan, 2021-05-29:
Note that you should NOT use a Makefile for GTSAM-related projects if you are not super familiar with GTSAM internals.

We recommend you use CMake and include GTSAM with `find_package`.

If you still want to know what went wrong, please first build with CMake, and then `make VERBOSE=1` to get the correct g++ arguments.

bryanhpchiang, 2021-05-29:
The Makefile was just for the minimal reproduction. 

The error occurs on the official examples under build/examples when I built GTSAM on the current develop branch.

varunagrawal, 2021-05-30:
I built the examples on my Ubuntu 20.04 machine and I cannot reproduce the error.

@bryanhpchiang can you please try using the CMake based build? I suspect some linking is not happening correctly.

```
cmake -DGTSAM_BUILD_EXAMPLES_ALWAYS=ON .. && make -j4 && ./examples/StereoVOExample
```

bryanhpchiang, 2021-05-30:
For sure, just tried building the `develop` branch again. The specific commit is 

Here's what I ran (I use the Python bindings):
```
cmake -DGTSAM_BUILD_EXAMPLES_ALWAYS=ON -DGTSAM_BUILD_PYTHON=1 -DGTSAM_PYTHON_VERSION=3.8.6 ..
make -j4
```

Result:
```
(base) bryan@ito:~/work/gtsam/build$ ./examples/StereoVOExample
./examples/StereoVOExample: symbol lookup error: ./examples/StereoVOExample: undefined symbol: _ZNK5gtsam6Factor9printKeysERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKSt8functionIFS6_mEE
```

Here's the full result from `cmake`:
```
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE - Success
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE - Failed
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.58") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- Found Eigen version: 3.3.7
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Found MKL: /home/bryan/miniconda3/include  
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found TBB: /usr/include (found suitable version "2020.1", minimum required is "4.4") found components: tbb tbbmalloc 
-- Building 3rdparty
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for getline
-- Looking for getline - found
-- checking for thread-local storage - found
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.1.0
-- Install prefix: /usr/local
-- Building GTSAM - shared: ON
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.1.0
[... 53 lines cut ...]
--  Eigen will use MKL and OpenMP                    : OpenMP found but GTSAM_WITH_EIGEN_MKL is disabled
--  Default allocator                                : TBB
--  Cheirality exceptions enabled                    : YES
--  Build with ccache                                : No
-- Packaging flags
--  CPack Source Generator                           : TGZ
--  CPack Generator                                  : TGZ
-- GTSAM flags                                               
--  Quaternions as default Rot3                      : Disabled
--  Runtime consistency checking                     : Disabled
--  Rot3 retract is full ExpMap                      : Enabled
--  Pose3 retract is full ExpMap                     : Enabled
--  Allow features deprecated in GTSAM 4.1           : Enabled
--  Metis-based Nested Dissection                    : Enabled
--  Use tangent-space preintegration                 : Enabled
-- MATLAB toolbox flags
--  Install MATLAB toolbox                           : Disabled
-- Python toolbox flags                                      
--  Build Python module with pybind                  : Enabled
--  Python version                                   : 3.8.6
-- ===============================================================
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bryan/work/gtsam/build
```

varunagrawal, 2021-05-30:
Okay, I've been using Clang-9 rather than GCC. Let me switch as get back to you. I don't think there is any other major difference (I use Boost 1.72.0).

varunagrawal, 2021-05-30:
Interesting. It works on GCC for me as well, I'm a bit confused at this point.

ProfFan, 2021-05-30:
I think it's something related to multiple GTSAM versions. 5 cents you have a global install of GTSAM that is not the same version as the source.

varunagrawal, 2021-05-30:
That's a great suggestion. Perhaps some of your 4.0.3 install is still lingering in your system.

bryanhpchiang, 2021-05-30:
Got it - that definitely makes sense since I might have installed GTSAM before with `apt-get`.

- How would I check for and remove a global install of GTSAM?
- Why would having multiple versions cause an error - is it the dynamic linking to the GTSAM .so file? 

Thanks!

ProfFan, 2021-05-31:
1. If you installed with apt-get, uninstall
2. Search for libgtsam.xxxxx in /usr/lib and /usr/local/lib, make sure there is none
3. make sure there is no /usr/include/gtsam and /usr/local/include/gtsam
4. Yes, it's a dynamic library, for details see ldconfig documentation.

varunagrawal, 2021-05-31:
Also check out `apt purge`

ProfFan, 2021-06-03:
Closing as stale

closed 2021-06-03 by ProfFan