borglab/gtsam #773: Redo data association on graph

fedetask, 2021-05-29:
I developed a Landmark based SLAM using the `NonlinearFactorGraph` and `LevenbergMarquardtOptimizer` and it works pretty well. I am now trying to correct bad data associations:

Assume a landmark `l` is measured twice and, due to noise, two landmarks `l_0` and `l_1` are created in the graph. GTSAM is able to correctly optimize their position, such that after optimization `l_0` and `l_1` are placed extremely close together. I would like then to merge the two, making them appear as a unique landmark to the `NonlinearFactorGraph`.

Is this possible somehow? I realized I cannot `rekey()` them (as explained [here](https://github.com/borglab/gtsam/issues/149)).
Do I need to store all measurements and re-create the graph every time after the optimization?

ProfFan, 2021-05-29:
Note that you are not supposed to `rekey`, but you can remove and add factors from/to a graph, no issues there. Just making copies of the factor is enough.

Hope this helps :)

fedetask, 2021-05-31:
I'm not sure I understood. Let's say I have the following graph, made of two poses `p1` and `p2` and two landmarks `l1` and `l2`
```c++
// ... noise models initialization
Symbol p0('P', 0);
Symbol p1('P', 1);
Symbol l0('L', 0);
Symbol l1('L', 1);

NonlinearFactorGraph graph;
graph.addPrior(p0, Pose2(0, 0, 0), pose_noise);

// Move from p0 to p1
graph.add(BetweenFactor<Pose2>(p0, p1, Pose2(2, 0, 0), pose_noise));

// From p1, we measure l0 and l1 that, due to bad data association, are detected as two different landmarks
graph.add(BearingRangeFactor<Pose2, Point2>(p1, l0, 0.01, 2.05, landmark_noise));
graph.add(BearingRangeFactor<Pose2, Point2>(p1, l1, -0.01, 1.98, landmark_noise));
```

Now, after optimizing the graph, I realize that `l0` and `l1` are actually the same landmark. I therefore want to remove `l1` and make the second `BearingRangeFactor` to point to `l0`. 

What I am trying to do is the following:

```c++
boost::shared_ptr<NonlinearFactor> f = graph.at(graph.size() - 1);  // Get factor of l1 measurement
// This is where I'm not sure:
// Modify factor so that it points to l0 instead of l1
std::map<Key, Key> rekey_map = {{f->keys()[1], l0.key()}}; 
boost::shared_ptr<NonlinearFactor> new_f = f->rekey(rekey_map);
graph.remove(graph.size() - 1);
graph.add(new_f);
```
Now, this throws a `Attempting to at the key "L1", which does not exist in the Values.` error if I try to optimize the graph.

fedetask, 2021-05-31:
While writing the comment above, I realized that simply doing 
```c++
graph.remove(graph.size() - 1);
graph.add(BearingRangeFactor<Pose2, Point2>(p1, l0, -0.01, 1.98, landmark_noise));
```
works. The problem is that this would require me to store all parameters of the measurement in order to re-create it with the new symbol. 

Therefore, my question becomes: Is it possible to either modify one of the factor symbols in-place, or is it possible to get the parameters of the factor returned by `graph.at()` so that I can `remove()` and then `add()`?

ProfFan, 2021-06-03:
Yes, please look at the code, every factor should expose a method to get the keys and params. If not, please add one and feel free to make a PR :)

closed 2021-06-03 by ProfFan