borglab/gtsam #778: Missing `marginalizeLeaves` for iSAM2

bryanhpchiang, 2021-05-30:
Ubuntu 20.04
gtsam 4.0.3

Using Python bindings.

According to the 4.1.0 documentation, there is a marginalizeLeaves function:

https://gtsam.org/doxygen/a04096.html

It's also in the 4.0.2 documentation as well:

https://gtsam-jlblanco-docs.readthedocs.io/en/latest/_static/doxygen/html/classgtsam_1_1ISAM2.html

But it's not available in the 4.0.3 version.

Minimal reproduction:
```
Python 3.8.6 | packaged by conda-forge | (default, Oct  7 2020, 19:08:05) 
[GCC 7.5.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gtsam
g>>>isam = gtsam.ISAM2()
>>> import inspect
>>> inspect.getmembers(isam)
[('__class__', <class 'gtsam.gtsam.ISAM2'>), ('__delattr__', <method-wrapper '__delattr__' of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__dir__', <built-in method __dir__ of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__doc__', None), ('__eq__', <method-wrapper '__eq__' of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__format__', <built-in method __format__ of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__ge__', <method-wrapper '__ge__' of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__getattribute__', <method-wrapper '__getattribute__' of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__gt__', <method-wrapper '__gt__' of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__hash__', <method-wrapper '__hash__' of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__init__', <method-wrapper '__init__' of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__init_subclass__', <built-in method __init_subclass__ of type object at 0x7ff5c8adb900>), ('__le__', <method-wrapper '__le__' of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__lt__', <method-wrapper '__lt__' of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__ne__', <method-wrapper '__ne__' of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__new__', <built-in method __new__ of type object at 0x7ff5c8adb900>), ('__pyx_vtable__', <capsule object NULL at 0x7ff5c77fb330>), ('__reduce__', <built-in method __reduce_cython__ of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__reduce_ex__', <built-in method __reduce_ex__ of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__repr__', <method-wrapper '__repr__' of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__setattr__', <method-wrapper '__setattr__' of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__setstate__', <built-in method __setstate_cython__ of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__sizeof__', <built-in method __sizeof__ of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__str__', <method-wrapper '__str__' of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('__subclasshook__', <built-in method __subclasshook__ of type object at 0x7ff5c8adb900>), ('calculateBestEstimate', <built-in method calculateBestEstimate of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimate', <built-in method calculateEstimate of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimateCal3Bundler', <built-in method calculateEstimateCal3Bundler of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimateCal3DS2', <built-in method calculateEstimateCal3DS2 of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimateCal3_S2', <built-in method calculateEstimateCal3_S2 of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimateEssentialMatrix', <built-in method calculateEstimateEssentialMatrix of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimateMatrix', <built-in method calculateEstimateMatrix of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimatePinholeCameraCal3_S2', <built-in method calculateEstimatePinholeCameraCal3_S2 of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimatePoint2', <built-in method calculateEstimatePoint2 of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimatePoint3', <built-in method calculateEstimatePoint3 of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimatePose2', <built-in method calculateEstimatePose2 of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimatePose3', <built-in method calculateEstimatePose3 of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimateRot2', <built-in method calculateEstimateRot2 of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimateRot3', <built-in method calculateEstimateRot3 of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimateSimpleCamera', <built-in method calculateEstimateSimpleCamera of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('calculateEstimateVector', <built-in method calculateEstimateVector of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('equals', <built-in method equals of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('getDelta', <built-in method getDelta of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('getFactorsUnsafe', <built-in method getFactorsUnsafe of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('getLinearizationPoint', <built-in method getLinearizationPoint of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('getVariableIndex', <built-in method getVariableIndex of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('marginalCovariance', <built-in method marginalCovariance of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('params', <built-in method params of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('printStats', <built-in method printStats of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('print_', <built-in method print_ of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('saveGraph', <built-in method saveGraph of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>), ('update', <built-in method update of gtsam.gtsam.ISAM2 object at 0x7ff5c8bfd930>)]

```

ProfFan, 2021-05-31:
Check `gtsam.i` to see if it is exported to Python. If it is not then you can export it yourself by adding the entry in gtsam.i

bryanhpchiang, 2021-06-01:
Got it, will do — thanks!

Sent via Superhuman ***@***.***>

closed 2021-06-03 by ProfFan