borglab/gtsam #782: Compilation error with Range factor

mc13756, 2021-06-04:
I want to create constraint for the distance between two Point2 variables (i.e. fixed distance of 1m). I'm trying to do it with a range factor like this:

`double door_width(1.0);
      
static SharedNoiseModel width_noise(noiseModel::Unit::Create(1));
      graph.emplace_shared<RangeFactor<Point2, Point2>>(next_door_num_left, next_door_num_right, door_width, width_noise);`

with next_door_num_left and next_door_num_right being integers used as variable keys. I get the following compilation error.:

/usr/local/include/gtsam/sam/RangeFactor.h:59:26: error: invalid use of incomplete type ‘struct gtsam::Range<Eigen::Matrix<double, 2, 1>, Eigen::Matrix<double, 2, 1> >’
     return Expression<T>(Range<A1, A2>(), a1_, a2_);
                          ^~~~~~~~~~~~~~~
In file included from /usr/local/include/gtsam/geometry/Pose2.h:23:0,
                 from /home/spyros/catkin_ws/src/door_detector_1/src/graph_creator_1.cpp:12:
/usr/local/include/gtsam/geometry/BearingRange.h:39:8: note: declaration of ‘struct gtsam::Range<Eigen::Matrix<double, 2, 1>, Eigen::Matrix<double, 2, 1> >’
 struct Range;

Where does this come from, and how can I fix it?

ProfFan, 2021-06-04:
Range is a trait and is not defined for `Point2`s. I think what you actually want is a BetweenFactor.

dellaert, 2021-06-04:
No, we just had this discussion on the Google group. It’s quite easy to add this trait for Point2, look at Point3.h. I offered that solution in the Google group and even asked for a PR, which might be forthcoming.

dellaert, 2021-06-04:
Not a bug, but a feature request. Solution discussed in Google group.

closed 2021-06-04 by dellaert

dellaert, 2021-06-04:
Will be fixed by #783