borglab/gtsam #787: Can not find PythonConfig.cmake when cmake Matlab wrapper

luokerenxxz, 2021-06-08:
The installation without Matlab wrapper is fine. But I got a cmake warning, which means cmake in “warp” can not find python, when setting ‘-DGTSAM_INSTALL_MATLAB_TOOLBOX=ON -DGTSAM_PYTHON_VERSION=3.6’.  This leads to an error when compilling.

I am working on a clean Ubuntu 18.04 using VMWare virtual machine.

I tried to set 

'-DPYTHON_INCLUDE_DIR=$(python3 -c "from distutils.sysconfig import get_python_inc; print(get_python_inc())")‘
’-DPYTHON_LIBRARY=$(python3 -c "import distutils.sysconfig as sysconfig; print(sysconfig.get_config_var('LIBDIR'))") ‘
’-DPYTHON_EXECUTABLE:FILEPATH=/usr/bin/python3'

But it doesn't work.

Thanks！
 
[image]
[image]
[image]

ProfFan, 2021-06-08:
Did you follow the tutorials *exactly* and install the package `python3-dev`?

luokerenxxz, 2021-06-09:
> Did you follow the tutorials _exactly_ and install the package `python3-dev`?

Yes, I followed the instructions in https://github.com/borglab/gtsam/blob/develop/matlab/README.md, and updated the wrap library. Also I tried `sudo apt-get install python3-dev`, it showed that python 3.6.9 has already been installed.

ProfFan, 2021-06-14:
Maybe @varunagrawal has some idea on this? Also @luokerenxxz Please post your entire log in text here.

ProfFan, 2021-07-15:
@luokerenxxz Do you still have this issue?

varunagrawal, 2021-08-01:
Closing due to lack of a response. Please re-open if the issue persists.

closed 2021-08-01 by varunagrawal