borglab/gtsam #801: Couldn't run the example code

OoooL19, 2021-06-21:
I tried to run ```make CameraResectioning```, but I got this. I am new to gtsam. I try to learn the gtsam from the example. Can anyone help me with this issue? I am on ubuntu 18 and using eigen 3.3.9.  I use developer branch of the gtsam. Thank you.
```
g++     CameraResectioning.cpp   -o CameraResectioning
/tmp/ccKitNic.o: In function `main':
CameraResectioning.cpp:(.text+0x18c): undefined reference to `gtsam::noiseModel::Diagonal::Sigmas(Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, bool)'
CameraResectioning.cpp:(.text+0x59f): undefined reference to `gtsam::Rot3::Rot3(double, double, double, double, double, double, double, double, double)'
CameraResectioning.cpp:(.text+0x63a): undefined reference to `gtsam::LevenbergMarquardtOptimizer::LevenbergMarquardtOptimizer(gtsam::NonlinearFactorGraph const&, gtsam::Values const&, gtsam::LevenbergMarquardtParams const&)'
CameraResectioning.cpp:(.text+0x65e): undefined reference to `gtsam::Values::Values(gtsam::Values const&)'
CameraResectioning.cpp:(.text+0x6c8): undefined reference to `gtsam::Values::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (unsigned long)> const&) const'
/tmp/ccKitNic.o: In function `__static_initialization_and_destruction_0(int, int)':
CameraResectioning.cpp:(.text+0x942): undefined reference to `gtsam::_defaultKeyFormatter[abi:cxx11](unsigned long)'
CameraResectioning.cpp:(.text+0x972): undefined reference to `gtsam::_multirobotKeyFormatter[abi:cxx11](unsigned long)'
/tmp/ccKitNic.o: In function `gtsam::Symbol::operator unsigned long() const':
CameraResectioning.cpp:(.text._ZNK5gtsam6SymbolcvmEv[_ZNK5gtsam6SymbolcvmEv]+0x14): undefined reference to `gtsam::Symbol::key() const'
/tmp/ccKitNic.o: In function `gtsam::ValuesKeyDoesNotExist::ValuesKeyDoesNotExist(char const*, unsigned long)':
CameraResectioning.cpp:(.text._ZN5gtsam21ValuesKeyDoesNotExistC2EPKcm[_ZN5gtsam21ValuesKeyDoesNotExistC5EPKcm]+0x23): undefined reference to `vtable for gtsam::ValuesKeyDoesNotExist'
/tmp/ccKitNic.o: In function `gtsam::ValuesKeyDoesNotExist::~ValuesKeyDoesNotExist()':
CameraResectioning.cpp:(.text._ZN5gtsam21ValuesKeyDoesNotExistD2Ev[_ZN5gtsam21ValuesKeyDoesNotExistD5Ev]+0xf): undefined reference to `vtable for gtsam::ValuesKeyDoesNotExist'
/tmp/ccKitNic.o: In function `gtsam::ValuesIncorrectType::ValuesIncorrectType(unsigned long, std::type_info const&, std::type_info const&)':
CameraResectioning.cpp:(.text._ZN5gtsam19ValuesIncorrectTypeC2EmRKSt9type_infoS3_[_ZN5gtsam19ValuesIncorrectTypeC5EmRKSt9type_infoS3_]+0x27): undefined reference to `vtable for gtsam::ValuesIncorrectType'
/tmp/ccKitNic.o: In function `gtsam::ValuesIncorrectType::~ValuesIncorrectType()':
CameraResectioning.cpp:(.text._ZN5gtsam19ValuesIncorrectTypeD2Ev[_ZN5gtsam19ValuesIncorrectTypeD5Ev]+0xf): undefined reference to `vtable for gtsam::ValuesIncorrectType'
/tmp/ccKitNic.o: In function `gtsam::NonlinearFactor::~NonlinearFactor()':
CameraResectioning.cpp:(.text._ZN5gtsam15NonlinearFactorD2Ev[_ZN5gtsam15NonlinearFactorD5Ev]+0xf): undefined reference to `vtable for gtsam::NonlinearFactor'
/tmp/ccKitNic.o: In function `gtsam::NoiseModelFactor::~NoiseModelFactor()':
CameraResectioning.cpp:(.text._ZN5gtsam16NoiseModelFactorD2Ev[_ZN5gtsam16NoiseModelFactorD5Ev]+0xf): undefined reference to `vtable for gtsam::NoiseModelFactor'
/tmp/ccKitNic.o: In function `gtsam::NonlinearOptimizerParams::~NonlinearOptimizerParams()':
CameraResectioning.cpp:(.text._ZN5gtsam24NonlinearOptimizerParamsD2Ev[_ZN5gtsam24NonlinearOptimizerParamsD5Ev]+0xf): undefined reference to `vtable for gtsam::NonlinearOptimizerParams'
/tmp/ccKitNic.o: In function `gtsam::NonlinearOptimizer::optimize()':
CameraResectioning.cpp:(.text._ZN5gtsam18NonlinearOptimizer8optimizeEv[_ZN5gtsam18NonlinearOptimizer8optimizeEv]+0x14): undefined reference to `gtsam::NonlinearOptimizer::defaultOptimize()'
CameraResectioning.cpp:(.text._ZN5gtsam18NonlinearOptimizer8optimizeEv[_ZN5gtsam18NonlinearOptimizer8optimizeEv]+0x20): undefined reference to `gtsam::NonlinearOptimizer::values() const'
/tmp/ccKitNic.o: In function `gtsam::NonlinearOptimizerParams::NonlinearOptimizerParams()':
CameraResectioning.cpp:(.text._ZN5gtsam24NonlinearOptimizerParamsC2Ev[_ZN5gtsam24NonlinearOptimizerParamsC5Ev]+0xf): undefined reference to `vtable for gtsam::NonlinearOptimizerParams'
/tmp/ccKitNic.o: In function `gtsam::LevenbergMarquardtParams::LevenbergMarquardtParams()':
CameraResectioning.cpp:(.text._ZN5gtsam24LevenbergMarquardtParamsC2Ev[_ZN5gtsam24LevenbergMarquardtParamsC5Ev]+0x1b): undefined reference to `vtable for gtsam::LevenbergMarquardtParams'
/tmp/ccKitNic.o: In function `gtsam::LevenbergMarquardtParams::~LevenbergMarquardtParams()':
CameraResectioning.cpp:(.text._ZN5gtsam24LevenbergMarquardtParamsD2Ev[_ZN5gtsam24LevenbergMarquardtParamsD5Ev]+0xf): undefined reference to `vtable for gtsam::LevenbergMarquardtParams'
/tmp/ccKitNic.o: In function `gtsam::LevenbergMarquardtOptimizer::~LevenbergMarquardtOptimizer()':
CameraResectioning.cpp:(.text._ZN5gtsam27LevenbergMarquardtOptimizerD2Ev[_ZN5gtsam27LevenbergMarquardtOptimizerD5Ev]+0xf): undefined reference to `vtable for gtsam::LevenbergMarquardtOptimizer'
CameraResectioning.cpp:(.text._ZN5gtsam27LevenbergMarquardtOptimizerD2Ev[_ZN5gtsam27LevenbergMarquardtOptimizerD5Ev]+0x32): undefined reference to `gtsam::NonlinearOptimizer::~NonlinearOptimizer()'
/tmp/ccKitNic.o: In function `gtsam::Pose3::operator*(gtsam::Pose3 const&) const':
CameraResectioning.cpp:(.text._ZNK5gtsam5Pose3mlERKS0_[_ZNK5gtsam5Pose3mlERKS0_]+0x50): undefined reference to `gtsam::Rot3::operator*(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&) const'
CameraResectioning.cpp:(.text._ZNK5gtsam5Pose3mlERKS0_[_ZNK5gtsam5Pose3mlERKS0_]+0xaa): undefined reference to `gtsam::Rot3::operator*(gtsam::Rot3 const&) const'
/tmp/ccKitNic.o: In function `gtsam::Cal3::Cal3(double, double, double, double, double)':
CameraResectioning.cpp:(.text._ZN5gtsam4Cal3C2Eddddd[_ZN5gtsam4Cal3C5Eddddd]+0x24): undefined reference to `vtable for gtsam::Cal3'
/tmp/ccKitNic.o: In function `gtsam::Cal3::~Cal3()':
CameraResectioning.cpp:(.text._ZN5gtsam4Cal3D2Ev[_ZN5gtsam4Cal3D5Ev]+0xb): undefined reference to `vtable for gtsam::Cal3'
/tmp/ccKitNic.o: In function `gtsam::Cal3_S2::Cal3_S2(double, double, double, double, double)':
CameraResectioning.cpp:(.text._ZN5gtsam7Cal3_S2C2Eddddd[_ZN5gtsam7Cal3_S2C5Eddddd]+0x65): undefined reference to `vtable for gtsam::Cal3_S2'
/tmp/ccKitNic.o: In function `gtsam::Cal3::Cal3(gtsam::Cal3 const&)':
CameraResectioning.cpp:(.text._ZN5gtsam4Cal3C2ERKS0_[_ZN5gtsam4Cal3C5ERKS0_]+0xf): undefined reference to `vtable for gtsam::Cal3'
/tmp/ccKitNic.o: In function `gtsam::Cal3_S2::Cal3_S2(gtsam::Cal3_S2 const&)':
CameraResectioning.cpp:(.text._ZN5gtsam7Cal3_S2C2ERKS0_[_ZN5gtsam7Cal3_S2C5ERKS0_]+0x26): undefined reference to `vtable for gtsam::Cal3_S2'
/tmp/ccKitNic.o: In function `void gtsam::Values::insert<gtsam::Pose3>(unsigned long, gtsam::Pose3 const&)':
CameraResectioning.cpp:(.text._ZN5gtsam6Values6insertINS_5Pose3EEEvmRKT_[_ZN5gtsam6Values6insertINS_5Pose3EEEvmRKT_]+0x65): undefined reference to `gtsam::Values::insert(unsigned long, gtsam::Value const&)'
/tmp/ccKitNic.o: In function `gtsam::NoiseModelFactor::NoiseModelFactor<boost::assign_detail::static_generic_list<unsigned long const, 1> >(boost::shared_ptr<gtsam::noiseModel::Base> const&, boost::assign_detail::static_generic_list<unsigned long const, 1> const&)':
CameraResectioning.cpp:(.text._ZN5gtsam16NoiseModelFactorC2IN5boost13assign_detail19static_generic_listIKmLi1EEEEERKNS2_10shared_ptrINS_10noiseModel4BaseEEERKT_[_ZN5gtsam16NoiseModelFactorC5IN5boost13assign_detail19static_generic_listIKmLi1EEEEERKNS2_10shared_ptrINS_10noiseModel4BaseEEERKT_]+0x2a): undefined reference to `vtable for gtsam::NoiseModelFactor'
/tmp/ccKitNic.o: In function `Eigen::Matrix<double, 2, 1, 0, 2, 1> gtsam::PinholeBaseK<gtsam::Cal3_S2>::_project<Eigen::Matrix<double, 3, 1, 0, 3, 1> >(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, gtsam::OptionalJacobian<2, 6>, gtsam::OptionalJacobian<2, gtsam::FixedDimension<Eigen::Matrix<double, 3, 1, 0, 3, 1> >::value>, gtsam::OptionalJacobian<2, 5>) const':
CameraResectioning.cpp:(.text._ZNK5gtsam12PinholeBaseKINS_7Cal3_S2EE8_projectIN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEEEENS5_IdLi2ELi1ELi0ELi2ELi1EEERKT_NS_16OptionalJacobianILi2ELi6EEENSB_ILi2EXsrNS_14FixedDimension [... cut ...]
CameraResectioning.cpp:(.text._ZN5gtsam8internal6handleINS_5Pose3EEclEmPKNS_5ValueE[_ZN5gtsam8internal6handleINS_5Pose3EEclEmPKNS_5ValueE]+0xbc): undefined reference to `typeinfo for gtsam::ValuesIncorrectType'
/tmp/ccKitNic.o: In function `gtsam::LieGroup<gtsam::Pose3, 6>::between(gtsam::Pose3 const&) const':
CameraResectioning.cpp:(.text._ZNK5gtsam8LieGroupINS_5Pose3ELi6EE7betweenERKS1_[_ZNK5gtsam8LieGroupINS_5Pose3ELi6EE7betweenERKS1_]+0x4d): undefined reference to `gtsam::Pose3::inverse() const'
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'CameraResectioning' failed
make: *** [CameraResectioning] Error 1
```

ProfFan, 2021-06-21:
Closed as non-issue.

Please follow the documentation on how to build examples properly **with CMake**. If you have further questions please use the mailing list.

closed 2021-06-21 by ProfFan

OoooL19, 2021-06-21:
> Closed as non-issue.
> 
> Please follow the documentation on how to build examples properly **with CMake**. If you have further questions please use the mailing list.

Yes, I did followed the example Readme, but I have this error. 
```
CMake Error at CMakeLists.txt:6 (gtsamAddExamplesGlob):
  Unknown CMake command "gtsamAddExamplesGlob".
```
When you said the use the mailing list. Did you mean email you guys? I am sorry I am also new to the github.

ProfFan, 2021-06-21:
Please post your entire CMake command line and log so we can diagnose the issue.

OoooL19, 2021-06-22:
> Please post your entire CMake command line and log so we can diagnose the issue.

In the examples folder I did
```
mkdir build
cd build
cmake ..
```
[CMakeOutput.log](https://github.com/borglab/gtsam/files/6690073/CMakeOutput.log)

ProfFan, 2021-06-22:
This is not the CMake log, sorry but what I mean is the command and result that appeared on your terminal.
Please look at the Issue template for details.

OoooL19, 2021-06-22:
> This is not the CMake log, sorry but what I mean is the command and result that appeared on your terminal.
> Please look at the Issue template for details.

I am sorry, these are the output. 
```
owen@owen-Terminator-T430:~/gtsam/examples$ mkdir build
owen@owen-Terminator-T430:~/gtsam/examples$ cd build
owen@owen-Terminator-T430:~/gtsam/examples/build$ cmake ..
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:6 (gtsamAddExamplesGlob):
  Unknown CMake command "gtsamAddExamplesGlob".

-- Configuring incomplete, errors occurred!
See also "/home/owen/gtsam/examples/build/CMakeFiles/CMakeOutput.log".
```

ProfFan, 2021-06-22:
This is not correct, but I think the documentation may need more clarification.

What you should do is run these commands under the *top* folder, not inside `example` folder.

```
mkdir build
cd build
cmake ..
```

We'll update the docs later.

OoooL19, 2021-06-22:
> This is not correct, but I think the documentation may need more clarification.
> 
> What you should do is run these commands under the _top_ folder, not inside `example` folder.
> 
> ```
> mkdir build
> cd build
> cmake ..
> ```
> 
> We'll update the docs later.

Ok, In that case, I already did the build in the top folder before. Is there anything I missed, so I cannot do ``` make CameraResectioning ```

ProfFan, 2021-06-22:
Again, please submit the console logs.

OoooL19, 2021-06-22:
> Again, please submit the console logs.

yes, the first pose is the log I had when I ran the ```make CameraResectioning```
or you want the ```ccmake ..``` log?

ProfFan, 2021-06-22:
1. remove everything
2. clone fresh repo
3. `mkdir build && cd build && cmake ..` (don't use ccmake if you don't understand CMake), logs
4. `make`, logs
5. `make CameraResectioning`, logs.

OoooL19, 2021-06-22:
> 1. remove everything
> 2. clone fresh repo
> 3. `mkdir build && cd build && cmake ..` (don't use ccmake if you don't understand CMake), logs
> 4. `make`, logs
> 5. `make CameraResectioning`, logs.

This is the log I had when I tried to run ```make CameraResectioning``` after I did the previous steps.
```
owen@owen-Terminator-T430:~/gtsam/build$ make CameraResectioning
[ 30%] Built target metis-gtsam
[100%] Built target gtsam
[100%] Built target CameraResectioning
owen@owen-Terminator-T430:~/gtsam/build$ make CameraResectioning.run
[ 30%] Built target metis-gtsam
[100%] Built target gtsam
[100%] Built target CameraResectioning
./CameraResectioning: symbol lookup error: ./CameraResectioning: undefined symbol: _ZTVN5gtsam6FactorE
examples/CMakeFiles/CameraResectioning.run.dir/build.make:70: recipe for target 'examples/CMakeFiles/CameraResectioning.run' failed
make[3]: *** [examples/CMakeFiles/CameraResectioning.run] Error 127
CMakeFiles/Makefile2:20669: recipe for target 'examples/CMakeFiles/CameraResectioning.run.dir/all' failed
make[2]: *** [examples/CMakeFiles/CameraResectioning.run.dir/all] Error 2
CMakeFiles/Makefile2:20676: recipe for target 'examples/CMakeFiles/CameraResectioning.run.dir/rule' failed
make[1]: *** [examples/CMakeFiles/CameraResectioning.run.dir/rule] Error 2
Makefile:8792: recipe for target 'CameraResectioning.run' failed
make: *** [CameraResectioning.run] Error 2
```

ProfFan, 2021-06-22:
You did not remove everything and start from clean. Make sure you post ALL logs from step one.

OoooL19, 2021-06-23:
> You did not remove everything and start from clean. Make sure you post ALL logs from step one.

Hi, after I tried couple times. I finally make it works. Thank you so much.

irisgro, 2023-04-19:
i meet the same question, could i know how did you solve this problem at last?