borglab/gtsam #809: #800 causes Levenberg-Marquardt optimisation to crash

HMellor, 2021-06-30:
## Description

When updating to the latest commit of the `develop` branch my existing codebase that uses GTSAM started crashing when running LM. 

I have narrowed it down to 6ee17c2ed7b2a89f4a25ae568009bb9385061ad1 which comes from #800.  Checking out the previous commit causes the program to no longer crash.

It may help to list some classes I'm using:
* NonlinearFactorGraph
* CombinedImuFactor
* GPSFactor

## Expected behavior

Should not crash

## Environment

 * **OS**
    * Distributor ID: Ubuntu
    * Description:    Ubuntu 20.04.2 LTS
    * Release:        20.04
    * Codename:       focal
* **TBB Version (relevant to #800)**
    * libtbb-dev/focal,now 2020.1-2 amd64 [installed]
    * libtbb2/focal,now 2020.1-2 amd64 [installed,automatic]
* **Language**
    * C++

ProfFan, 2021-07-06:
@acxz Could you take a look?

ProfFan, 2021-07-06:
Also, @HMellor could you provide a minimal reproducer (one cpp file, no other dependencies)?

dellaert, 2021-07-07:
@acxz this was your PR - could you weigh in? @HMellor thatnks for letting us know the TBB versions you used.

acxz, 2021-07-07:
This might be an issue similar to why the testGncOptimize test failed once out of 100 runs. I could take a deeper look into this issue with a minimal reproducing example.

In the meanwhile, it may be better to just revert my PR.

closed 2021-07-17 by ProfFan