borglab/gtsam #810: Serializing ISAM2

ScottMcMichael, 2021-07-06:
## Description

I am trying to back up an ISAM2 solver object to disk but it does not work properly once it is loaded back from file.  This is the gtest code I used to test it (I can also provide the includes and exports used):

```
TEST(Gtsam, serializeTest) {

  gtsam::ISAM2Params parameters;
  gtsam::ISAM2 solver(parameters);
  gtsam::NonlinearFactorGraph graph;
  gtsam::Values initialValues;
  initialValues.clear();

  gtsam::Vector6 temp6;
  for (int i = 0; i < 6; ++i) {
    temp6[i] = 0.0001;
  }
  gtsam::noiseModel::Diagonal::shared_ptr noiseModel = gtsam::noiseModel::Diagonal::Sigmas(temp6);

  gtsam::Pose3 pose0(gtsam::Rot3(), gtsam::Point3(0, 0, 0));
  gtsam::Symbol symbol0('x', 0);
  graph.add(gtsam::PriorFactor<gtsam::Pose3>(symbol0, pose0, noiseModel));
  initialValues.insert(symbol0, pose0);

  solver.update(graph, initialValues, gtsam::FastVector<size_t>());

  std::string binaryPath = "saved_solver.dat";
  {
    std::ofstream outputStream(binaryPath);
    boost::archive::binary_oarchive outputArchive(outputStream);
    outputArchive << solver;
  }

  gtsam::ISAM2 solverFromDisk;
  {
    std::ifstream ifs(binaryPath);
    boost::archive::binary_iarchive inputArchive(ifs);  
    inputArchive >> solverFromDisk;
  }

  EXPECT_NO_THROW(solver.calculateEstimate<gtsam::Pose3>(symbol0)); // Works
  EXPECT_NO_THROW(solverFromDisk.calculateEstimate<gtsam::Pose3>(symbol0)); // Throws
}
```

The exception text is:  Requested variable 'x0' is not in this VectorValues.

## Environment

I am using gtsam version 4.0.3 on Ubuntu 20.04.

## Additional information

Looking through the source code, the ISAM2 class inherits the "serialize" function from the BayesTree parent class.  ISAM2 adds several member values such as delta_ and theta_ which are not included in the serialize function and this seems to be where the problem comes from.  I'm not sure if it would be appropriate to write a new serialize function which includes these or if ISAM2 is not intended to be serialized.

varunagrawal, 2021-07-07:
This seems straightforward to fix by adding the necessary Boost Name-Value-Pairs (`nvp`). Am I missing something?

@ScottMcMichael I don't have a lot of bandwidth right now to fix this for the next few weeks, but if you submit a PR I can provide reviews and recommendations.

ScottMcMichael, 2021-07-09:
Created PR #816

closed 2021-07-19 by ProfFan