borglab/gtsam #815: CMake handling of a missing TBB leads to crash

johnwlambert, 2021-07-09:
On Mac, I would expect a missing TBB to be handled gracefully. Instead, cmake crashes:

```bash
mkdir build
cd build
cmake .. -DGTSAM_BUILD_PYTHON=1 -DGTSAM_PYTHON_VERSION=3.8.10
```
WIth the following error report

```bash
-- The CXX compiler identification is AppleClang 12.0.5.12050022
-- The C compiler identification is AppleClang 12.0.5.12050022
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE - Success
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE - Failed
-- Found Boost: /usr/local/lib/cmake/Boost-1.76.0/BoostConfig.cmake (found suitable version "1.76.0", minimum required is "1.58") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- Found Eigen version: 3.3.7
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES) 
-- Could NOT find OpenMP_CXX (missing: OpenMP_CXX_FLAGS OpenMP_CXX_LIB_NAMES) 
-- Could NOT find OpenMP (missing: OpenMP_C_FOUND OpenMP_CXX_FOUND) 
CMake Error at cmake/FindTBB.cmake:184 (file):
  file failed to open for reading (No such file or directory):

    /usr/local/include/tbb/tbb_stddef.h
Call Stack (most recent call first):
  cmake/HandleTBB.cmake:3 (find_package)
  CMakeLists.txt:50 (include)

-- Could NOT find TBB: Found unsuitable version ".", but required is at least "4.4" (found /usr/local/include, found components: tbb tbbmalloc)
-- Building 3rdparty
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for getline
-- Looking for getline - found
-- checking for thread-local storage - found
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.1.0
-- Install prefix: /usr/local
-- Building GTSAM - shared: ON
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.1.0
-- Install prefix: /usr/local
-- Found Python: /Users/johnlam/opt/miniconda3/envs/gtsam-v2/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter Development Development.Module Development.Embed 
-- gtwrap Package config : /usr/local/lib/cmake/gtwrap
-- gtwrap version        : 1.0
-- gtwrap CMake path     : /usr/local/lib/cmake/gtwrap
-- gtwrap library path   : /usr/local/lib/gtwrap
-- gtwrap binary path    : /usr/local/bin/gtwrap
-- gtwrap header path    : /usr/local/include/gtwrap
-- Checking Python Version
-- Setting Python version for wrapper
-- pybind11 v2.6.0 dev1
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Performing Test HAS_FLTO_THIN
-- Performing Test HAS_FLTO_THIN - Success
-- Wrote /Users/johnlam/Documents/gtsam/build/GTSAMConfig.cmake
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- ===============================================================
-- ================  Configuration Options  ======================
--  CMAKE_CXX_COMPILER_ID type                       : Clang
--  CMAKE_CXX_COMPILER_VERSION                       : 12.0.5.12050022
--  CMake version                                    : 3.20.3
--  CMake generator                                  : Unix Makefiles
--  CMake build tool                                 : /usr/bin/make
-- Build flags                                               
--  Build Tests                                      : Enabled
--  Build examples with 'make all'                   : Enabled
--  Build timing scripts with 'make all'             : Disabled
--  Build shared GTSAM libraries                     : Enabled
--  Put build type in library name                   : Enabled
--  Build libgtsam_unstable                          : Enabled
--  Build GTSAM unstable Python                      : Enabled
--  Build MATLAB Toolbox for unstable                : Disabled
--  Build for native architecture                    : Enabled
--  Build type                                       : Release
--  C compilation flags                              :  -O3 -DNDEBUG
--  C++ compilation flags                            :  -O3 -DNDEBUG
--  GTSAM_COMPILE_FEATURES_PUBLIC                    : cxx_std_11
--  GTSAM_COMPILE_OPTIONS_PUBLIC                     : -ftemplate-depth=1024;-march=native
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC                 : 
--  GTSAM_COMPILE_OPTIONS_PUBLIC_DEBUG               : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_DEBUG           : 
--  GTSAM_COMPILE_OPTIONS_PUBLIC_RELEASE             : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_RELEASE         : 
--  GTSAM_COMPILE_OPTIONS_PUBLIC_TIMING              : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_TIMING          : 
--  GTSAM_COMPILE_OPTIONS_PUBLIC_PROFILING           : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_PROFILING       : 
--  GTSAM_COMPILE_OPTIONS_PUBLIC_RELWITHDEBINFO      : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_RELWITHDEBINFO  : 
--  GTSAM_COMPILE_OPTIONS_PUBLIC_MINSIZEREL          : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_MINSIZEREL      : 
--  Use System Eigen                                 : OFF (Using version: 3.3.7)
--  Use Intel TBB                                    : TBB not found
--  Eigen will use MKL                               : MKL not found
--  Eigen will use MKL and OpenMP                    : OpenMP not found
--  Default allocator                                : STL
--  Cheirality exceptions enabled                    : YES
--  Build with ccache                                : No
-- Packaging flags
--  CPack Source Generator                           : TGZ
--  CPack Generator                                  : TGZ
-- GTSAM flags                                               
--  Quaternions as default Rot3                      : Disabled
--  Runtime consistency checking                     : Disabled
--  Rot3 retract is full ExpMap                      : Enabled
--  Pose3 retract is full ExpMap                     : Enabled
--  Allow features deprecated in GTSAM 4.1           : Enabled
--  Metis-based Nested Dissection                    : Enabled
--  Use tangent-space preintegration                 : Enabled
-- MATLAB toolbox flags
--  Install MATLAB toolbox                           : Disabled
-- Python toolbox flags                                      
--  Build Python module with pybind                  : Enabled
--  Python version                                   : 3.8.10
-- ===============================================================
CMake Warning at cmake/HandleFinalChecks.cmake:3 (message):
  TBB 4.4 or newer was not found - this is ok, but note that GTSAM
  parallelization will be disabled.  Set GTSAM_WITH_TBB to 'Off' to avoid
  this warning.
Call Stack (most recent call first):
  CMakeLists.txt:120 (include)

-- Configuring incomplete, errors occurred!
See also "/Users/johnlam/Documents/gtsam/build/CMakeFiles/CMakeOutput.log".
See also "/Users/johnlam/Documents/gtsam/build/CMakeFiles/CMakeError.log".
```

Environment:
Python 3.8.10
macOS Big Sur, Version 11.2.3

varunagrawal, 2021-07-10:
The error message says it all

```
CMake Error at cmake/FindTBB.cmake:184 (file):
  file failed to open for reading (No such file or directory):

    /usr/local/include/tbb/tbb_stddef.h
Call Stack (most recent call first):
  cmake/HandleTBB.cmake:3 (find_package)
  CMakeLists.txt:50 (include)

-- Could NOT find TBB: Found unsuitable version ".", but required is at least "4.4" (found /usr/local/include, found components: tbb tbbmalloc)
```

Either you haven't turned off the TBB flag or your TBB install is messed up somehow.

johnwlambert, 2021-07-10:
Hi @varunagrawal, thanks for taking a look. It looks like I was trying `-DGTSAM_WITH_TBB=0` instead of `-DGTSAM_WITH_TBB=OFF`. We could clarify that in the docs.

But I guess my two cents are that if TBB is truly optional, then CMake should be able to gracefully check if it's found, and if not, should set the flag to `OFF` itself and proceed with a successful compilation.

ProfFan, 2021-07-10:
I think the current thing should be documented better but the behavior should not change: one could accidentally build a GTSAM without TBB and think he is building with it, leading to serious performance regressions.

johnwlambert, 2021-07-10:
Fair point. I took a crack at clarifying the docs: https://github.com/borglab/gtsam/pull/818

varunagrawal, 2021-07-10:
I believe CMake setting flags based on internal logic can be a bad idea since then the behavior is no longer opaque. The user will be wondering why CMake is trying to find TBB when the flag was never set. If TBB is not found at all, then CMake does handle it correctly and lets the user know.

In general, the issue is more that TBB is currently default to ON, whereas the correct default (for it to be truly optional) is OFF.

closed 2021-07-11 by varunagrawal