borglab/gtsam #824: Drafting 4.1 Release

ProfFan, 2021-07-19:
The `develop` branch has been pretty stable now, so it's high time that we actually release a version 4.1.0 of GTSAM.

The Python packages are now built automatically on https://github.com/borglab/gtsam-manylinux-build/actions. Please download the artifacts there and test with your code base.

Outstanding issues and PRs:
- #595 

Nice to haves:
- #433 

@varunagrawal @jlblancoc @varunagrawal @gchenfc @dellaert  and all,
feel free to add the issues and PRs you think are important to this list :)

jlblancoc, 2021-07-22:
Sure! I just edited it to add the 2nd and 3rd items in the "nice to have" section ;-)

varunagrawal, 2021-07-25:
System Eigen to `ON`? That seems suspect since we might get more issues about Eigen related errors from people who don't read error logs.

ProfFan, 2021-07-25:
It's twofold:
- `ON`, and we get issues from people with bad Eigen installs
- `OFF`, and we get issues from ABI compatibility and mysterious errors originated from different options of Eigen.

I think `ON` is slightly better, since these should mostly be `it doesn't work` type of issues...

asa, 2021-08-01:
Is there a python osx wheel I could test for this build? I can't seem to get it to build locally for me. Pip installs an old build (4.0.3).

ProfFan, 2021-08-01:
@asa https://github.com/borglab/gtsam-manylinux-build/actions

Click artifacts and download

asa, 2021-08-01:
I was able to do that and successfully install an artifact, thank you. These actions are not run on develop branch as far as I see however.  I suppose a branch will be cut for 4.1 in a pr and the artifacts will be built then... Looks like I should keep banging on getting my local builds to work...  These artifacts get me relatively recent bits though.   Thanks!

ProfFan, 2021-11-22:
We are very close to doing a 4.1 release (since the PyPI packages are very out-of-date now). Currently we only need #188 blocking, but we do have a lot of nice-to-haves outstanding.

Pinging everyone here for a last sprint on getting more in the release :)

@varunagrawal  @dellaert  @lucacarlone @jlblancoc @acxz

varunagrawal, 2021-11-23:
#368 is solved but needs some major documentation updates which will only be done in December. We can push it to the next release.

lucacarlone, 2021-11-23:
I'm happy to push on PR #861 if there is interest. It has been ready for a while, but I'll need to fix a CI error.

jlblancoc, 2021-11-25:
There's a remaining issue which might need to be listed as "must fix" before release: apparently, when building a subgraph by selecting factors from a FG and inserting them into a new FG, some kind of memory corruption happens.
I've been avoding this problem in an application by serializing-and-deserializing to a temporary file to ensure the new subgraph doesn't make the optimizer to crash, but I should put together a proper bug report with a minimal test case... I'll try to do it asap.

ProfFan, 2021-11-25:
@jlblancoc Could you post a PoC in an issue so we can help reproduce?

lucacarlone, 2021-11-25:
I think we have seen similar segfaults on our side. @jlblancoc : is this in conjunction with smart factors or also happens for other factors?

jlblancoc, 2021-11-28:
@ProfFan  It was 100% reproducible in my app but now I can't reproduce it anymore so can't provide a PoC... I'll report in a new issue if it shows up again.
@lucacarlone Nope, it was with "non-smart" factors only...

ProfFan, 2021-12-05:
@jlblancoc Can you reproduce it? If not we may just do the release and fix in 4.1.1?

ProfFan, 2021-12-05:
Let's cut the release next Tuesday

dellaert, 2021-12-05:
That would be amazing. Note I do not want to change the default behavior of Eigen at this moment, at least not without extensive discussion on the pros and cons

jlblancoc, 2021-12-06:
> @jlblancoc Can you reproduce it? If not we may just do the release and fix in 4.1.1?

@ProfFan Sure, go ahead.

ProfFan, 2021-12-07:
we may delay cutting branch due to #954

ProfFan, 2021-12-19:
I am back from quals (yay!) so let's do that. I'll cut the branch, compile and release artifacts at the time @dellaert makes the call.

dellaert, 2021-12-20:
Great. Actually, before cutting anything, can we talk? I’ve since then also looked at the pypi distributions from GTSam, and I wonder whether we should tag versions that have the same version number. Send me an invite for tomorrow if you can make it

ProfFan, 2021-12-20:
4.1.1 is out! We'll do a bigger 4.2 shortly after the discrete changes per discussions w/ @dellaert .

closed 2021-12-20 by ProfFan