borglab/gtsam #825: Python gtwrap: ModuleNotFoundError: No module named 'loguru'

carlmorgan, 2021-07-19:
## Description
Build failure for gtsam wrapper - additional dependency added / not checked for / installed.

Probably introduced as part of commit 4c410fcd0ebe1869400f6ea601e21792da74f019 by @varunagrawal 

gtwrap/matlab_wrapper/wrapper.py:14
+from loguru import logger

## Build log
```[ 84%] Built target gtsam
[ 84%] Generating ../wrap/gtsam/gtsam_wrapper.cpp
Traceback (most recent call last):
  File "/export/bulk/jenkins/workspace/gtsam_crown_upstream_develop/wrap/scripts/matlab_wrap.py", line 11, in <module>
    from gtwrap.matlab_wrapper import MatlabWrapper
  File "/export/bulk/jenkins/workspace/gtsam_crown_upstream_develop/wrap/gtwrap/matlab_wrapper/__init__.py", line 3, in <module>
    from .wrapper import MatlabWrapper
  File "/export/bulk/jenkins/workspace/gtsam_crown_upstream_develop/wrap/gtwrap/matlab_wrapper/wrapper.py", line 14, in <module>
    from loguru import logger
ModuleNotFoundError: No module named 'loguru'
make[2]: *** [matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:72: wrap/gtsam/gtsam_wrapper.cpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:30720: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....```

## Expected behavior
To complete compile :)

## Environment
Ubuntu Focal Fossa 20.04 with MatLab 2020b.  

## Additional information
CI based nightly upstream builds - has been running fine for previous weeks.

varunagrawal, 2021-07-20:
Can you do a `pip install loguru` and try again? I am pretty sure I configured CMake to install the python dependencies (even setup.py should do it for us) but I guess this is specific to your python environment.

varunagrawal, 2021-07-20:
Basically the problem is that we use loguru for logging in the wrapper and while installing the wrapper, it should also be installed as a dependency.

Knowing more about your python environment and version would be super useful.

carlmorgan, 2021-07-20:
Understand the loguru is the dependency / issue. 

Stock Focal platform, gcc 9.3.0 / Python 3.8 etc

Questions:
* Is it really needed?  
gtwrap/matlab_wrapper/wrapper.py and tests/test_matlab_wrapper.py are the only imports I can see, and only tests uses for a change to info logging. 

wrap/tests/test_matlab_wrapper.py
```
# set the log level to INFO by default
logger.remove()  # remove the default sink
logger.add(sys.stderr, format="{time} {level} {message}", level="INFO")
```

* Installing pip requirements into the host OS isn't a great solution in our case - the CI build farm infrastructure is shared across multiple projects / languages / solutions and has minimal base OS dependencies (i.e. risks of conflicts).  (Also multiple platforms / archs in play)
* If that lib is _really_ needed - building a venv setups as part of the tooling environment might be an option.
* I can see the requirements.txt has it in - but doesn't look like standard cmake / make is picking that up and/or failing

Is this only needed for the generation of the MEX wrapper components or needed more generally for py bindings etc?

[ I've also just done a local build and even with the loguru locally installed, and looks like other wrapper functions are failing to compile - I'll raise separately after a little more digging ]

ProfFan, 2021-07-20:
@varunagrawal Maybe we should add a wrapper on `logger.log` and make loguru an optional dependency?

varunagrawal, 2021-07-21:
@carlmorgan thanks for the input. It seems like `loguru` isn't quite needed, and reducing dependencies is always a good idea. Let me remove it.

PR #830 (2021-07-24): Wrap Update
Merge new wrapper updates.

Fixes #825.
closed 2021-07-24 by varunagrawal