borglab/gtsam #834: Better Windows 10 Build Manual

geoeo, 2021-07-31:
## Feature

Better Windows 10 Installation/Build Manual

## Motivation

Not sure about other Cmake Gurus, but I personally could not get this thing to build on Windows 10.
And it is stated that it is a supported platform.#

There are multiple steps of 3rd party libs that have to be built (Boost, Metis) and even then I found that 
I had to edit the CMakeLists.txt to get it to at least accept my CMake config.

I tried with VSCode and Visual Studio 2019.

## Pitch

A section in the readme detailing the tools and steps necessary for building this on windows

varunagrawal, 2021-08-01:
We already have one: https://github.com/borglab/gtsam/blob/develop/INSTALL.md#windows-installation

You need to make sure Visual Studio is able to find Metis and link to it (aka this is a pure linker issue). On Linux, we can do `sudo ldconfig` so you may want to google something similar for Windows.

closed 2021-08-01 by varunagrawal

geoeo, 2021-08-01:
Thanks for the link. I seem to have missed that. A note about the 3rd party libs which have to be built separately would be nice imho. 

Thanks again.

geoeo, 2021-08-04:
P.s The build on windows is failing on 4.03 maybe thats part of the problem

varunagrawal, 2021-08-04:
Gotcha. You may want to try the develop branch since we've made lots of improvements for Windows support since. I wrote up that doc after weeks of trying to figure out the right way to build on Visual Studio so if you have any feedback/input, I'd highly welcome a PR. :)

geoeo, 2021-08-07:
I had to disable the `ShonanAveragingCLI.cpp`  example to get it to compile on windows 10 develop -  4.1rc. But it works now! I can file an issue request with the output if thats of interest to you

varunagrawal, 2021-08-09:
Yeah absolutely! I actually like writing docs, what I hate is developing for Windows. :sweat_smile:

NickPerezCarletonUniversity, 2021-08-13:
> We already have one: https://github.com/borglab/gtsam/blob/develop/INSTALL.md#windows-installation

@varunagrawal it should be more visible... as in viewable on the main Github page.

stefangachter, 2024-12-11:
I work under Windows using conan packages for boost, metis, etc. (based on https://github.com/conan-io/conan-center-index/tree/master/recipes/gtsam). I can successfully build gtsam (in Release and RelWithDebInfo, but not Debug yet). I adapted the recipe for development purposes. However, I am not yet able to build and run the test. Built test is enabled:
````
Build Tests                                      : Enabled
````
But I cannot find the "binaries", that is, I have not yet figured out how to run the tests under Windows. Any guidance is welcome.

stefangachter, 2024-12-12:
I got it: if build test is enabled, solutions are created, but tests are not built because they are excluded from cmake target `all`.  The individual test can be build with `cmake --build build --config Debug --target check_base_program` and then test can be run in the build directory with `ctest -C Debug -R check_base_program`

stefangachter, 2024-12-12:
Note that the above conan recipe (`conanfile.py`) is not designed for "development" and does some unnecessary patching.
1) conan recipe is designed for gtsam 4.2 (tags/4.2, 4f66a491f). Otherwise, patching will fail.
2) Remove patching of `GtsamBuildTypes.cmake` in `conanfile.py`. Otherwise, it is not possible to switch between build types.
3) If you want to build tests, then set the flag to true in  `conanfile.py`
4) After first build `conan build .`, disable  `self._patch_sources()` in `conanfile.py`

For installing `conan install . gtsam/4.2@ --build=missing --profile:host=windows --profile:build=windows` and building `conan build .` gtsam, I used the following options stored in `windows`:
```
    [settings]
    os=Windows
    arch=x86_64
    compiler=Visual Studio
    compiler.version=17
    build_type=Debug

    [options]
    boost:without_test=True
    boost:without_stacktrace=True
    boost:without_locale=True
    gtsam:build_type_postfixes=False
    gtsam:enable_consistency_checks=True
    gtsam:install_cppunitlite=False
    gtsam:pose3_expmap=True
    gtsam:use_quaternions=False
    gtsam:rot3_expmap=True
    gtsam:slow_but_correct_betweenfactor=True
    gtsam:tangent_preintegration=False
    gtsam:with_TBB=False
    metis:with_64bit_types=False
 ```
 Note that patching changes some of the cmake files: `HandleBoost.cmake`,  `HandleEigen.cmake`, `HandleGlobalBuildFlags.cmake`, `HandleMetis.cmake`, `HandleMetis.cmake`, `CMakeLists.txt` of `libmetis`, and main `CMakeLists.txt`.