borglab/gtsam #837: gtsam_unstable IncrementalFixedLagSmoother.h vtable error

epicjung, 2021-08-02:
## Description
Hi, I am trying to use "IncrementalFixedLagSmoother" in the GTSAM unstable. The compile error occurs when I want to compile a c++ program as follows: 
**Version 1:**
void optimization::optimize()
{
  double lag = 10.0;
  ISAM2Params params; 
  params.relinearizeThreshold = 0.1;
  params.relinearizeSkip = 1;
  gtsam::IncrementalFixedLagSmoother smoother(lag, params);
}

**Version 2:** (This compiles fine)
void optimization::optimize()
{
  double lag = 10.0;
  ISAM2Params params; 
  params.relinearizeThreshold = 0.1;
  params.relinearizeSkip = 1;
  gtsam::IncrementalFixedLagSmoother smoother();
}

The version 2 works fine, where there are no parameters in the constructor. However, when I input parameters as in Version 1, the terminal shows something like this:

CMakeFiles/sensor_fusion_optimization.dir/src/optimization.cpp.o: In function `optimization::optimize()':
/usr/local/include/gtsam_unstable/nonlinear/IncrementalFixedLagSmoother.h:43: undefined reference to `vtable for gtsam::IncrementalFixedLagSmoother'
CMakeFiles/sensor_fusion_optimization.dir/src/optimization.cpp.o: In function `optimization::optimize()':
/usr/local/include/gtsam_unstable/nonlinear/FixedLagSmoother.h:69: undefined reference to `vtable for gtsam::FixedLagSmoother'
CMakeFiles/sensor_fusion_optimization.dir/src/optimization.cpp.o: In function `gtsam::IncrementalFixedLagSmoother::~IncrementalFixedLagSmoother()':
/usr/local/include/gtsam_unstable/nonlinear/IncrementalFixedLagSmoother.h:47: undefined reference to `vtable for gtsam::IncrementalFixedLagSmoother'
CMakeFiles/sensor_fusion_optimization.dir/src/optimization.cpp.o: In function `gtsam::IncrementalFixedLagSmoother::~IncrementalFixedLagSmoother()':
/usr/local/include/gtsam_unstable/nonlinear/FixedLagSmoother.h:69: undefined reference to `vtable for gtsam::FixedLagSmoother'
CMakeFiles/sensor_fusion_optimization.dir/src/optimization.cpp.o: In function `gtsam::IncrementalFixedLagSmoother::~IncrementalFixedLagSmoother()':
/usr/local/include/gtsam_unstable/nonlinear/IncrementalFixedLagSmoother.h:47: undefined reference to `vtable for gtsam::IncrementalFixedLagSmoother'
CMakeFiles/sensor_fusion_optimization.dir/src/optimization.cpp.o: In function `gtsam::IncrementalFixedLagSmoother::~IncrementalFixedLagSmoother()':
/usr/local/include/gtsam_unstable/nonlinear/FixedLagSmoother.h:69: undefined reference to `vtable for gtsam::FixedLagSmoother'
collect2: error: ld returned 1 exit status
sensor_fusion/CMakeFiles/sensor_fusion_optimization.dir/build.make:808: recipe for target '/home/euigon/fusion_ros/devel/lib/sensor_fusion/sensor_fusion_optimization' failed
make[2]: *** [/home/euigon/fusion_ros/devel/lib/sensor_fusion/sensor_fusion_optimization] Error 1
CMakeFiles/Makefile2:1151: recipe for target 'sensor_fusion/CMakeFiles/sensor_fusion_optimization.dir/all' failed
make[1]: *** [sensor_fusion/CMakeFiles/sensor_fusion_optimization.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

## Steps to reproduce

1. 
2.

## Expected behavior

## Environment
C++, Ubuntu 18.04, GTSAM "develop" branch

## Additional information

varunagrawal, 2021-08-20:
@epicjung can you please provide a minimal reproducible example? Just the code snippets provided don't give us sufficient info, a full example would be much more insightful.

Binbin418916, 2021-11-24:
add `gtsam_unstable` in target_link_libraries, 
like
target_link_libraries(exe_file
    gtsam
    gtsam_unstable
)

dellaert, 2021-12-22:
@Binbin418916 Small PR?

Williamwenda, 2022-01-13:
> add `gtsam_unstable` in target_link_libraries, like target_link_libraries(exe_file gtsam gtsam_unstable )

This solution works out

varunagrawal, 2022-01-13:
From the comments it seems that the author was not linking gtsam_unstable correctly in their project. Since a solution has been reached and there is no need for a PR, we're closing this.

closed 2022-01-13 by varunagrawal

RomanStadlhuber, 2024-12-08:
@varunagrawal When installing from [`ros-noetic-gtsam`](https://index.ros.org/p/gtsam/github-borglab-gtsam/#noetic), I cannot do this, i.e. adding

```cmake
find_package(GTSAM REQUIRED)

add_library(xyz xyz.cxx)

target_link_libraries(xyz
  gtsam
  gtsam_unstable
)
```

and then `cmake --build ../build`  I get

```bash
[build] /usr/bin/ld: cannot find -lgtsam_unstable
```

I am using CMake v. 3.16.3 from ROS Noetic on Ubuntu 20.04.

When omitting `gtsam_unstable` stuff in my source code and the respective library in CMakeLists.txt, linking to regular gtsam works just fine.
Is there something I am missing?

varunagrawal, 2024-12-09:
@RomanStadlhuber how are you installing GTSAM? From source or from the ubuntu package?

RomanStadlhuber, 2024-12-09:
@varunagrawal I have installed it from the ubuntu package with `apt install ros-noetic-gtsam`.

I have found out in the meantime, that I also needed to add `find_package(GTSAM_UNSTABLE REQUIRED)` in order to make it work.