borglab/gtsam #840: ShonanAveragingCLI example fails to build on Windows 10 on version 4.1rc

geoeo, 2021-08-09:
## Description

ShonanAveragingCLI does not compile on Windows 10 4.1rc.
 I removed the example for the build process via 
`gtsamAddExamplesGlob("*.cpp" "elaboratePoint2KalmanFilter.cpp;ShonanAveragingCLI.cpp;" "gtsam;${Boost_PROGRAM_OPTIONS_LIBRARY}")`

for the CMakeLists.txt in the examples folder in order to compile.

(Disabling eloboratePont2KalmanFilter was already there)

## Steps to reproduce

1.  Check out 4.1rc on Windows 10
2. Add Boost dependencies
3. Build All

You will receive 
```
       Creating library D:/Workspace/Libraries/gtsam/build/x64-Debug (default)/bin/Debug/ShonanAveragingCLI.lib and object D:/Workspace/Libraries/gtsam/build/x64-Debug (default)/bin/Debug/ShonanAveragingCLI.exp
D:\Workspace\Libraries\gtsam\build\x64-Debug (default)\gtsam\ShonanAveragingCLI.obj : error LNK2019: unresolved external symbol "public: __cdecl gtsam::ShonanAveraging2::ShonanAveraging2(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,struct gtsam::ShonanAveragingParameters<2> const &)" (??0ShonanAveraging2@gtsam@@QEAA@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBU?$ShonanAveragingParameters@$01@1@@Z) referenced in function main 
D:\Workspace\Libraries\gtsam\build\x64-Debug (default)\gtsam\ShonanAveragingCLI.obj : error LNK2019: unresolved external symbol "public: __cdecl gtsam::ShonanAveraging3::ShonanAveraging3(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,struct gtsam::ShonanAveragingParameters<3> const &)" (??0ShonanAveraging3@gtsam@@QEAA@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBU?$ShonanAveragingParameters@$02@1@@Z) referenced in function main 
D:\Workspace\Libraries\gtsam\build\x64-Debug (default)\bin\Debug\ShonanAveragingCLI.exe : fatal error LNK1120: 2 unresolved externals
```

## Expected behavior

A clean compilation

## Environment

Windows 10 Visual Studio 2019

## Additional information

varunagrawal, 2021-08-20:
Interesting. This is usually a `GTSAM_EXPORT` issue but `ShonanAveraging2` and `ShonanAveraging3` already have that specifier declared. This might need a lot of investigation.

chrisgitguud, 2021-12-17:
This appears to be fixed in commit 347ea5d805328a1fb5e014ee1f838cdc890fcccd.

varunagrawal, 2021-12-18:
@geoeo can you check if the latest `develop` branch works for you?