borglab/gtsam #841: gtsam.utilities.extractPoint2() doesn't work

dishank-b, 2021-08-10:
gtsam.utilities.extractPoint2(Values) return empty array, even if Values have Point2 variables. 

## Steps to reproduce

1. `initial = gtsam.Values()`
2.  `initial.insert(1, gtsam.Pose2(0.0, 0.1, 0.1)`
3.   `initial.insert(2, gtsam.Point2(0.0, 0.1))`
4.   `gtsam.utilities.extractPoint2(initial)` returns `array([], shape=(0, 2), dtype=float64)`
5.  Where as `gtsam.utilities.extractPose2(initial)` return correctly which is `array([[0. , 0.1, 0.1]])`
  

Expted behavior of `gtsam.utilities.extractPoint2(initial)` is returning `array([[0. , 0.1]])`

varunagrawal, 2021-08-12:
Valid bug. The issue is the underlying C++ function.

varunagrawal, 2021-08-20:
Okay so did a bit of investigating and the underlying C++ function is correct (verified with a unit test).
The issue seems to be that the `values.filter` call expects a `Point2` in C++ but that type info is not being relayed in Python. As a result, when I print out the filtered values in C++, it prints as empty.

I believe this is because `Point2` is stored as an `Eigen::Matrix21` but when converting from numpy to Eigen, Pybind changes the type to `Eigen::Matrix<double, -1, -1>` and thus the filter no longer works.

dishank-b, 2021-08-20:
So what can be done about this? Do you know any alternative way to filter out Point2 type from Values() ?

varunagrawal, 2021-08-20:
Already on it. Will make the PR in about 30 minutes.

closed 2021-08-20 by varunagrawal

dishank-b, 2021-08-23:
How do I get the fixed version? Will I have to pull and install again?

varunagrawal, 2021-08-23:
Yup. `git pull origin develop` should get you the latest fixes.

PR #851 (2021-08-20): Fix utilities and add tests
Fixes #841