borglab/gtsam #847: NonlinearFactorGraph deserialization doesn't work

lynnkse, 2021-08-15:
## Description

FactorGraph deserialization from string doesn't work. Deserialize function returns with no errors, but the factorgraph remains empty

## Steps to reproduce
Code that reproduces the issue and prints "deserialization error!" if the deserialization fails, "deserialization successfull" otherwise
```
import gtsam

#create NonlinearFactorGraph with one factor
graph = gtsam.NonlinearFactorGraph()

current_key = gtsam.symbol('A', 0)
current_rot = gtsam.Rot3(0.000796, 1, -4.907, -1, 0.000796, 3.5175, -3.522, -5.418, 1)
current_point = gtsam.Point3(20, 0, 0)
current_pose = gtsam.Pose3(current_rot, current_point)
prior_model = gtsam.noiseModel.Diagonal.Sigmas([0.000001, 0.000001, 0.000001, 0.001, 0.001, 0.001])
f_prior = gtsam.PriorFactorPose3(current_key, current_pose, prior_model)

graph.push_back(f_prior)

#serialize the FG
ser_graph = graph.serialize()

#create new empty factor graph and deserialize from the string above
deser_graph = gtsam.NonlinearFactorGraph()
deser_graph.deserialize(ser_graph)

#lengths of the original FG and the one created above must be equal now
if deser_graph.size() == graph.size():
    print("deserialization successfull")
else:
    print("deserialization error!")
```

## Expected behavior

deser_graph must contain the same factors as the original FG (graph) after deserialization

## Environment

GTSAM 4.1.0
Python 3.8.10
Ubuntu 20.04

## Additional information

The same works in C++. The problem seems to be in Python wrapper

varunagrawal, 2021-08-20:
I cannot reproduce this bug. @lynnkse are you using the 4.1.0 release or the latest develop branch?

lynnkse, 2021-08-22:
@varunagrawal yes I'm using the 4.1.0 release. Had it worked correctly for you, i.e. "deserialization successfull" is printed out?

varunagrawal, 2021-08-22:
Yes, I get the success message.

ProfFan, 2021-08-22:
@lynnkse Could you try building in clean (fresh clone, fresh build folder and no `ccache`?

lynnkse, 2021-08-29:
@ProfFan yes I did just that, on multiple computers in fact. Which Python release do you use?

varunagrawal, 2021-08-29:
I personally use 3.8.11. Did you try using the latest develop branch?

lynnkse, 2021-09-01:
@varunagrawal ok, it works correctly on the latest develop branch

varunagrawal, 2021-09-01:
Sweet. Closing this as resolved.

closed 2021-09-01 by varunagrawal