borglab/gtsam #849: Verbose cheirality in ProjectionFactorPPP and ProjectionFactorPPPC prints wrong key for landmark

eirik-ff, 2021-08-17:
## Description

If verbose cheirality is enabled in ProjectionFactorPPP(C), it will print out a message similar to `CheiralityException: Landmark c0 moved behind camera x0`. Here, `C(0)` is the calibration key and `X(0)` is the pose. The error is that the landmark key is in fact the calibration key. 

## Steps to reproduce

1. Fail the cheirality check with verbose cheirality enabled when using ProjectionFactorPPP(C).

Example code to reproduce:

```c++
#include <gtsam/inference/Symbol.h>
#include <gtsam/nonlinear/Values.h>
#include <gtsam/nonlinear/NonlinearFactorGraph.h>
#include <gtsam_unstable/slam/ProjectionFactorPPP.h>
#include <gtsam_unstable/slam/ProjectionFactorPPPC.h>
#include <gtsam/nonlinear/LevenbergMarquardtOptimizer.h>

using gtsam::symbol_shorthand::C; // body-sensor calibration
using gtsam::symbol_shorthand::K; // camera matrix
using gtsam::symbol_shorthand::L; // landmarks
using gtsam::symbol_shorthand::X; // body pose

int main(int argc, char *argv[])
{
    gtsam::NonlinearFactorGraph graph;
    gtsam::Values initial;

    gtsam::Cal3_S2::shared_ptr cameraMatrix(new gtsam::Cal3_S2(1, 1, 0, 0, 0));
    auto noise = gtsam::noiseModel::Isotropic::Sigma(2, 1e-2);

    gtsam::Point2 p(1, 2);
    graph.add(gtsam::ProjectionFactorPPP<gtsam::Pose3, gtsam::Point3, gtsam::Cal3_S2>(p, noise, X(0), C(0), L(0), cameraMatrix, false, true));
    graph.add(gtsam::ProjectionFactorPPPC<gtsam::Pose3, gtsam::Point3, gtsam::Cal3_S2>(p, noise, X(0), C(0), L(1), K(0), false, true));

    initial.insert<gtsam::Pose3>(X(0), gtsam::Pose3());
    initial.insert<gtsam::Pose3>(C(0), gtsam::Pose3());
    initial.insert<gtsam::Cal3_S2>(K(0), *cameraMatrix);
    initial.insert<gtsam::Point3>(L(0), gtsam::Point3(0, 0, -1));
    initial.insert<gtsam::Point3>(L(1), gtsam::Point3(0, 0, -1));

    gtsam::LevenbergMarquardtOptimizer optimizer(graph, initial);
    gtsam::Values result = optimizer.optimize();

    return 0;
}
```

## Expected behavior

Instead, the expected output when a cheirality exception is thrown is e.g. `CheiralityException: Landmark l0 moved behind camera x0`.

## Environment

Using code from commit `f7ef6f703`, however it is not fixed in the most recent commit as of writing this issue. The relevant files are: [ProjectionFactorPPP.h:145](https://github.com/borglab/gtsam/blob/develop/gtsam_unstable/slam/ProjectionFactorPPP.h#L145) and [ProjectionFactorPPPC.h:142](https://github.com/borglab/gtsam/blob/develop/gtsam_unstable/slam/ProjectionFactorPPPC.h#L141). 

## Fix

This is simply fixed by changing the mentioned lines above to:

```c++
if (verboseCheirality_)
          std::cout << e.what() << ": Landmark "<< DefaultKeyFormatter(this->key3()) <<
              " moved behind camera " << DefaultKeyFormatter(this->key1()) << std::endl;
```

I.e., the change is changing `this->key2()` to `this->key3()`. 

This is a very small thing, but it is nice to make sure it reports errors correctly.

varunagrawal, 2021-09-06:
Makes sense that this would be buggy since it is a part of `gtsam_unstable`. If it is urgent @eirik-ff, can you please submit a PR? I'll be happy to help review and get it merged at the earliest.

varunagrawal, 2021-09-15:
Gotcha. This issue is not on our radar anytime soon, which is why if you need it resolved for something, we would be gladly welcome a PR.