borglab/gtsam #852: Please include enough detail to run the tutorial code

FranzEricSchneider, 2021-08-20:
This will be the most beginner of questions, but could you include the appropriate #include lines to build and run tutorial code? I am following along with [the tutorial](https://gtsam.org/tutorials/intro.html), which looks great, but I am trying to compile and run the example code alongside to both get a feel for the process and make sure everything is installed correctly. For example, in section 2.2 there is the first tutorial code segment. I tried copying it in chunk by chunk to see if I could figure out the right #include statements, and started with

```
// Add a Gaussian prior on pose x_1
Pose2 priorMean(0.0, 0.0, 0.0);
noiseModel::Diagonal::shared_ptr priorNoise =
  noiseModel::Diagonal::Sigmas(Vector3(0.3, 0.3, 0.1));
```

When I tried to compile it complained about Pose2 that's fine, I looked through the gtsam source code and found examples of `#include <gtsam/geometry/Pose2.h>`. Then it starts complaining about noiseModel:
```
#include <gtsam/geometry/Pose2.h>
using namespace gtsam;
int main()
{
    // Add a Gaussian prior on pose x_1
    Pose2 priorMean(0.0, 0.0, 0.0);
    noiseModel::Diagonal::shared_ptr priorNoise =
      noiseModel::Diagonal::Sigmas(Vector3(0.3, 0.3, 0.1));
    return 0;
}
___________________________________________________
||=== Build: Debug in gtsam_tutorial (compiler: GNU GCC Compiler) ===|
gtsam_tutorial/main.cpp||In function ‘int main()’:|
gtsam_tutorial/main.cpp|9|error: ‘noiseModel’ has not been declared|
||=== Build failed: 1 error(s), 0 warning(s) (0 minute(s), 5 second(s)) ===|
```

At this point I got a bit stuck. Couldn't find a related #include anywhere in the source code. I think it would be really valuable to make the tutorial examples easily runnable, since then people learning can alter and re-run them to learn the tools a bit.

dellaert, 2021-08-21:
The examples are all on GitHub. I don’t have time to fix this myself but feel free to do a PR on gtsam.org, and I’ll review and merge.

closed 2021-08-21 by dellaert

FranzEricSchneider, 2021-08-21:
Thanks for the suggestion to look at the examples directly, the README there was very useful. Added a very small PR (https://github.com/borglab/gtsam/pull/853) against some README instructions.