borglab/gtsam #878: Details about IMU Integration Covariance

varunagrawal, 2021-09-18:
## Feature

@lucacarlone and I need to document what the Integration Covariance is and its role in the `ImuFactor`.

dellaert, 2021-09-18:
Please be careful: ImuFactor.pdf is about my new Tangent-space formulation. The Forster et al. papers are the reference for the manifold pre-integration version. I don't mind expanding and re-capping that math in this file as well, but that is a more substantial undertaking.

mattking-smith, 2023-03-13:
@lucacarlone or @dellaert I haven't come across the paper detailing the math on the `Integration Covariance` from Forster. Do you have specific paper in mind? So far I am able to calibrate my IMU setup with an Alan Variance (AV) analysis, but I essentially "trial and error" the integration covariance as the AV analysis doesn't determine this information.

xavierhubbard-mitll, 2024-02-21:
Wondering if there are any details on this anywhere now? I have been treating integration covariance as a hyperparameter to be tuned, which feels incorrect.