borglab/gtsam #881: Not installing the libgtsam-unstable-dev package causes build failures

mantkiew, 2021-09-28:
## Description

On Ubuntu, when a user installs only `libgtsam-dev` package and then tries to use the library using CMake, the following error is produced:

```
CMake Error at /usr/lib/cmake/GTSAM/GTSAM-exports.cmake:100 (message):
  The imported target "gtsam_unstable" references the file
     "/usr/lib/x86_64-linux-gnu/libgtsam_unstable.so.4.0.3"
  but this file does not exist.
```

That is, the cmake file references `libgtsam_unstable.so.4.0.3` file which belongs to the `libgtsam-unstable-dev` package. 

Consequently, it is not possible to only install `libgtsam-dev` without installing `libgtsam-unstable-dev`.

## Steps to reproduce

1. `apt-get install libgtsam-dev`
2. find the library in CMakeLists.txt as usual
3. build and it'll fail with the abovementioned error.

## Expected behavior

When the user installs only  `libgtsam-dev`  package, the CMake files installed should not refer to the files from the `libgtsam-unstable-dev` package.

## Environment

Ubuntu 18.04 Bionic

mantkiew, 2021-09-28:
Maybe this could be added to https://github.com/borglab/gtsam/issues/824

ProfFan, 2021-09-28:
Pinging @jlblancoc who is the expert on packaging.

mantkiew, 2021-09-30:
Thank you, I'll try that. However, I suggest making this option to be `OFF` by default if the unstable package is not required. Otherwise, the option can stay `ON` by default if the unstable package becomes installed automatically.

jlblancoc, 2021-10-03:
It seems a real CMake exports bug: 

> CMake Error at /usr/lib/cmake/GTSAM/GTSAM-exports.cmake:100 (message):

There should be two export files, one for `gtsam` and another for `gtsam_unstable`. 

I have no time for full testing now, but I think the error is the repeated ` EXPORT GTSAM-exports` in both: 
- https://github.com/borglab/gtsam/blob/develop/gtsam/CMakeLists.txt#L177-L182
- and https://github.com/borglab/gtsam/blob/develop/gtsam_unstable/CMakeLists.txt#L101-L106

The latter should be ` EXPORT gtsam-unstable-exports`, but probably other minor changes should be made around for consistency...