borglab/gtsam #883: PriorFactor<Pose3> susceptible to rotation when sigma is infinity

Jochen0x90h, 2021-10-01:
## Description

The PriorFactor<Pose3> seems to be susceptible to the rotation of the prior pose even if the sigma of the noise model is infinity. In this case it should have no influence. The IMUKittiExampleGPS.cpp example uses a current estimate for the rotation. When this is replaced by an identity rotation (Rot3()) then the result doesn't look good any more. Is this a bug or do I conceptually miss something here?

## Steps to reproduce

Take examples/IMUKittiExampleGPS.cpp and modify "Create GPS factor" at line 283 as follows so that the rotation component of the pose of the prior factor is identity:

            // Create GPS factor
            auto gps_pose = Pose3(current_pose_global.rotation(), gps_measurements[i].position);
            auto gps_pose2 = Pose3(Rot3(), gps_measurements[i].position);
            if ((i % gps_skip) == 0) {
                new_factors.emplace_shared<PriorFactor<Pose3>>(current_pose_key, gps_pose2, noise_model_gps);
                new_values.insert(current_pose_key, gps_pose);

                printf("################ POSE INCLUDED AT TIME %lf ################\n", t);
                cout << gps_pose.translation();
                printf("\n\n");
            }

## Expected behavior

When the sigma for rotation is infinity, there should be no difference in the output IMUKittiExampleGPSResults.csv

## Environment

Latest develop branch, but the behavior is the same for version 4.0.3

Jochen0x90h, 2021-10-01:
Using PoseTranslationPrior<Pose3> instead of PriorFactor<Pose3> with infinity sigma for rotation seems to work

varunagrawal, 2022-04-02:
If the covariance is infinity, why is the prior being added? It is the same as saying there is no prior so it makes sense that rotation will be affected since the optimization is free to move the values around as required to minimize the total error.

If you're still having issues, I recommend asking on the Google Group since this doesn't seem to be a code related issue.

closed 2022-04-02 by varunagrawal