borglab/gtsam #886: Buggy axis-angle computation in GTSAM

johnwlambert, 2021-10-05:
GTSAM returns angle's of well over 180 degrees when transforming a Rot3() object with `.axisAngle()`.

Here's an example below, where the angle returned is 204 degrees. Scipy's Rotation package returns the same axis vector as GTSAM, but returns the angle as 179 degrees, which seems legitimate.

I believe the angle of the angle-axis representation should be confined to [0, pi] radians, per [Duke](http://motion.pratt.duke.edu/RoboticSystems/3DRotations.html#:~:text=Axis%2Dangle%20representation%20uses%20a,3%2DD%20space%20of%20rotations.) and the notion of being smallest rotation to take us there (we would go the opposite way, instead, otherwise):

Given the following 3x3 matrix:
```python
(Pdb) p relative_rot
R: [
	-0.0309726, -0.789768, -0.612625;
	-0.787144, -0.358463, 0.501909;
	-0.615993, 0.497768, -0.610558
]
```
GTSAM provides an axis-angle decomposition:
```python
(Pdb) p relative_rot.axisAngle()
(:-0.696212
 0.566258
 0.441181
, 3.576080425974451)
```
Angle is 204 degrees:
```python
(Pdb) np.rad2deg( 3.576080425974451)
204.89431560768165
```

Scipy states the angle is 179.93 degrees:
```python
(Pdb) from scipy.spatial.transform import Rotation
(Pdb) Rotation.from_matrix(relative_rot.matrix()).as_rotvec()
array([-2.18469647,  1.77759913,  1.38497925])
(Pdb) axisangle = Rotation.from_matrix(relative_rot.matrix()).as_rotvec()
(Pdb) np.linalg.norm(axisangle)
3.138618297175063
(Pdb) np.rad2deg(3.138618297175063)
179.8295819306683
```

Scipy's axis of rotation is identical:
```python
(Pdb) axisangle /= np.linalg.norm(axisangle)
(Pdb) axisangle
array([-0.6960695 ,  0.56636359,  0.44127037])
```

johnwlambert, 2021-10-06:
The angle can be 300 - 800 degrees in other cases:
```python
>>> import numpy as np
>>> import gtsam
>>> R = np.array([[ -0.999957, 0.00922903, 0.00203116],[ 0.00926964, 0.999739, 0.0208927],[ -0.0018374, 0.0209105, -0.999781 ]])
>>> R.T @ R
array([[ 1.00000330e+00,  1.66519550e-07, -4.07243092e-07],
       [ 1.66519550e-07,  1.00000049e+00,  7.20413748e-08],
       [-4.07243092e-07,  7.20413748e-08,  1.00000268e+00]])
>>> np.linalg.det(R)
1.0000032373629155
>>> from gtsam import Rot3
>>> Rot3(R)
R: [
	-0.999957, 0.00922903, 0.00203116;
	0.00926964, 0.999739, 0.0208927;
	-0.0018374, 0.0209105, -0.999781
]
>>> Rot3(R).axisAngle()
(:0.00460089
  0.999934
 0.0104968
, 6.075185318083525)
>>> np.rad2deg(6.075185318083525)
348.08247848602855
>>> Rot3(R).matrix().T @ Rot3(R).matrix()
array([[ 1.00000330e+00,  1.66519550e-07, -4.07243092e-07],
       [ 1.66519550e-07,  1.00000049e+00,  7.20413748e-08],
       [-4.07243092e-07,  7.20413748e-08,  1.00000268e+00]])
>>> from scipy.spatial.transform import Rotation
>>> Rotation.from_matrix(Rot3(R).matrix()).as_rotvec()
array([0.01452083, 3.13945318, 0.03281409])
>>> scipy_angle_axis = Rotation.from_matrix(Rot3(R).matrix()).as_rotvec()
>>> np.linalg.norm(scipy_angle_axis)
3.139658247605024
>>> np.rad2deg(np.linalg.norm(scipy_angle_axis))
179.88916670120787
>>> scipy_angle_axis / np.linalg.norm(scipy_angle_axis)
array([0.00462497, 0.99993469, 0.01045148])
```

ProfFan, 2021-10-07:
Hey @johnwlambert could you put up a PR with a unit test that reproduces this?

dellaert, 2021-10-08:
Do you have an idea how to fix it?

dellaert, 2021-10-11:
Also, ClosestTo can be expensive, so the computational expectations of this piece of code would suddenly change, so we should be careful or find a computation less sensitive to small deviations from the manifold.

johnwlambert, 2021-10-11:
Thanks for looking into this, @ProfFan.

It's interesting to see how this very small deviation from the manifold creates issues. When I looked at R^T @ R and det(R), they seemed close to I and 1, so I assumed they were still "on-manifold." But I guess the 6th or 7th decimal place is decisive?

```python
>>> R.T @ R
array([[ 1.00000330e+00,  1.66519550e-07, -4.07243092e-07],
       [ 1.66519550e-07,  1.00000049e+00,  7.20413748e-08],
       [-4.07243092e-07,  7.20413748e-08,  1.00000268e+00]])
>>> np.linalg.det(R)
1.0000032373629155
```

I see the [SO3::ClosestTo](https://github.com/borglab/gtsam/blob/develop/gtsam/geometry/SO3.cpp#L145) uses Eigen's SVD. Some say there is a closed form solution for eigenvectors of a 3x3 matrix:

- https://dsp.stackexchange.com/questions/1911/analytical-expression-for-the-eigenvectors-of-a-3x3-real-symmetric-matrix
- https://twitter.com/jon_barron/status/1307356783082496000

But it may be numerically ill-behaved?

shteren1, 2022-02-20:
@ProfFan , @dellaert i just stumbled upon a similar issue, i have an incremental vslam optimization that runs hundreds of optimization cycles and uses results from previous iterations as initial state for new key frames i add, long story short i came across a very unpleasant event where the later Pose3 values i had in my problem were VERY far from orthogonality, as far as one of the entries of the diagonal on R*R.T was 43!

a very easy patch that solved the entire issue on my side was normalizing the rotation matrices using quaternions before initializing gtsam Pose3 objects.
I think that enforcing SO3::ClosestTo on the constructor of Rot3 can prevent most of those problems without a very large run time overhead, calling it in every step of optimization is expensive, but if at least you start with orthogonal matrices at the end of an optimization cycle you don't stray too far away from it, perhaps as a last step when an optimization cycle ends its also worth calling it.