borglab/gtsam #912: Bearing range 3D factor in python

xieshuqin, 2021-11-01:
hi,

Thanks for the awesome work!

I wonder if the BearingRangeFactor for Pose3 and Point3 is also ported to Python? I install gtsam via pip and it doesn't have BearingRangeFactor3D.

ProfFan, 2021-11-11:
Hi, if you want to export some function not available in Python, please edit the corresponding `.i` files and add the export yourself :)

xieshuqin, 2021-11-11:
hi Frank,

Thanks for the reply. I have exported the function. I have one more question about the noise model used by BearingRange3D. I wonder what is the correct physical meaning for the 3 variables of BearingRange3D's noise model? Is it XYZ?

dellaert, 2021-11-11:
> hi Frank,
> 
> Thanks for the reply. I have exported the function. I have one more question about the noise model used by BearingRange3D. I wonder what is the correct physical meaning for the 3 variables of BearingRange3D's noise model? Is it XYZ?

Will you create a PR? That would be awesome :-)

I’m not near a computer, but 3-D bearing is two dimensional, and range is one dimensional. So I think the first two are radians and the second is distance.You should look at the code to be sure.

xieshuqin, 2021-11-12:
No problem, I will try to create a PR this weekend.

dellaert, 2021-11-12:
Why don't you close when the PR lands. Can be done automatically by mentioning issue # in PR comment.

ayushsharma-crypto, 2022-05-24:
Hello @xieshuqin  and @ProfFan, I'm also trying to use BearingRangeFactor3D in gtsam python wrapper. Could you please provide the link for the `.i` file that you were referring to?

varunagrawal, 2022-05-25:
@ayushsharma-crypto please look in `sam.i`.

PR #1240 (2022-07-08): Wrap BearingRange Factors
Fixes #912
closed 2022-07-08 by varunagrawal