borglab/gtsam #923: How to deal with the IMU data with gravity ,cause TangentPreintegration.cpp does not consider graviry

zhaoworking, 2021-11-11:
In TangentPreintegration.cpp, updatePreintegrated() function:

"Vector9 preintegratedPlus;
preintegratedPlus << // new preintegrated vector:
theta + w_tangent * dt, // theta
position + velocity * dt + a_nav * dt22, // position
velocity + a_nav * dt; // velocity"

Why not consider gravity?
Forgive me that I used the similar question as others. But I can not find the answer in that issue.So , I am here to open another issue, hope can solve my confusion.
Due to the gravity which is in included in my data , how do I deal with this situation. Should I feed the data into preintegration without any preprocess?

dellaert, 2021-11-11:
Gravity is handled. A quick search in the navigation folder yields 44 results on "gravity" in 12 files. You might want to look at `testImuFactor`, for starters.

closed 2021-11-11 by dellaert

zhaoworking, 2021-11-11:
> gravity

But I did not see any information about gravity handled in `ImuFactorsExample.cpp` . Cause I want to modify this file to adapt to my own dataset