borglab/gtsam #928: make check: ambiguous reference

duembgen, 2021-11-12:
## Description

When compiling the gtsam tests on my Linux machine, I get two errors that _1 is not correctly resolved, for the two tests `testMagFactor.cpp` and `testGPSFactor.cpp`. Extract from error message:
```
In file included from /home/duembgen/Packages/gtsam/CppUnitLite/TestHarness.h:23:0,                                                                                              
                 from /home/duembgen/Packages/gtsam/gtsam/navigation/tests/testMagFactor.cpp:25:                                                                                 
/home/duembgen/Packages/gtsam/gtsam/navigation/tests/testMagFactor.cpp:101:51: error: reference to ‘_1’ is ambiguous                                                             
       (std::bind(&MagFactor3::evaluateError, &f3, _1, dir, bias, none, none, none), s),//                                                                                       
                                                   ^                                                                                                                             
/home/duembgen/Packages/gtsam/CppUnitLite/Test.h:152:9: note: in definition of macro ‘EXPECT’                                                                                    
 { if (!(condition)) \                                                                                                                                                           
         ^~~~~~~~~                                                                                                                                                               
In file included from /home/duembgen/Packages/gtsam/gtsam/3rdparty/Eigen/Eigen/Core:281:0,                                                                                       
                 from /home/duembgen/Packages/gtsam/gtsam/3rdparty/Eigen/Eigen/Dense:1,                                                                                          
                 from /home/duembgen/Packages/gtsam/gtsam/base/OptionalJacobian.h:22,                                                                                            
                 from /home/duembgen/Packages/gtsam/gtsam/base/Matrix.h:27,                                                                                                      
                 from /home/duembgen/Packages/gtsam/gtsam/base/Manifold.h:22,                                                                                                    
                 from /home/duembgen/Packages/gtsam/gtsam/base/GenericValue.h:22,                                                                                                
                 from /home/duembgen/Packages/gtsam/gtsam/nonlinear/Values.h:27,                                                                                                 
                 from /home/duembgen/Packages/gtsam/gtsam/nonlinear/NonlinearFactor.h:23,                                                                                        
                 from /home/duembgen/Packages/gtsam/gtsam/navigation/MagFactor.h:19,                                                                                             
                 from /home/duembgen/Packages/gtsam/gtsam/navigation/tests/testMagFactor.cpp:19:                                                                                 
/usr/include/c++/7/functional:275:34: note: candidates are: const std::_Placeholder<1> std::placeholders::_1                                                                     
     extern const _Placeholder<1> _1;       
```

## Steps to reproduce

1. mkdir build; cd build; cmake ..; make check

## Expected behavior

I guess according to #819 the goal was to use `std::placeholders` everywhere, but there seems to be a namespace confusion in the two mentioned files. Replacing `_1` by `std::placeholders::_1` fixes the problem. 

## Environment

I am using 5.4.0-87-generic, 18.04.1-Ubuntu, gcc version 7.5.0

## Additional information

duembgen, 2021-11-12:
If this issue is specific to my installation (which I guess it might be given that all tests seem to be running in CI) then feel free to ignore, since I fixed it locally. Otherwise, I am happy to open a pull requested with the fixes (essentially 5 extra std::placeholders:: statements)

varunagrawal, 2021-11-15:
Please open the PR. This is a welcome addition to the codebase.

closed 2021-11-15 by ProfFan