borglab/gtsam #935: Values memory allocation does not adhere GTSAM_DEFAULT_ALLOCATOR flag

beetleskin, 2021-11-22:
## Description

The [GTSAM_DEFAULT_ALLOCATOR](https://github.com/borglab/gtsam/blob/8cb22624e0520315f402cf2255857301634a9041/cmake/HandleAllocators.cmake#L25) flag determines what allocators are used in the [FastDefaultAllocator](https://github.com/borglab/gtsam/blob/a18875b598eb454378cf05de3ed4df5278fd87c9/gtsam/base/FastDefaultAllocator.h#L51). This again is used in FastList, FastSet and FastMap. It was introduced in https://github.com/borglab/gtsam/commit/e3ea39f182a5c2a07c516e2cf97943046ea8b89a.

However, [Values](https://github.com/borglab/gtsam/blob/dc8b5e58ff2b8b48ba5b5fc329271daf3b8c2468/gtsam/nonlinear/Values.h#L75) still uses the _fast_pool_allocator_ for its internal _boost:ptr_map_. I think this also should adhere the allocator-flag.

**Background**: We set up a 120-core machine doing gtsam-landmark-SLAM in parallel threads (different threads, disjoint data i.e. individual graphs and optimizations) within the same process, and noticed that all the threads were busy with kernel-tasks. We tracked it down to the _fast_pool_allocator_  in _Values_, which made the thread spend most of the time in memory-management.

## Steps to reproduce

1. Set any valid value to _GTSAM_DEFAULT_ALLOCATOR_
2. Observe that the allocator does not change for Values

## Steps to reproduce our specific situation (no gtsam needed)

Here is a minimal running example that exposes the memory-management issue with the _fast_pool_allocator_:
```c++
#include <future>
#include <thread>
#include <boost/ptr_container/ptr_map.hpp>
#include <boost/pool/pool_alloc.hpp>
#include <tbb/tbb_allocator.h>
#include <vector>

using namespace std;

struct Test {
    int a, b, c;
    double d;
};

using TestValue = Test;

typedef boost::ptr_map<
        int,
        TestValue,
        std::less<>,
        boost::heap_clone_allocator,
        //tbb::tbb_allocator<std::pair<const int, void*> > > TestMap;
        //std::allocator<std::pair<const int, void *> > > TestMap;
        boost::fast_pool_allocator<std::pair<const int, void*> > > TestMap;
        //boost::pool_allocator<std::pair<const int, void*> > > TestMap;

int worker() {
    TestMap values;
    int ops = 100000;
    while (true) {
        //std::cout << std::this_thread::get_id() << ": insert..." << std::endl;
        for (int i = 0; i < ops; ++i) {
            values.insert(i, new Test());
        }
        //std::cout << std::this_thread::get_id() << ": delete again..." << std::endl;
        for (int i = 0; i < ops; ++i) {
            values.erase(i);
        }
    }
    return 123;
}

int main(int argc, char **argv) {
    int threads = 180;
    std::vector<std::future<int>> futures;
    for (int i = 0; i < threads; ++i) {
        auto fut = std::async(std::launch::async, worker);
        futures.push_back(std::move(fut));
    }

    // right now we abort with Ctrl+C so it doesn't matter how we wait
    futures.front().get();
    return 0;
}
```
* Compile and run it: `g++ PoolAllocatorTest.cpp /usr/lib/x86_64-linux-gnu/libtbb.so -lpthread -o pool_allocator_example` 
* Observe (with e.g. htop) red bars for your threads allover the place

## Expected behavior

Values should adhere the compiler allocator-flag.

## Environment

Ubuntu 20.04 (same for 18.04), Dell Precision 5540 but also big-ass cloud worker (120 logical cores).

## Additional information

I'm not the expert with allocators and memory arenas and their details; I just tracked our issue down to the _fast_pool_allocator_ and found the compiler-flag. So maybe there are reasons that I'm not aware of to not use the user-specified allocator, but the _fast_pool_allocator_ at this point?

Also, I did not try all tests with all allocator-compiler flags, and neither did timing-measurements as I was hoping your CI does that ..?

beetleskin, 2021-11-22:
I do have a fix for this, but I can't push branches. Shall I fork and PR or how do you do it? I can't find a how-to-contribute thingy.

ProfFan, 2021-11-22:
We were planning to retire the boost allocator thingy (they are really legacy of the 2010s). Thanks for the catch!

PR #936 (2021-11-22): Use FastDefaultAllocator for Values to adhere allocation compile flags
Fixes #935
closed 2021-11-22 by varunagrawal