borglab/gtsam #937: Retire Boost in light of Default C++17 Support in Ubuntu 18.04+

ProfFan, 2021-11-22:
There is no need to use boost anymore because C++17 is supported on all platforms we currently support.

I have a branch (https://github.com/ProfFan/gtsam/pull/8) that I put together in my weekends, which uses cereal (https://uscilab.github.io/cereal/index.html) for serialization and ripped all boost components except `ptr_map` (see also #935 #936) and some leftovers in `discrete`.

All unit tests pass locally including the serialization tests.

This thread is for open discussion on the direction we should move on.

asa, 2021-11-22:
Amazing! Yes!

varunagrawal, 2021-11-23:
We can use `std::optional` rather than pointers. Make the code safer.

ProfFan, 2021-11-23:
The issue is optionals should never take references. In C++17 this is explicitly disallowed.

jlblancoc, 2021-11-24:
Awesome move, I would love to see C++17 finally here, Yay!

On optional references, they can be used, indeed: check this and search in
the project for some uses for syntax examples and such:

https://github.com/MRPT/mrpt/blob/develop/libs/core/include/mrpt/core/optional_ref.h

Feel free of copying that type definition and play with it as a direct
replacement of boost's optional.

dellaert, 2021-11-25:
This is very cool. Probably should leave this until after the journal submission deadline :-) two questions:
– you say all platforms, but is C++ 17 supported on all popular embedded platforms?
– We should definitely do a 4.1 release before switching.

ProfFan, 2021-12-19:
> This is very cool. Probably should leave this until after the journal submission deadline :-) two questions:
> – you say all platforms, but is C++ 17 supported on all popular embedded platforms?

Yes

> – We should definitely do a 4.1 release before switching.

And yes. Since quals is done I'll start the release.

varunagrawal, 2021-12-21:
We could replace `boost::optional` with `std::optional`. That should make porting far easier than switching to pointers.

ProfFan, 2021-12-21:
std::optional + std::reference_wrapper has some issues - I tried but it broke something that I have not identified. Since I am free of quals now I can start investigating. @varunagrawal do you want to pair program on this?

dellaert, 2021-12-31:
I have been thinking about this. While there can be some (mild) API changes in 4.2, 4.3..., I see removing boost and moving to c++17 is more of "5.0" move. Also, in terms of reviewing this and assessing consequences on correctness, performance, it would have to be executed in a series of small, targeted PRs that tackle different issues, and are easy to review. Happy to chat more in person on creating a time line/roadmap for this.

varunagrawal, 2023-03-10:
@ProfFan I am closing this since it is superseded by #1414 and we've already moved to C++17. But this was an awesome RFC 😄

closed 2023-03-10 by varunagrawal