borglab/gtsam #939: Indeterminant linear system detected while working near variable 1

qianlong2059, 2021-11-24:
what I did was to execute the quick test in tagslam_root repository
and the link is here https://github.com/berndpfrommer/tagslam_root
but I encountered a problem,the error is

[ERROR] [1637719401.554352107]: /tagslam:
Indeterminant linear system detected while working near variable
1 (Symbol: 1).

Thrown when a linear system is ill-posed. The most common cause for this
error is having underconstrained variables. Mathematically, the system is
underdetermined. See the GTSAM Doxygen documentation at
http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
more information.
[tagslam-2] process has died [pid 3241, exit code 255, cmd /home/turtlebot3/tagslam_root/devel/lib/tagslam/tagslam_node __name:=tagslam __log:=/home/turtlebot3/.ros/log/acac2748-4cca-11ec-a093-b827eb751c4a/tagslam-2.log].
log file: /home/turtlebot3/.ros/log/acac2748-4cca-11ec-a093-b827eb751c4a/tagslam-2*.log

does anybody encounter the same problem with me and how I can solve it?

ProfFan, 2021-11-26:
This is not a GTSAM issue, please first ask in the tagslam repo.

Next time, please use the mailing list for questions: https://groups.google.com/g/gtsam-users

closed 2021-11-26 by ProfFan

ghm0819, 2021-12-09:
As far as I know, there are two reasons for such a problem

1.1 compilation option

cmake .. -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF 

1.2 parameter option

Matrix decomposition is by default using LLT decomposition, which is fast but sometimes numerically unstable. Use QR decomposition to resolve this issue. QR is slow but stable to use

parameters.factorization = gtsam::ISAM2Params::QR;

whw-create, 2023-03-12:
> 

it seems not work