borglab/gtsam #940: Eigen compilation issue

EnderLord0042, 2021-11-25:
## Description

When compiling on Ubuntu 21.10 after installing dependencies through apt, the project fails to compile
(I have attached the output of `make check` here: [gtsamMakeCheckLog.txt](https://github.com/borglab/gtsam/files/7600584/gtsamMakeCheckLog.txt))

## Steps to reproduce

1. Install dependencies via apt on Ubuntu 21.10 
2. Attempt to build

## Expected behavior

The library is expected to compile

## Environment

I am using Ubuntu 21.10, gcc 11.2.0, cmake 3.18.4, and boost 1.74

## Additional information

This issue has already been mentioned in this pull request, but the patch mentioned is no longer available
https://github.com/borglab/gtsam/pull/391#issuecomment-663814121

jlblancoc, 2021-11-25:
Could you show the full command line of the failing file? i.e. the last big fat line when doing `VERBOSE=1 make -j1`

EnderLord0042, 2021-11-25:
```
cd /home/quincy/Documents/LinuxVR/gtsam/build/gtsam && /usr/bin/c++ -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SERIALIZATION_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBOOST_TIMER_DYN_LINK -DNDEBUG -Dgtsam_EXPORTS -I/home/quincy/Documents/LinuxVR/gtsam -I/home/quincy/Documents/LinuxVR/gtsam/build -I/home/quincy/Documents/LinuxVR/gtsam/CppUnitLite -I/home/quincy/Documents/LinuxVR/gtsam/gtsam/3rdparty/Eigen -I/home/quincy/Documents/LinuxVR/gtsam/gtsam/3rdparty/metis/include -I/home/quincy/Documents/LinuxVR/gtsam/gtsam/3rdparty/metis/libmetis -I/home/quincy/Documents/LinuxVR/gtsam/gtsam/3rdparty/metis/GKlib -isystem /home/quincy/Documents/LinuxVR/gtsam/gtsam/3rdparty/SuiteSparse_config -isystem /home/quincy/Documents/LinuxVR/gtsam/gtsam/3rdparty/Spectra -isystem /home/quincy/Documents/LinuxVR/gtsam/gtsam/3rdparty/CCOLAMD/Include -O3 -DNDEBUG -fPIC -march=native -Wall -fPIC -Wreturn-local-addr -Werror=return-local-addr -Wreturn-type -Werror=return-type -Wformat -Werror=format-security -Wsuggest-override -O3 -Wno-unused-local-typedefs -o CMakeFiles/gtsam.dir/navigation/CombinedImuFactor.cpp.o -c /home/quincy/Documents/LinuxVR/gtsam/gtsam/navigation/CombinedImuFactor.cpp
```

EnderLord0042, 2021-11-25:
The comment on the pull request I mentioned earlier (this: https://github.com/borglab/gtsam/pull/391#issuecomment-663814121) said that this issue was only with c++17, so I ran the compile command I commented earlier with an added `-std=c++11` to compile only that file in c++11, and because make checks for object files before attempting to compile, I managed to get the library to build.

jlblancoc, 2021-11-28:
Ah! It must be a new issue with g++11.2
Perhaps g++11 should be added to the CI build matrix?

From [here](https://www.phoronix.com/scan.php?page=news_item&px=GCC-11-Cpp-17-Default#:~:text=GCC%2011%20Now%20Defaults%20To%20C%2B%2B17%20Dialect%20By%20Default%20%2D%20Phoronix&text=Following%20the%20proposal%20at%20the,has%20been%20C%2B%2B14.):

> GCC 11 Now Defaults To C++17 Dialect By Default 

So this is gonna become a real issue for most users if they get the latest g++ but not the latest Eigen3...
An ugly workaround might be a special build flag for that file (via cmake source file properties), or wait for the move to c++17.

varunagrawal, 2021-11-29:
Oh I encountered this exact issue and have proposed a fix to it here #944
This is indeed an issue with GCC 8+.

ProfFan, 2021-11-29:
There is a catch here: most people will use system Eigen which is not 3.3.9. I would suggest we do some kind of work-around in *our* code instead.

varunagrawal, 2021-11-29:
The workaround isn't something we can do trivially. We can use CMake to force 3.3.9 and this shouldn't be a major issue since it is a patch version update.

ProfFan, 2021-11-29:
@varunagrawal it is easy to do:  https://github.com/humanoid-path-planner/hpp-fcl/pull/226/files

varunagrawal, 2021-11-29:
Cool! I still think we should update eigen since there are other big fixes that we can benefit from.

Asking users to update eigen is a small task for the rewards.

dellaert, 2021-11-29:
@varunagrawal upgrading system eigen is not trivial so I vote to work-around in our code. Please update PR #944 with other reasons to upgrade. Please do not merge until serialization is validated with old Eigen version.

varunagrawal, 2021-11-29:
Okay, I am updating #944 with the workaround that Fan linked.

varunagrawal, 2021-11-29:
Huh @ProfFan that fix doesn't work. I am getting the error
```
In file included from /usr/local/include/gtsam/base/serializationTestHelpers.h:26,
                 from /home/varun/borglab/GTDynamics/tests/testHelicalJoint.cpp:17:
/usr/local/include/gtsam/base/serialization.h:49:19: error: ‘traits’ is not a class template
   49 | template<> struct traits<boost::serialization::U> {enum {Flags=0};};
      |                   ^~~~~~
/usr/local/include/gtsam/base/serialization.h:49:51: error: explicit specialization of non-template ‘Eigen::internal::traits’
   49 | template<> struct traits<boost::serialization::U> {enum {Flags=0};};
      |                                                   ^
```

varunagrawal, 2021-11-29:
Also, maintainers of Eigen also call it an "ugly hack" https://gitlab.com/libeigen/eigen/-/issues/1676#note_709679901

varunagrawal, 2021-11-29:
Let me take a closer look at this.

varunagrawal, 2021-11-29:
Figured it out. We need to include `Eigen/Core` as well. This was mentioned in the [original SO answer](https://stackoverflow.com/a/54535484/1236990) and not in the PR.

closed 2021-11-30 by varunagrawal