borglab/gtsam #948: Bug in BatchFixLagSmoother "enforceConsistency_" not working

DeepRobot2020, 2021-12-02:
I am working on some monocular SFM with a sliding window and notice that the 
_enforceConsistency__ flag seems not doing anything and _linearKeys__ are never assigned with marginalized values.

This can be easily reproduced by using "FixedLagSmootherExample.cpp" by adding some assertions. In the example, one key and their factors will get marginalized out, and to make the system consistent (In my case, I want the prior factor to get carried forward), we need to fix the linearization point for the variables related to marginalized ones. I think the fix is not hard, and I am happy to put up a PR but would like to get confirmation that this is an issue. 

```cpp
          if (enforceConsistency_ && (linearKeys_.size() > 0)) {
            assert(0); // never triggered when enforceConsistency_ is on even when we doing marginalization 
            theta_.update(linearKeys_);
            for(const auto key_value: linearKeys_) {
              delta_.at(key_value.key) = newDelta.at(key_value.key);
            }
          }
```

## Description

## Steps to reproduce

1. 
2.

## Expected behavior

## Environment

## Additional information

ProfFan, 2021-12-02:
@dellaert

varunagrawal, 2021-12-06:
`BatchFixLagSmoother` comes under `gtsam_unstable` so this should come as no surprise that the code is super buggy.

dellaert, 2021-12-06:
- Unstable does not mean super buggy. All of the code is unit tested and reviewed. But code that is not frequently used is expected to have more bugs.
– I have no time to look into this. But I’ll gladly review a PR to fix it