borglab/gtsam #950: Possible Wrapper Issue with MATLAB Toolbox

mattking-smith, 2021-12-04:
## Description

I recently installed a fresh 20.04 Ubuntu OS on a new computer to test GTSAM + MATLAB toolbox. After following available documentation it appeared that I successfully wrapped GTSAM into the MATLAB environment, noting that the MATLAB toolbox folder was created in the correct file path destination and there was a MATLAB MEX file in the appropriate location. However, I am unable to run any GTSAM code in MATLAB (including the basic unit tests) due to the wrapper being unable to find the functions. The errors I am currently experiencing are not listed under the MATLAB wrapper documentation. 

## Steps to reproduce

1. Build GTSAM from source, clone or download the latest release from the [GTSAM Github repo](https://github.com/borglab/gtsam). Then follow the build & install instructions posted on [GTSAM build page](https://gtsam.org/build/). 
2. To flag the MATLAB toolbox, before the build process, in the terminal, run the command 
> `cmake -DGTSAM_TOOLBOX_INSTALL_PATH:PATH= $HOME/toolbox ..` 

A successfully flagged MATLAB toolbox enabled cmake appears as:

[image]
Noting that MATLAB root and MEX binary paths have been located.
3. Run the command `make install`
4. Follow the step given the [MATLAB Wrapper documentation](https://github.com/borglab/gtsam/tree/develop/matlab).
5. In the MATLAB environment attempt the following:
> `cd /Users/yourname/toolbox  % Change to wherever you installed the toolbox`
> `cd gtsam_examples           % Change to the examples directory`
> `gtsamExamples               % Run the GTSAM examples GUI`

## Expected behavior

Expected to see the following messages in the command window of the MATLAB environment:

```
Starting: testJacobianFactor
Starting: testKalmanFilter
Starting: testLocalizationExample
Starting: testOdometryExample
Starting: testPlanarSLAMExample
Starting: testPose2SLAMExample
Starting: testPose3SLAMExample
Starting: testSFMExample
Starting: testStereoVOExample
Starting: testVisualISAMExample
Tests complete!
```
## Observed behavior
The output of the MATLAB command window:
```
>> testCal3Unified
Unrecognized function or variable 'Cal3Unified'.
```

## Environment

MATLAB version: 2021a
Linux OS: Ubuntu 20.04
Standard GTSAM + MATLAB Toolbox build.

## Additional information

ProfFan, 2021-12-05:
@varunagrawal Could you take a look at this?

mattking-smith, 2021-12-05:
Added high-priority tag because I am trying to make a write-up/documentation for corporate sponsors on a sponsor-shared GitHub repository. I need to have reproducible steps for company to follow on their own computers.

dellaert, 2021-12-05:
> @varunagrawal Could you take a look at this?

Why Varun? Who is using the Matlab toolbox regularly? That will be better than distracting Varun at this time.

@mattking-smith , do you have no ideas on how to fix this? You are now our Matlab go-to guy :-)

mattking-smith, 2021-12-06:
> > @varunagrawal Could you take a look at this?
> 
> Why Varun? Who is using the Matlab toolbox regularly? That will be better than distracting Varun at this time.
> 
> @mattking-smith , do you have no ideas on how to fix this? You are now our Matlab go-to guy :-)

I do not have immediate fixes to the issue, as I have not looked deeply enough into the issue at this point. My suspicion is the way the MEX binding is being generated via the wrapper, but I have not looked or worked on the wrapper code so far. 

I do not have much experience with using Python to wrap libraries into MATLAB, (as is done currently in the [wrap library](https://github.com/borglab/wrap))  as I've only integrated C and C++ code into MATLAB and Simulink via traditional MEXing and writing S-functions. I can turn my attention to fixing this issue, but it will take me some time to catch up to speed on the wrap library and python interface for GTSAM to MATLAB. 

However, I will continue to probe the matter to try and figure out why the MEX file cannot find these function definitions.

mikesheffler, 2021-12-06:
This (my question) is a stupid question, but is it possible we're just looking at a MATLAB path issue,?

To me, 
```
>> testCal3Unified
Unrecognized function or variable 'Cal3Unified'
```
looks like a generic message from MATLAB saying "I don't know where / what that is" and would be exactly the same as
```
>> RandomFunctionNotKnownToMatlab
```
I haven't built the MATLAB wrapper in quite a while, but when I _have_ built it in the past, and there was an issue _at_ MATLAB (i.e., not when building the wrapper), it would look like a linkage issue, and I would see something like:
```
>> gtsam.Values
Invalid MEX-file '/usr/local/gtsam_toolbox/gtsam_wrapper.mexa64' : libboost_serialization.so.1.65.1: cannot open shared object file: No such file or directory

Error in gtsam.Values (line 90)
         my_ptr = gtsam_wrapper(1403);
```
That is just an old error message I happened to have saved somewhere, but it was typical of a problem with the wrapper: MATLAB would complain about the MEX file and maaaaybe offer a hint at where the issue originates (in my personal experience, it's always a problem with serialization / deserialization, because boost serialization exists to cause me suffering).

I haven't built the wrap in forever, though, so I don't know if this is relevant 😬

mattking-smith, 2021-12-06:
> I haven't built the wrap in forever

Thankfully I have compiled and integrated GTSAM + MATLAB toolbox as early as last April on another computer. Perhaps the wrapper library has had a major update in that time?

> looks like a generic message from MATLAB saying "I don't know where / what that is" and would be exactly the same as...

This comment got me thinking that I should compare my currently built and working GTSAM + MATLAB computer with the computer that is not currently working. 

When looking, I did find a major dependency and the source of the error being generated on my end. 

On the computer that is not working with GTSAM + MATLAB when running the `make install`, the .m files defining the class definitions of GTSAM functions via the wrapper are not generated in the +gtsam folder, such as the Cal3Unified class (which is why MATLAB cannot find the function definition). 

I am unsure currently why the standard GTSAM + MATLAB toolbox instructions would not construct and define these class definitions as it has done in the past, but I would again think this would hint at possible wrapping issues during the build process.

mattking-smith, 2021-12-06:
> the .m files defining the class definitions of GTSAM functions via the wrapper are not generated in the +gtsam folder, such as the Cal3Unified class (which is why MATLAB cannot find the function definition).

To follow up on this problem, I zipped up and sent the +gtsam folder from the working GTSAM + MATLAB computer to the computer which lacked all of the proper class defining .m files. Unzipping the .m files into this +gtsam folder resulting in MATLAB being able to find the GTSAM + Wrapper function calls, but I am finding this error now:  
```
>> testCal3Unified
One or more output arguments not assigned during call to "gtsam_wrapper".

Error in gtsam.Cal3Unified (line 40)
        [ my_ptr, base_ptr ] = gtsam_wrapper(526);

Error in testCal3Unified (line 4)
K = Cal3Unified;
```
The `One or more output arguments not assigned during call to "gtsam_wrapper".` is a persistent error across all test cases. This error may further support the claim that the wrapper library is not binding GTSAM into MATLAB properly.

varunagrawal, 2021-12-06:
@mattking-smith PR #953 should address this. Can you please take a look at it?

closed 2021-12-07 by dellaert