borglab/gtsam #952: derivate in Point3::normalize()

TobiasLangner-TomTom, 2021-12-05:
https://github.com/borglab/gtsam/blob/c7781961e66b79cfe17e4084b2ea15b94a2d2a95/gtsam/geometry/Point3.cpp#L58

Shouldn't this line be : 
*H /= pow(x2 + y2 + z2, **0.5**);

https://math.stackexchange.com/a/2161888

varunagrawal, 2021-12-05:
Nope, 1.5 is correct. I guess you're missing that s = v / λ so the extra λ terms come into the denominator in the final expression.

I recommend deriving this yourself on pen and paper and you'll see for yourself. If you still think this is wrong, we would really appreciate a unit test demonstrating it is incorrect. I believe we already have a unit test for checking the correctness of this Jacobian. :slightly_smiling_face:

closed 2021-12-05 by varunagrawal

TobiasLangner-TomTom, 2021-12-05:
My bad. The derivation I have linked defines H in terms of the normalized point, while the GTSAM implementation uses the unnormalized point.

varunagrawal, 2021-12-05:
No worries. This is non-trivial stuff.