borglab/gtsam #954: MATLAB Wrapper Does Not Support Default Arguments

mattking-smith, 2021-12-07:
It appears that the wrapper definitions generated when binding GTSAM to MATLAB need additional arguments then the default arguments provided for the MATLAB examples.

I am seeing that errors when testing all the GTSAM examples in the gtsam_examples folder. 

For example when looking at StereoVOExample.m when trying to execute 
`stereo_model = noiseModel.Isotropic.Sigma(3,1);`
as written when compiled throws the error:
```
>> stereo_model = noiseModel.Isotropic.Sigma(3,1);
Error using gtsam.noiseModel.Isotropic.Sigma (line 76)
Arguments do not match any overload of function
Isotropic.Sigma
```
 However looking at the wrapper definition under Isotropic.m states
`%Sigma(size_t dim, double sigma, bool smart) : returns gtsam::noiseModel::Isotropic`
where the Isotropic wrapper definition in MATLAB is:

```
    function varargout = Sigma(varargin)
      % SIGMA usage: Sigma(size_t dim, double sigma, bool smart) : returns gtsam.noiseModel.Isotropic
      % Doxygen can be found at https://gtsam.org/doxygen/
      if length(varargin) == 3 && isa(varargin{1},'numeric') && isa(varargin{2},'double') && isa(varargin{3},'logical')
        varargout{1} = gtsam_wrapper(1092, varargin{:});
        return
      end

      error('Arguments do not match any overload of function Isotropic.Sigma');
    end
```
Hence executing 
`>> stereo_model = noiseModel.Isotropic.Sigma(3,1,true);`
is handled without error and GTSAM can properly optimize the StereoVOExample.m file:
```
Optimizing
Elapsed time is 0.014737 seconds.
```
It appear that some wrapper definitions have changed and the baseline example codes in MATLAB need to updated to reflect this.

_Originally posted by @mattking-smith in https://github.com/borglab/gtsam/pull/953#pullrequestreview-825608801_

dellaert, 2021-12-07:
This is the default argument issue: I think previously there were *wt* versions of the Sigma function: one with 2 arguments and one with 3. This was then replaced with one, with a default argument, but that (silently) broke the MATLAB examples. A quick fix, @mattking-smith , if you're up for it, is to push a quick PR with the "true" argument added in all examples that fail. Then we can take a bit more time fixing the wrapper code - and as I alluded to, it might be beneficial for you to be involved in that more permanent fix.

mattking-smith, 2021-12-07:
> This is the default argument issue: I think previously there were _wt_ versions of the Sigma function: one with 2 arguments and one with 3. This was then replaced with one, with a default argument, [...]

Yeah I can do this to give a quick fix a go.

mattking-smith, 2021-12-08:
> Yeah I can do this to give a quick fix a go.

As I am working on this I have been able to fix simple bugs, like adding `true` to `>> stereo_model = noiseModel.Isotropic.Sigma(3,1,true);`, but I am struggling to fix the `graph.print()` feature of a MATLAB-based graph, which is called in many examples. 

For example in the LocalizationExample.m file, the command 
`graph.print(sprintf('\nFactor graph:\n'));`  results in the error 
```
Error using gtsam.NonlinearFactorGraph/print (line 406)
Arguments do not match any overload of function gtsam.NonlinearFactorGraph.print
```
The print function is defined in the NonlinearFactorGraph.m file as
```
    function varargout = print(this, varargin)
      % PRINT usage: print(string s, KeyFormatter keyFormatter) : returns void
      % Doxygen can be found at https://gtsam.org/doxygen/
      if length(varargin) == 2 && isa(varargin{1},'char') && isa(varargin{2},'gtsam.KeyFormatter')
        gtsam_wrapper(1575, this, varargin{:});
        return
      end
      error('Arguments do not match any overload of function gtsam.NonlinearFactorGraph.print');
    end
```
The new additional argument here is the  `'gtsam.KeyFormatter'`. I am struggling to find an example on how to create a KeyFormatter for this function in the MATLAB environment. @gchenfc do you have any thoughts on how I could generate the KeyFormatter in MATLAB for the print command?

dellaert, 2021-12-08:
If it is limited to one or two classes, you can always add the two versions of the function back into the wrapper, ie One with the KeyFormatter and one without. Then the default argument will be selected in C++

mattking-smith, 2021-12-13:
@varunagrawal, I believe once the `master` version of the wrapper library is merged into this repository we can close this issue.

varunagrawal, 2021-12-13:
Yeah I'm already on that :)

closed 2021-12-14 by varunagrawal