borglab/gtsam #961: Range only expression factor

vifi2021, 2021-12-11:
Hi,

I am a beginner of GTSAM and I am following one of the example (Pose3SLAMExampleExpressions_BearingRangeWithTransform.cpp) to come up with a way to construct an expression graph based on range only expression factors.

following the example code:
```C++
auto prediction_ = Expression<BearingRange3D>( BearingRange3D::Measure, Pose3_('x',i)*body_T_sensor_, Point3_('l',j));
auto measurement = BearingRange3D(world_T_sensor.bearing(points[j]), world_T_sensor.range(points[j]));
graph.addExpressionFactor(prediction_, measurement, bearingRangeNoise);
```

I am trying to define a range expression as the following:
```C++
auto prediction_r_ = Expression<double>( BearingRange3D::MeasureRange, Pose3_('x',i)*body_T_sensor_, Point3_('l',j));
auto measurement = BearingRange3D(world_T_sensor.bearing(points[j]), world_T_sensor.range(points[j]));
double measurement_r = measurement.range();
graph.addExpressionFactor(prediction_r_, measurement_r, RangeNoise);
```

However my first line gives me compiler error.
```
hans@hans-XPS-15-9560:~/Documents/RAN_SLAM/learning_gtsam/build$ make Pose3SLAMExampleExpressions_BearingRangeWithTransform
Scanning dependencies of target Pose3SLAMExampleExpressions_BearingRangeWithTransform
[ 50%] Building CXX object CMakeFiles/Pose3SLAMExampleExpressions_BearingRangeWithTransform.dir/Pose3SLAMExampleExpressions_BearingRangeWithTransform.cpp.o
In file included from /usr/include/boost/function/detail/maybe_include.hpp:33:0,
                 from /usr/include/boost/function/detail/function_iterate.hpp:14,
                 from /usr/include/boost/preprocessor/iteration/detail/iter/forward1.hpp:67,
                 from /usr/include/boost/function.hpp:64,
                 from /usr/local/include/gtsam/inference/Key.h:28,
                 from /usr/local/include/gtsam/inference/Symbol.h:21,
                 from /home/hans/Documents/RAN_SLAM/learning_gtsam/Pose3SLAMExampleExpressions_BearingRangeWithTransform.cpp:8:
/usr/include/boost/function/function_template.hpp: In instantiation of ‘static R boost::detail::function::function_invoker4<FunctionPtr, R, T0, T1, T2, T3>::invoke(boost::detail::function::function_buffer&, T0, T1, T2, T3) [with FunctionPtr = double (*)(const gtsam::Pose3&, const Eigen::Matrix<double, 3, 1>&); R = double; T0 = const gtsam::Pose3&; T1 = const Eigen::Matrix<double, 3, 1>&; T2 = gtsam::OptionalJacobian<1, 6>; T3 = gtsam::OptionalJacobian<1, 3>]’:
/usr/include/boost/function/function_template.hpp:925:38:   required from ‘void boost::function4<R, T1, T2, T3, T4>::assign_to(Functor) [with Functor = double (*)(const gtsam::Pose3&, const Eigen::Matrix<double, 3, 1>&); R = double; T0 = const gtsam::Pose3&; T1 = const Eigen::Matrix<double, 3, 1>&; T2 = gtsam::OptionalJacobian<1, 6>; T3 = gtsam::OptionalJacobian<1, 3>]’
/usr/include/boost/function/function_template.hpp:716:7:   required from ‘boost::function4<R, T1, T2, T3, T4>::function4(Functor, typename boost::enable_if_c<(! boost::is_integral<Functor>::value), int>::type) [with Functor = double (*)(const gtsam::Pose3&, const Eigen::Matrix<double, 3, 1>&); R = double; T0 = const gtsam::Pose3&; T1 = const Eigen::Matrix<double, 3, 1>&; T2 = gtsam::OptionalJacobian<1, 6>; T3 = gtsam::OptionalJacobian<1, 3>; typename boost::enable_if_c<(! boost::is_integral<Functor>::value), int>::type = int]’
/usr/include/boost/function/function_template.hpp:1061:16:   required from ‘boost::function<R(T0, T1, T2, T3)>::function(Functor, typename boost::enable_if_c<(! boost::is_integral<Functor>::value), int>::type) [with Functor = double (*)(const gtsam::Pose3&, const Eigen::Matrix<double, 3, 1>&); R = double; T0 = const gtsam::Pose3&; T1 = const Eigen::Matrix<double, 3, 1>&; T2 = gtsam::OptionalJacobian<1, 6>; T3 = gtsam::OptionalJacobian<1, 3>; typename boost::enable_if_c<(! boost::is_integral<Functor>::value), int>::type = int]’
/home/hans/Documents/RAN_SLAM/learning_gtsam/Pose3SLAMExampleExpressions_BearingRangeWithTransform.cpp:77:122:   required from here
/usr/include/boost/function/function_template.hpp:101:19: error: too many arguments to function
           return f(BOOST_FUNCTION_ARGS);
                  ~^~~~~~~~~~~~~~~~~~~~~

``` 
I have been scratching my head for several days but still can't figure out how to solve this. Could anyone help me with this?

dellaert, 2021-12-11:
`MeasureRange` does not have the requisite OptionalJacobians. RangeFactor.h probably does what you want, but if you want to use expressions, you can use the `Range` template for the types you’re interested in. Range fractured or age does that, so you can crib from that example. Probably as easy as replacing  MeasureRange with Range<Pose3,Point3>.

closed 2021-12-11 by dellaert

vifi2021, 2021-12-11:
Thank you very much!
```C++
auto prediction_r_ = Expression<double>( Range<Pose3, Point3>(), Pose3_('x',i)*body_T_sensor_, Point3_('l',j));
```
This works for me.