borglab/gtsam #964: Indeterminant linear system detected while working near variable

zhaoworking, 2021-12-13:
```
################ NEW FACTORS AT TIME 1626318282.750650 ################
NonlinearFactorGraph: size: 3

Factor 0: ImuFactor(x88,v88,x89,v89,b88)
preintegrated measurements:
    deltaTij 0.05
    deltaRij.ypr = (-4.58681042e-05 -0.000265717352  0.000372633753)
    deltaPij [-0.000490940523, 0.000735574774, 0.0120388693]'
    deltaVij [-0.0191920688, 0.0299153734, 0.482425743]'
    gyrobias [0, 0, 0]'
    acc_bias [0, 0, 0]'

    preintMeasCov 
[ 1.53125001e-09   1.2509485e-17  2.15427521e-18 -1.65205415e-16 -5.26200298e-12  3.35800697e-13 -1.04223302e-14 -3.33090037e-10  2.09902469e-11
  1.2509485e-17  1.53125002e-09 -1.53640589e-18  5.26191276e-12 -1.59239783e-16  2.07899034e-13  3.33084412e-10 -1.00285151e-14  1.28838896e-11
 2.15427521e-18 -1.53640589e-18  1.53125003e-09 -3.36785831e-13 -2.07192214e-13 -8.33493979e-17 -2.10526005e-11 -1.28391533e-11 -5.18923917e-15
-1.65205415e-16  5.26191276e-12 -3.36785831e-13   4.1562842e-09  8.47799052e-17  1.34069245e-15  1.25001812e-07  4.46190586e-15  7.02545197e-14
-5.26200298e-12 -1.59239783e-16 -2.07192214e-13  8.47799052e-17  4.15628412e-09  -2.1525147e-15  4.49788672e-15  1.25001807e-07 -1.14005568e-13
 3.35800697e-13  2.07899034e-13 -8.33493979e-17  1.34069245e-15  -2.1525147e-15  4.15625019e-09  7.11711822e-14 -1.14580265e-13   1.2500001e-07
-1.04223302e-14  3.33084412e-10 -2.10526005e-11  1.25001812e-07  4.49788672e-15  7.11711822e-14  5.00010227e-06  2.50441129e-13  3.96094251e-12
-3.33090037e-10 -1.00285151e-14 -1.28391533e-11  4.46190586e-15  1.25001807e-07 -1.14580265e-13  2.50441129e-13  5.00010201e-06  -6.4393506e-12
 2.09902469e-11  1.28838896e-11 -5.18923917e-15  7.02545197e-14 -1.14005568e-13   1.2500001e-07  3.96094251e-12  -6.4393506e-12  5.00000056e-06]
  noise model sigmas: 3.91311897e-05 3.91311898e-05 3.91311899e-05 6.44692501e-05 6.44692494e-05 6.44689863e-05  0.00223609084  0.00223609079   0.0022360681

Factor 1: BetweenFactor(b88,b89)
  measured: acc = [0, 0, 0]' gyro = [0, 0, 0]'
  noise model: diagonal sigmas[0.001; 0.001; 0.001; 1e-05; 1e-05; 1e-05];

Factor 2: BetweenFactor(x88,x89)
  measured: R:
[
	      0.9999987,  0.000816484903,  -0.00139033375;
 	-0.000815426224,     0.999999377,  0.000761853454;
 	  0.00139095493, -0.000760718749,     0.999998743
]
[-0.00238359758, -0.0981021114, 0.00579207297]';
  noise model: diagonal sigmas[2.15791864e-05; 2.15791864e-05; 4.37019372; 1; 1; 1];

time consuming per episode is 0.024776717
current_bias:  
acc = [9.8821951e+90, -1.17842217e+92, 1.21878767e+92]' gyro = [-7.14350686e+88, -2.34964682e+89, 2.48906269e+89]'
   0.999986527 -0.00517254398 0.000435631792   0.0236690054
  0.0051716756    0.999984683  0.00197145619   -0.267376937
-0.00044582256 -0.00196917668    0.999997962  0.00374918896
             0              0              0              1
    0.999986527  -0.00517254398  0.000435631791
   0.0051716756     0.999984683   0.00197145619
-0.000445822562  -0.00196917668     0.999997962
 0.0236690054
 -0.267376937
0.00374918896
cur_state:

|0.389296034, -0.3907936, -0.834103686|
|0.83344345, 0.535012545, 0.138324231|
|0.392199712, -0.749027329, 0.533982628|

[1.17262331e+105, -5.39972378e+104, 1.588715e+103]';

 7.16926278e+105
-7.30289123e+105
 1.31886773e+106
pro_state:

|-0.0999942018, 0.951112991, 0.292207526|
|-0.716640183, 0.134884699, -0.684275504|
|-0.690237646, -0.277831238, 0.6681181|

[1.53108645e+105, -9.05116939e+104, 6.75321015e+104]';
[... 128 lines cut ...]
kill -l [exitstatus]

```

Here I use IMU between Factor and 6DOF visual odometry pose Between factor  to construct my graph using isam2 to update it. But I found this problem after a short time. Obviously, the IMU bias displayed which after optimization is too big. But I don't know what cause the bias increases in such a pace that reaches at **1e91**  magnitude that cause the system just shut down. 
So,What is the root reason of such problem? And for sake of the convenience and intuition , I past the code blow.
```
            new_factors.emplace_shared<BetweenFactor<Pose3>>(
                previous_pose_key,
                current_pose_key,
                Pose3(Rot3(imu_cur_T_pre_.rotation()), imu_cur_T_pre_.translation()),
                noise_model_pose);
```
```

            new_factors.emplace_shared<ImuFactor>(previous_pose_key, previous_vel_key,
                                                  current_pose_key, current_vel_key,
                                                  previous_bias_key, *current_summarized_measurement);
```
```
            new_factors.emplace_shared<BetweenFactor<imuBias::ConstantBias>>(previous_bias_key,
                                                                             current_bias_key,
                                                                             imuBias::ConstantBias(),
                                                                             sigma_between_b);
```

dellaert, 2021-12-13:
Not a bug, please subscribe to Google group and ask there. But even there I think no one will be able to help you based on only this information. Working with IMU factors is hard and people might only help you if they know the steps you have taken to debug etc.

closed 2021-12-13 by dellaert

zhaoworking, 2021-12-13:
> 

Thanks a lot whatsoever. Notable to say, I change the 6DOF factor pose_noise of which the orientation noise is set to INT_MAX ,to the zero. And , it works fine. But I am confused with this noise model, could u tell me what the role that the noise parameter here represent in the noise model. Cause when I set INT_MAX, i found the noise model is [0,0,0], however , when I set zeros, i found it is [inf ,inf, inf] in `factor.print()`
I just wanna know what is the matter with this noise parameter. :-)