borglab/gtsam #981: Incorrect reconstruction results with SmartFactor and Cal3Fisheye

junzhang3141, 2021-12-22:
I am debugging some incorrect 3D recon issues using SmartFactor + Fisheye camera model. 
For some reason all the 3d points near the edge are all wrong and I am now able to reproduce the issue 
with gtsam example`SFMExample_SmartFactor.cpp`. 
All I did is to modify the camera model to be fisheye and I can see the optimized points are quite off compared to the ground truth:

There is the modified code to reproduce the issue:
https://github.com/junzhang3141/gtsam/commit/4a86b2f0406e93dc97df4df1340e919fb92587a3

Here is the output with the fisheye  camera model and the expected values should be 

```
  points.push_back(gtsam::Point3(10.0,10.0,10.0));
  points.push_back(gtsam::Point3(-10.0,10.0,10.0));
  points.push_back(gtsam::Point3(-10.0,-10.0,10.0));
  points.push_back(gtsam::Point3(10.0,-10.0,10.0));
  points.push_back(gtsam::Point3(10.0,10.0,-10.0));
  points.push_back(gtsam::Point3(-10.0,10.0,-10.0));
  points.push_back(gtsam::Point3(-10.0,-10.0,-10.0));
  points.push_back(gtsam::Point3(10.0,-10.0,-10.0));
```

```
Values with 8 values:
Value 0: (Eigen::Matrix<double, 3, 1, 0, 3, 1>)
[
        9.15607959;
        9.17752954;
        9.28096226
]

Value 1: (Eigen::Matrix<double, 3, 1, 0, 3, 1>)
[
        -9.0805853;
        9.22227012;
        9.28433874
]

Value 2: (Eigen::Matrix<double, 3, 1, 0, 3, 1>)
[
        -9.12772385;
        -9.01612418;
        9.28647311
]

Value 3: (Eigen::Matrix<double, 3, 1, 0, 3, 1>)
[
        9.11506908;
        -9.06124787;
        9.28290342
]

Value 4: (Eigen::Matrix<double, 3, 1, 0, 3, 1>)
[
        9.15180669;
        9.17685577;
        -9.26849095
]

Value 5: (Eigen::Matrix<double, 3, 1, 0, 3, 1>)
[
        -9.08505838;
        9.22121502;
        -9.26292606
]

Value 6: (Eigen::Matrix<double, 3, 1, 0, 3, 1>)
[
        -9.13232163;
        -9.01699544;
        -9.26318632
]

Value 7: (Eigen::Matrix<double, 3, 1, 0, 3, 1>)
[
        9.11026593;
        -9.06183102;
        -9.26867751
]

final error: 142.732316
number of iterations: 4
```

## Description

## Steps to reproduce

1. 
2.

## Expected behavior

## Environment

## Additional information

junzhang3141, 2021-12-22:
I have not yet dug into the code yet but my gut feeling is we are not applying distortion to the projected points correctly as I can see more errors if 3d points are near the edges of the image. I will update once I have more info.

varunagrawal, 2023-12-30:
@junzhang3141 closing this issue. Please re-open once you have more info or if this is still an issue.

closed 2023-12-30 by varunagrawal