borglab/gtsam #995: Similarity(3) estimation returns a negative scale for some poorly aligned pose graphs

johnwlambert, 2021-12-30:
For some badly aligned pose graphs, `Similarity3.Align(` returns an object with **negative scale**, e.g. 

```python
Sim(3) estimate: 
R: [
	-0.786299, 0.313693, -0.532288;
	-0.569328, -0.0332175, 0.821439;
	0.239998, 0.948943, 0.204713
]
translation = [786.49152258 275.49551117 122.02794014]
scale =  -0.0009678177616769815
```
This wreaks havoc on reprojection errors, as cheirality exceptions begin to occur everywhere when using camera poses that have been transformed by the Sim(3) object with negative scale.

When the scale is clamped to a positive number, then reprojection errors are unmodified by applying SIm(3) to points and cameras.

According to Ethan Eade (see Section 10.1 of [this doc](https://ethaneade.com/lie_groups.pdf)), Sim(3) scale must be non-negative, so the GTSAM code is currently buggy.

## How can I reproduce it?
Call:
```python
    ab_pairs = Pose3Pairs(list(zip(aTi_list, bTi_list)))
    aSb = Similarity3.Align(ab_pairs)
```
on the following 2 pose graphs (using 17 out of 32 cameras for correspondences):
```python
        poses_gt = [
            Pose3(
                Rot3(
                    [
                        [0.696305769, -0.0106830792, -0.717665705],
                        [0.00546412488, 0.999939148, -0.00958346857],
                        [0.717724415, 0.00275160848, 0.696321772],
                    ]
                ),
                Point3(5.83077801, -0.94815149, 0.397751679),
            ),
            Pose3(
                Rot3(
                    [
                        [0.692272397, -0.00529704529, -0.721616549],
                        [0.00634689669, 0.999979075, -0.00125157022],
                        [0.721608079, -0.0037136016, 0.692291531],
                    ]
                ),
                Point3(5.03853323, -0.97547405, -0.348177392),
            ),
            Pose3(
                Rot3(
                    [
                        [0.945991981, -0.00633548292, -0.324128225],
                        [0.00450436485, 0.999969379, -0.00639931046],
                        [0.324158843, 0.00459370582, 0.945991552],
                    ]
                ),
                Point3(4.13186176, -0.956364218, -0.796029527),
            ),
            Pose3(
                Rot3(
                    [
                        [0.999553623, -0.00346470207, -0.0296740626],
                        [0.00346104216, 0.999993995, -0.00017469881],
                        [0.0296744897, 7.19175654e-05, 0.999559612],
                    ]
                ),
                Point3(3.1113898, -0.928583423, -0.90539337),
            ),
            Pose3(
                Rot3(
                    [
[... 480 lines cut ...]
                    Rot3(
                        [
                            [-0.766833992, -0.537641809, -0.350580824],
                            [-0.389506676, -0.0443270797, 0.919956336],
                            [-0.510147213, 0.84200736, -0.175423563],
                        ]
                    ),
                    Point3(234.185458, 326.007989, -691.769777),
                ),
                calib,
            ),
            30: PinholeCameraCal3Bundler(
                Pose3(
                    Rot3(
                        [
                            [-0.754844165, -0.559278755, -0.342662459],
                            [-0.375790683, -0.0594160018, 0.92479787],
                            [-0.537579435, 0.826847636, -0.165321923],
                        ]
                    ),
                    Point3(-5.93398168, 41.9107972, -3.07385081),
                ),
                calib,
            ),
        }

dellaert, 2021-12-30:
Good find.! Do you have an idea how to fix it?

johnwlambert, 2021-12-30:
@dellaert I propose one potential fix here: https://github.com/borglab/gtsam/pull/997

dellaert, 2021-12-31:
@johnwlambert , when I run `make python-install python-test` I get:
```
======================================================================
FAIL: test_align_via_Sim3_to_poses_skydio32 (test_Sim3.TestSim3)
Ensure scale estimate of Sim(3) object is non-negative.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/dellaert/git/github/build/python/gtsam/tests/test_Sim3.py", line 640, in test_align_via_Sim3_to_poses_skydio32
    assert rSe.scale() >= 0
AssertionError
```

johnwlambert, 2021-12-31:
@dellaert interesting. I'll take another look on my local machine. Was passing that test on all CI configurations yesterday:
```
test_align_poses_along_straight_line (test_Sim3.TestSim3)
Test Align Pose3Pairs method. ... ok
test_align_poses_along_straight_line_gauge (test_Sim3.TestSim3)
Test if Align Pose3Pairs method can account for gauge ambiguity. ... ok
test_align_poses_scaled_squares (test_Sim3.TestSim3)
Test if Align Pose3Pairs method can account for gauge ambiguity. ... ok
test_align_via_Sim3_to_poses_skydio32 (test_Sim3.TestSim3)
Ensure scale estimate of Sim(3) object is non-negative. ... ok
```

johnwlambert, 2022-01-01:
@dellaert I built locally and all the tests pass for me.

I did have to uninstall 4 previous versions of gtsam first, though. Can you `pip uninstall gtsam` until nothing is left, and then pull the latest, and try one more time to `make python-install`? This is what I had to do:
```bash
jlambert:~/gtsam$ cd build
jlambert:~/gtsam/build$ pip uninstall gtsam
Found existing installation: gtsam 4.1.1
Uninstalling gtsam-4.1.1:
  Would remove:
    /home/jlambert/python/lib/python3.8/site-packages/gtsam-4.1.1.dist-info/*
    /home/jlambert/python/lib/python3.8/site-packages/gtsam/*
    /home/jlambert/python/lib/python3.8/site-packages/gtsam_unstable/*
Proceed (y/n)? y
  Successfully uninstalled gtsam-4.1.1
jlambert:~/gtsam/build$ pip uninstall gtsam
Found existing installation: gtsam 4.2a0
Uninstalling gtsam-4.2a0:
  Would remove:
    /home/jlambert/python/lib/python3.8/site-packages/gtsam-4.2a0-py3.8.egg
Proceed (y/n)? y
  Successfully uninstalled gtsam-4.2a0
jlambert:~/gtsam/build$ pip uninstall gtsam
Found existing installation: gtsam 4.1.0
Uninstalling gtsam-4.1.0:
  Would remove:
    /home/jlambert/python/lib/python3.8/site-packages/gtsam-4.1.0-py3.8.egg
Proceed (y/n)? y
  Successfully uninstalled gtsam-4.1.0
jlambert:~/gtsam/build$ pip uninstall gtsam
WARNING: Skipping gtsam as it is not installed.
jlambert:~/gtsam/build$ make python-install
```

Then, when I run
```bash
jlambert: make python-test
----------------------------------------------------------------------
Ran 148 tests in 0.125s

OK (skipped=3)
[100%] Built target python-test
```

dellaert, 2022-01-03:
Thanks for checking!  I get the same now
I think there is an issue when compiling GTSAM with "Timing" mode, in that it links with the wrong shared library. Closing this again now :-)

closed 2022-01-03 by dellaert

dellaert, 2022-01-03:
> I think there is an issue when compiling GTSAM with "Timing" mode, in that it links with the wrong shared library.

@varunagrawal I looked at this, and think this might be a cmake issue with CMAKE_BUILD_TYPE_POSTFIXES. Specifically, when I enable Timing, I think it builds a `gtsamTiming.cpython-39-darwin.so` but it might not link with the correct `libgtsamTiming.4.2.0.dylib`? Not sure, and no idea how to fix.

varunagrawal, 2022-01-03:
That's interesting. I will need to take a look at this with a much fresher mind though but I will get to it asap.