borglab/gtsam #1076: Support direct STL container wrapping in the Pybind11 wrapper

varunagrawal, 2022-01-30:
## Feature

By including `pybind11/stl.h`, we can have Pybind11 directly translate STL container elements (even those that are typedef'd) rather than go the roundabout way of defining a custom opaque type that may collide with another type.

For example, recently #1069 added a `Rot3Vector` to the wrapped code and this shouldn't be necessary. We should just be able to say `std::vector<gtsam::Rot3>` and use a regular list on the python side. This makes the wrapper more powerful and more intuitive.

The downside is that this will considerably change the Python API (using python based data types rather than our custom types) but it is the more beneficial approach in the long term.

## Motivation

The hassle we had with #1069 is some pretty good motivation.

## Pitch

I took the day off yesterday and worked a bit on this to see how much effort it was, and I made surprisingly great progress. I can make a PR as early as today.

dellaert, 2022-01-30:
If it is very much API breaking I want to hold off until next major version (5.x). The book development can't tolerate this amount of disruption at this time.. Also, IROS....

varunagrawal, 2022-01-30:
Yup thought as much. I'll still open a draft PR on my fork so we're not starting from scratch in the future and not wasting CI cycles.

closed 2023-06-16 by varunagrawal