borglab/gtsam #1081: Problem with python wrapper build

fatrybl, 2022-01-31:
Hello everyone, 
i faced a problem with building GTSAM with python wrapper/
What i do:
1) dowloaded repo, mkdir build && cd build
2) cmake -DCMAKE_BUILD_TYPE=Debug -DGTSAM_BUILD_PYTHON=1 -DCMAKE_INSTALL_PREFIX:PATH=$HOME/mark/gtsam_lib/ ..   (tried to install to default usr/local/ , same result)
3) make check -j6 . All tested passed 0 errors out of 265
4) make install -j6.  All installed , no errors , all is fine
5) Now: make python-install 
and get this:
[image]
6) make python-test
and give me this:
[image]

ProfFan, 2022-01-31:
I think we should change the README to advise people to NOT use `python-install`. As demonstrated in #1075 , the existence of the `python-install` appears to be rather confusing to first-time users.

@fatrybl I suggest you to install the python toolbox with:
```
cd <your build folder> # normally <gtsam>/build
pip install --user ./python
```

As for the error you are seeing, please report back your current operating system and GCC version.

dellaert, 2022-01-31:
I love python-install ! What's wrong with it?

dellaert, 2022-01-31:
I think the issue - which we should document properly - is that users that don't have install privileges in their regular python build should build gtsam withn a virtualenv or conda environment.

dellaert, 2022-01-31:
On Mac, for instance, system python is still 2.7. I also think *no one* should install in /usr/local

fatrybl, 2022-02-01:
@ProfFan thank you for a decision. I used pip install --user ./python and python wrapper has been successfully installed to /home/mark/.local/lib/Python3.8/site-packages/ . My GCC is 9.3.0, Linux Mint 20.3 Cinnamon, kernel 5.4.0-97-generic.
@dellaert Could you please clarify how to install with python-install and proper permissions? sudo make python-install did not help. I dont really want to use any environments as conda or others. Seems like there should be an option to install wrapper to $HOME/.local/lib/Python3.8/site-packages/ as with a method above

NickPerezCarletonUniversity, 2022-03-23:
I tried everything in the following thread: https://github.com/borglab/gtsam/issues/714 as well as in this current thread. I am still getting the error listed [here](https://github.com/borglab/gtsam/issues/714) when trying to build.

Does anyone have any tips/tricks/advice? Thanks in advance.

NickPerezCarletonUniversity, 2022-03-23:
@ProfFan @dellaert @fatrybl

varunagrawal, 2022-03-24:
If you're getting a PythonConfig.cmake error, that means you most likely don't have the dev files for your interpreter installed. For a vanilla Ubuntu system, that would be `apt-get install python3-dev`.

closed 2022-03-25 by varunagrawal