borglab/gtsam #1087: Unable to Build 4.1.1 On Windows

magicbycalvin, 2022-02-04:
## Description

I have been trying to compile GTSAM from source on at least 3 different Windows 10 machines but keep running into problems. My goal is to compile the most recent stable version (4.1.1 at the time of writing) in release and generate the MATLAB toolbox alongside it. I am first trying to just get release to successfully compile before playing with the MATLAB build so that I can hopefully isolate problems. Unfortunately, since the compile time takes so long (2-3 hours at least), I can't quickly try a new configuration to see if it works. For now I am getting by with a precompiled GTSAM MATLAB toolbox I found somewhere (can't recall where) which is labeled as version 3.2.0. However, I would like to get it working with the most recent version and especially with the release version instead of debug for speed.

## Steps to reproduce

1. Download Boost from [this page](https://www.boost.org/users/history/version_1_78_0.html), unzip it, run `bootstrap.bat` followed by `b2.exe`
2. Download latest stable release of GTSAM and unzip it.
3. Run the cmake GUI and set the source code to point to the unzipped GTSAM directory (i.e., `Where is the source code: C:/Users/<USERNAME>/Desktop/GTSAM/gtsam-4.1.1`). Create a `build` directory inside the `gtsam-4.1.1` directory and point cmake to that for building (i.e., `Where to build the binaries: C:/Users/<USERNAME>/Desktop/GTSAM/gtsam-4.1.1/build`).
4. The ONLY cmake parameter I added/modified is setting `BOOST_ROOT` to point to the unzipped and precompiled Boost directory (i.e., `Add Entry` -> `Name: BOOST_ROOT`, `Type: PATH`, `Value: C:/Users/<USERNAME>/Desktop/boost_1_78_0`).
5. Click configure. Configuring finishes successfully but does spit out several cmake warnings. The following warning message is what is printed out 10 times (I do also get the warning about TBB but that seems irrelevant to this situation, especially since I can change `GTSAM_WITH_TBB` to `off` to suppress it):
```
CMake Warning at cmake/FindBoost.cmake:1147 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  cmake/FindBoost.cmake:1269 (_Boost_COMPONENT_DEPENDENCIES)
  cmake/FindBoost.cmake:1914 (_Boost_MISSING_DEPENDENCIES)
  cmake/HandleBoost.cmake:28 (find_package)
  CMakeLists.txt:44 (include)
```
6. Click on generate. Generation completes successfully with no messages other than `Generating done`.
7. Click on `Open Project`, next to the button it says `Current Generator: Visual Studio 17 2022`.
8. Within Visual Studio 2022, click on the dropdown to change `debug` to `release`. Then click `Build` -> `Build Solution`.
9. Wait for about an hour and then come back to 461 errors, 102 warnings, and 103 messages.

Below are the first few error codes Visual Studio provided for me. I am omitting the majority of them for space but would be happy to provide more information upon request.

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: class Eigen::Matrix<double,2,1,0,2,1> __cdecl gtsam::PinholeBaseK<class gtsam::Cal3Bundler>::project(class Eigen::Matrix<double,3,1,0,3,1> const &,class gtsam::OptionalJacobian<2,6>,class gtsam::OptionalJacobian<2,3>,class gtsam::OptionalJacobian<2,3>)const " (__imp_?project@?$PinholeBaseK@VCal3Bundler@gtsam@@@gtsam@@QEBA?AV?$Matrix@N$01$00$0A@$01$00@Eigen@@AEBV?$Matrix@N$02$00$0A@$02$00@4@V?$OptionalJacobian@$01$05@2@V?$OptionalJacobian@$01$02@2@2@Z) referenced in function main	timePinholeCamera	C:\Users\<My Windows Username>\Desktop\GTSAM\gtsam-4.1.1\build\timing\timePinholeCamera.obj	1	
```

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1120	1 unresolved externals	timePinholeCamera	C:\Users\<My Windows Username>\Desktop\GTSAM\gtsam-4.1.1\build\bin\Release\timePinholeCamera.exe	1	
```

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: class Eigen::Matrix<double,2,1,0,2,1> __cdecl gtsam::PinholeCamera<class gtsam::Cal3Bundler>::project2(class Eigen::Matrix<double,3,1,0,3,1> const &,class gtsam::OptionalJacobian<2,9>,class gtsam::OptionalJacobian<2,3>)const " (__imp_?project2@?$PinholeCamera@VCal3Bundler@gtsam@@@gtsam@@QEBA?AV?$Matrix@N$01$00$0A@$01$00@Eigen@@AEBV?$Matrix@N$02$00$0A@$02$00@4@V?$OptionalJacobian@$01$08@2@V?$OptionalJacobian@$01$02@2@@Z) referenced in function "public: virtual class Eigen::Matrix<double,-1,1,0,-1,1> __cdecl gtsam::GeneralSFMFactor<class gtsam::PinholeCamera<class gtsam::Cal3Bundler>,class Eigen::Matrix<double,3,1,0,3,1> >::evaluateError(class gtsam::PinholeCamera<class gtsam::Cal3Bundler> const &,class Eigen::Matrix<double,3,1,0,3,1> const &,class boost::optional<class Eigen::Matrix<double,-1,-1,0,-1,-1> &>,class boost::optional<class Eigen::Matrix<double,-1,-1,0,-1,-1> &>)const " (?evaluateError@?$GeneralSFMFactor@V?$PinholeCamera@VCal3Bundler@gtsam@@@gtsam@@V?$Matrix@N$02$00$0A@$02$00@Eigen@@@gtsam@@UEBA?AV?$Matrix@N$0?0$00$0A@$0?0$00@Eigen@@AEBV?$PinholeCamera@VCal3Bundler@gtsam@@@2@AEBV?$Matrix@N$02$00$0A@$02$00@4@V?$optional@AEAV?$Matrix@N$0?0$0?0$0A@$0?0$0?0@Eigen@@@boost@@2@Z)	timeSFMBAL	C:\Users\<My Windows Username>\Desktop\GTSAM\gtsam-4.1.1\build\timing\timeSFMBAL.obj	1	
```

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1120	1 unresolved externals	timeSFMBAL	C:\Users\<My Windows Username>\Desktop\GTSAM\gtsam-4.1.1\build\bin\Release\timeSFMBAL.exe	1	
```

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol "__declspec(dllimport) class std::vector<class gtsam::BinaryMeasurement<class gtsam::Rot3>,class std::allocator<class gtsam::BinaryMeasurement<class gtsam::Rot3> > > __cdecl gtsam::parseMeasurements<class gtsam::Rot3>(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class boost::shared_ptr<class gtsam::noiseModel::Diagonal> const &,unsigned __int64)" (__imp_??$parseMeasurements@VRot3@gtsam@@@gtsam@@YA?AV?$vector@V?$BinaryMeasurement@VRot3@gtsam@@@gtsam@@V?$allocator@V?$BinaryMeasurement@VRot3@gtsam@@@gtsam@@@std@@@std@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@2@AEBV?$shared_ptr@VDiagonal@noiseModel@gtsam@@@boost@@_K@Z) referenced in function main	timeShonanFactor	C:\Users\<My Windows Username>\Desktop\GTSAM\gtsam-4.1.1\build\timing\timeShonanFactor.obj	1	
```

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol "__declspec(dllimport) class std::map<unsigned __int64,class gtsam::Pose3,struct std::less<unsigned __int64>,class std::allocator<struct std::pair<unsigned __int64 const ,class gtsam::Pose3> > > __cdecl gtsam::parseVariables<class gtsam::Pose3>(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,unsigned __int64)" (__imp_??$parseVariables@VPose3@gtsam@@@gtsam@@YA?AV?$map@_KVPose3@gtsam@@U?$less@_K@std@@V?$allocator@U?$pair@$$CB_KVPose3@gtsam@@@std@@@4@@std@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@2@_K@Z) referenced in function main	timeShonanFactor	C:\Users\<My Windows Username>\Desktop\GTSAM\gtsam-4.1.1\build\timing\timeShonanFactor.obj	1	
```

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1120	2 unresolved externals	timeShonanFactor	C:\Users\<My Windows Username>\Desktop\GTSAM\gtsam-4.1.1\build\bin\Release\timeShonanFactor.exe	1	
```

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::PinholeCamera<class gtsam::Cal3Bundler>::PinholeCamera<class gtsam::Cal3Bundler>(void)" (__imp_??0?$PinholeCamera@VCal3Bundler@gtsam@@@gtsam@@QEAA@XZ) referenced in function main	timeAdaptAutoDiff	C:\Users\<My Windows Username>\Desktop\GTSAM\gtsam-4.1.1\build\timing\timeAdaptAutoDiff.obj	1	
```

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol "__declspec(dllimport) public: class Eigen::Matrix<double,2,1,0,2,1> __cdecl gtsam::PinholeCamera<class gtsam::Cal3Bundler>::project2(class Eigen::Matrix<double,3,1,0,3,1> const &,class gtsam::OptionalJacobian<2,9>,class gtsam::OptionalJacobian<2,3>)const " (__imp_?project2@?$PinholeCamera@VCal3Bundler@gtsam@@@gtsam@@QEBA?AV?$Matrix@N$01$00$0A@$01$00@Eigen@@AEBV?$Matrix@N$02$00$0A@$02$00@4 [... cut ...]
Running on Windows 10 with an AMD Ryzen Threadripper 3990X and 256 GB of RAM (I have also tried these same steps on a much less powerful home computer also running Windows 10). Using the GUI version of cmake and Visual Studio 2022 (used Visual Studio 2019 on the home computer).

GTSAM Version: 4.1.1 (downloaded from the releases section on GitHub)

Boost Version: 1.78.0 (I have tried both Boost from source and also the [prebuilt windows binaries](https://sourceforge.net/projects/boost/files/boost-binaries/) link provided on the Boost website)

## Additional information

N/A

ProfFan, 2022-02-04:
Out of curiosity, could you try building GTSAM with the VS Clang instead of MSVC? MSVC is very foreign to the entire team, and I've heard a lot of good comments from friends using Clang.

magicbycalvin, 2022-02-04:
Looked back at it and I believe I ran into the same problems as before despite using Clang instead of MSVC. Is there anything else I should try out? Perhaps I am not doing something right within VS?

dellaert, 2022-02-05:
Hi, I think what we test in CI and currently is MSVC. @ProfFan, could you possibly help with debugging that? That would then help everyone if there actually is an issue.

magicbycalvin, 2022-02-05:
I'd be happy to hop on Zoom to take a look if that would make things easier. Otherwise, I am more than happy to provide as much info here as you need. Thanks!

magicbycalvin, 2022-02-11:
Just wanted to bump this issue to see if anything has been discovered. If not, no worries! I understand how busy everyone must be!

dellaert, 2022-02-12:
Yeah, not only busy, but the core issue is most of us are on Linux or Mac. I tried compiling on an old windows laptop but could not get past boost library issues. It's odd, as our CI passed for 4.1.1., right? Can you look at the CI workflow and see what might be different on your system?

If nothing jumps out, I have this: a lot of the warnings in your original post concern timing scripts. How about trying a simpler target, e.g. `testGaussianFactorGraph`, which should just compile the basics to get that running. What are the unresolved symbols in that case?

And, finally, did you try latest develop? Sometimes a missing GTSAM_EXPORT is what is needed, so if develop works and we see what GTSAM_EXPORTs have been added since 4.1.1, we can do a patch 4.1.2 to make it compile on windows...

dellaert, 2022-02-13:
Wow, thanks @mikesheffler ! Two things:
1. Seems like one way forward is immediate: turn unstable OFF :-)
2. To get unstable to link (because seems compile is not an issue?) maybe check how many build errors you can get rid of by adding GTSAM_UNSTABLE_EXPORT to the *classes* that are mentioned above.

*However* I just looked at the first file, `SingleValue.h,` and *it is already there* ! This makes me think that something is wrong with handling exports in unstable. I could find the following commit from @varunagrawal that tinkered with this, so (a) you might be able to spot a problem, (b) @varunagrawal might be able to weight in: https://github.com/borglab/gtsam/commit/48cc70c5068e4fef0ed6a489a2739001e01a7534

It's telling that the windows CI does *not* build any unstable targets. Again, latest edits by @varunagrawal in the workflow, so he might be able to tell us more.

varunagrawal, 2022-02-13:
Yeah I ran into this issue a ton when trying to speedup the windows CI. The unstable build was never turned on which is why this issue wasn't caught before.

I tried adding GTSAM_UNSTABLE_EXPORT to the complaining classes and they didn't seem to do anything? At that point I sort of gave up because I had more pressing things to attend to and the Windows dev skill needed was above my pay-grade.

We may have to ask the folks at Robostack for help at this since they have the needed expertise.

mikesheffler, 2022-02-14:
> It's telling that the windows CI does not build any unstable targets. Again, latest edits by @varunagrawal in the workflow, so he might be able to tell us more.

> The unstable build was never turned on which is why this issue wasn't caught before

I don't actually know how to read the CI configurations, so I'm going to take your words for it, but looking at [this action](https://github.com/borglab/gtsam/runs/5168480167?check_suite_focus=true#step:5:202), I thought that that meant unstable _was_ being built. There are a ton of references in there that seem to suggest that it is.

-----

I created a pull request (#1097) that fixes the issues in `gtsam_unstable\Discrete`.

-----

However, I don't know how to fix the errors that are similar to

`LocalOrientedPlane3Factor.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor3<class gtsam::Pose3,class gtsam::Pose3,class gtsam::OrientedPlane3>::NoiseModelFactor3<class gtsam::Pose3,class gtsam::Pose3,class gtsam::OrientedPlane3>(void)" (__imp_??0?$NoiseModelFactor3@VPose3@gtsam@@V12@VOrientedPlane3@2@@gtsam@@QEAA@XZ) referenced in function "public: __cdecl gtsam::LocalOrientedPlane3Factor::LocalOrientedPlane3Factor(void)" (??0LocalOrientedPlane3Factor@gtsam@@QEAA@XZ)`

One part of this that I don't know how to untangle is that the definition for `LocalOrientedPlane3Factor` goes like this:

```c++
class GTSAM_UNSTABLE_EXPORT LocalOrientedPlane3Factor
    : public NoiseModelFactor3<Pose3, Pose3, OrientedPlane3>
```

and `NoiseModelFactor3` has 

```c++
template<class VALUE1, class VALUE2, class VALUE3>
class GTSAM_EXPORT NoiseModelFactor3: public NoiseModelFactor
```

are we seeing shenanigans relating to a `GTSAM_UNSTABLE_EXPORT` inheriting from a `GTSAM_EXPORT`? I don't know if that is / should be a problem.

-----

I'm willing to look at the MATLAB build (w/o `gtsam_unstable`) as well, but I have not tried to yet.

dellaert, 2022-02-14:
Yeah, that is weird. I think abstract classes like NoiseModelFactor3 should *not* have EXPORT, actually. Could you see whether removing that works?
PS, if that works and you're doing another PR, maybe turn off warning C4834, as it clutters everything up in logs :-/

mikesheffler, 2022-02-14:
>  I think abstract classes like NoiseModelFactor3 should not have EXPORT, actually. Could you see whether removing that works?

Removing `GTSAM_EXPORT` from `NoiseModelFactor3` and `NoiseModelFactor2` fixes all of the link errors associated with `LocalOrientedPlane3` and `RelativeElevationFactor`. I didn't PR it because I wasn't sure if it was implied that I would also strip `GTSAM_EXPORT` from `NoiseModelFactor1`, `4`, `5`, and `6` (probably), and (less obvious -- it's not completely abstract) `NoiseModelFactor`.

After making those local changes, I tried tackling the link errors associated with `ProjectionFactorRollingShutter` and didn't get far. The proximate error seems to be that `ProjectionFactorRollingShutter::evaluateError` has a `PinholeCamera<Cal3_S2> camera` and the linker can't figure out `camera.project(`. I think the `project` that gets called comes from `PinholeBaseK`, and that definition is not making back to the linker for reasons that aren't clear to me (but are probably `GTSAM_EXPORT`-related).

Throwing a dart at the rest of the list, I looked at `SmartStereoProjectionPoseFactor`. It had (among other things)

```
unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::SmartFactorBase<class gtsam::StereoCamera>::SmartFactorBase<class gtsam::StereoCamera>(class gtsam::SmartFactorBase<class gtsam::StereoCamera> const &)"
```

and I noticed the same structure mentioned before where `SmartStereoProjectionPoseFactor` has `GTSAM_UNSTABLE_EXPORT` and it derives from `SmartStereoProjectionFactor`, which is also `GTSAM_UNSTABLE_EXPORT`, which derives from `SmartFactorBase<StereoCamera>`, with `SmartFactorBase` carrying `GTSAM_EXPORT`.

I still don't know what happens when we have a `GTSAM_UNSTABLE_EXPORT` class derive from a `GTSAM_EXPORT` class (or maybe even just the derived class relying on non-overridden methods from the parent, grandparent, etc. class), so I didn't chase it any further.

> ... if that works and you're doing another PR, maybe turn off warning C4834 ...

Sure! Does that mean adding `#pragma warning(suppress : 4834)` somewhere (many places?), or is there a way for me to ask the CI to suppress it in the logs? I'm not sure what you have in mind.

dellaert, 2022-02-14:
Yeah, I don't know either. I tried re-reading https://github.com/borglab/gtsam/blob/develop/Using-GTSAM-EXPORT.md and it seems to say that as long as a class has *at least one* function defined in a .cpp, it should be OK to do GTSAM_EXPORT.

FWIW I think mixing GTSAM_EXPORT and GTSAM_UNSTABLE_EXPORT is not an issue, and they probably are `#defined` to the same thing underneath. Hence, the naming is unintuitive and *if* what I think is actually the case, we should probably just rename it to GTSAM_EXPORT.

About #pragma: most warnings I see seem related to BOOST_CONCEPT_CHECK, which is not used in many places, so maybe that's the way to go, yes.

Finally: you can add some unstable checks in the CI script (at the very end) to make sure things actually work

mikesheffler, 2022-02-15:
**tl;dr** I think I got it. The key seemed to be _removing_ some export statements. I can put together a PR later tonight, once I try to quiet some of the warnings, and learn how to run the tests that @dellaert asked for.

-----

> ... it seems to say that as long as a class has at least one function defined in a .cpp, it should be OK to do GTSAM_EXPORT

Aaaaaaaaaah. I was not aware of that. It makes sense, but I didn't think of it at all. Would you like to guess what the genealogies of all of the complaining classes have in common? There was a `GTSAM_EXPORT` or `GTSAM_UNSTABLE_EXPORT` statement in a parent class that (the parent class) was completely defined in a header file (no corresponding .cpp file). I did a little tidying up, and now I can get `gtsam` and `gtsam_unstable` to both build. It's super minor changes to like five files.

-----

> FWIW I think mixing GTSAM_EXPORT and GTSAM_UNSTABLE_EXPORT is not an issue, and they probably are #defined to the same thing underneath. Hence, the naming is unintuitive and if what I think is actually the case, we should probably just rename it to GTSAM_EXPORT.

That sounds plausible to me, but that wouldn't (I don't think) explain why I had to change
 ```c++
 class GTSAM_EXPORT Constraint : public DiscreteFactor
 ```
 to
 ```c++
 class GTSAM_UNSTABLE_EXPORT Constraint : public DiscreteFactor
 ```
 in my previous PR.
 
-----

> About #pragma: most warnings I see seem related to BOOST_CONCEPT_CHECK, which is not used in many places, so maybe that's the way to go, yes.

Okay, I will try to add some #pragma statements to a few files and try to get the compiler to settle down a little with the warnings.

-----

> Finally: you can add some unstable checks in the CI script (at the very end) to make sure things actually work

Sorry for asking about such a basic issue, but I don't actually know how to do this. Looking at [the Windows build Actions](https://github.com/borglab/gtsam/actions/workflows/build-windows.yml), I see what _was_ run for recent commits (I think). I don't know what to modify to run the tests you are asking for. Could you please explain (or, if there's something that I can read, that works too).

dellaert, 2022-02-15:
Woah, good sleuthing :-)

about the last point, I’m not an expert either on these workflows :-) however, at the end of this file: https://github.com/borglab/gtsam/blob/develop/.github/workflows/build-windows.yml

you can add any target that is defined in GTSam. You can do `make help`  to see all the targets. The most hard-core target is simply `check`, which would certify that every unit test works in both gtsam and unstable….

mikesheffler, 2022-02-15:
There are some more details in #1102 , but I had to cave on silencing the warnings and adding checks.

* Using #pragma statements in .cpp files wound up touching like 85 files (not exaggerating), so maybe there's a CMake way to do it? I have a patch saved that just does all the pragmas, and I could apply it an make a PR, if that's the route you want to go.

* The `check`s are in bad shape for MSVC / Windows. The six that are in the Windows CI actions right now are (basically) the only six that pass. Some checks seem to be agglomerations of checks, so that probably doesn't count. It's worth mentioning that `gtsam`, `gtsam_unstable`, and `wrap` are included (and presently pass) in the Windows CI.

Here's the list from my notes:

Pass
-----
check.base
check.base_unstable
check.linear
check_base_program
check_base_unstable_program
check_dynamics_unstable_program
check_geometry_unstable_program
check_inference_program
check_linear_program
check_symbolic_program

Fail
-----
check.basis
check.discrete
check.discrete_unstable
check.dynamics_unstable
check.geometry
check.geometry_unstable
check.inference
check.linear_unstable
check.navigation
check.nonlinear
check.nonlinear_unstable
check.partition
check.sam
check.sfm
check.slam
check.slam_unstable
check.symbolic
check.tests
check_basis_program
check_discrete program
check_discrete unstable_program
check_geometry_program
check_linear_unstable_program
check_navigation_program
check_nonlinear_unstable_program
check_partition_program
check_sam_program
check_sfm_program
check_slam_program
check_slam_unstable_program
check_tests_program

I have not looked into why any of these are failing. Many of them might be straightforward to fix, but I don't know, at present.

mikesheffler, 2022-02-15:
Getting back to the original question of getting the MATLAB toolbox building, here is the current build output, for posterity. This is just the 'regular' toolbox, and _not_ attempting to add `GTSAM_UNSTABLE_INSTALL_TOOLBOX`.

The formatting here is really frustrating, and many of the lines are 'Hint on symbols that are defined and could potentially match` 🙄 , but I don't really know a better way to present it.

Build Errors for `gtsam_matlab_wrapper`
-----
```
Build started...
1>------ Build started: Project: gtsam_matlab_wrapper, Configuration: Debug x64 ------
1>   Creating library C:/Mike/Work/Build/gtsam-msheffler/wrap/gtsam_mex/Debug/gtsam_wrapper.lib and object C:/Mike/Work/Build/gtsam-msheffler/wrap/gtsam_mex/Debug/gtsam_wrapper.exp
1>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "public: class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl gtsam::RedirectCout::str(void)const " (?str@RedirectCout@gtsam@@QEBA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ) referenced in function "void __cdecl gtsamRedirectCout_str_106(int,struct mxArray_tag * * const,int,struct mxArray_tag const * * const)" (?gtsamRedirectCout_str_106@@YAXHQEAPEAUmxArray_tag@@HQEAPEBU1@@Z)
1>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "public: __cdecl gtsam::RedirectCout::~RedirectCout(void)" (??1RedirectCout@gtsam@@QEAA@XZ) referenced in function "public: void * __cdecl gtsam::RedirectCout::`scalar deleting destructor'(unsigned int)" (??_GRedirectCout@gtsam@@QEAAPEAXI@Z)
1>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor1<class Eigen::Matrix<double,-1,1,0,-1,1> >::NoiseModelFactor1<class Eigen::Matrix<double,-1,1,0,-1,1> >(void)" (__imp_??0?$NoiseModelFactor1@V?$Matrix@N$0?0$00$0A@$0?0$00@Eigen@@@gtsam@@QEAA@XZ) referenced in function "public: __cdecl gtsam::PriorFactor<class Eigen::Matrix<double,-1,1,0,-1,1> >::PriorFactor<class Eigen::Matrix<double,-1,1,0,-1,1> >(void)" (??0?$PriorFactor@V?$Matrix@N$0?0$00$0A@$0?0$00@Eigen@@@gtsam@@QEAA@XZ)
1>  Hint on symbols that are defined and could potentially match:
1>    "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor1<class gtsam::Pose3>::NoiseModelFactor1<class gtsam::Pose3>(class gtsam::NoiseModelFactor1<class gtsam::Pose3> const &)" (__imp_??0?$NoiseModelFactor1@VPose3@gtsam@@@gtsam@@QEAA@AEBV01@@Z)
1>    "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor1<class gtsam::Pose3>::NoiseModelFactor1<class gtsam::Pose3>(class boost::shared_ptr<class gtsam::noiseModel::Base> const &,unsigned __int64)" (__imp_??0?$NoiseModelFactor1@VPose3@gtsam@@@gtsam@@QEAA@AEBV?$shared_ptr@VBase@noiseModel@gtsam@@@boost@@_K@Z)
1>    "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor1<class gtsam::Pose3>::NoiseModelFactor1<class gtsam::Pose3>(void)" (__imp_??0?$NoiseModelFactor1@VPose3@gtsam@@@gtsam@@QEAA@XZ)
1>    "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor1<class gtsam::Rot3>::NoiseModelFactor1<class gtsam::Rot3>(class gtsam::NoiseModelFactor1<class gtsam::Rot3> const &)" (__imp_??0?$NoiseModelFactor1@VRot3@gtsam@@@gtsam@@QEAA@AEBV01@@Z)
1>    "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor1<class gtsam::Rot3>::NoiseModelFactor1<class gtsam::Rot3>(class boost::shared_ptr<class gtsam::noiseModel::Base> const &,unsigned __int64)" (__imp_??0?$NoiseModelFactor1@VRot3@gtsam@@@gtsam@@QEAA@AEBV?$shared_ptr@VBase@noiseModel@gtsam@@@boost@@_K@Z)
1>    "__declspec(dllimport) public: __cdecl gtsam::NoiseModelFactor1<class gtsam::Rot3>::NoiseModelFactor1<class gtsam::Rot3>(void)" (__imp_??0?$NoiseModelFactor1@VRot3@gtsam@@@gtsam@@QEAA@XZ)
1>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: virtual __cdecl gtsam::NoiseModelFactor1<class Eigen::Matrix<double,-1,1,0,-1,1> >::~NoiseModelFactor1<class Eigen::Matrix<double,-1,1,0,-1,1> >(void)" (__imp_??1?$NoiseModelFactor1@V?$Matrix@N$0?0$00$0A@$0?0$00@Eigen@@@gtsam@@UEAA@XZ) referenced in function "int `public: __cdecl gtsam::PriorFactor<class Eigen::Matrix<double,-1,1,0,-1,1> >::PriorFactor<class Eigen::Matrix<double,-1,1,0,-1,1> >(class PriorFactor<class Eigen::Matrix<double,-1,1,0,-1,1> >::dtor$0 const &)'::`1'::dtor$0" (?dtor$0@?0???0?$PriorFactor@V?$Matrix@N$0?0$00$0A@$0?0$00@Eigen@@@gtsam@@QEAA@AEBV01@@Z@4HA)
1>  Hint on symbols that are defined and could potentially match:
1>    "__declspec(dllimport) public: virtual __cdecl gtsam::NoiseModelFactor1<class gtsam::Pose3>::~NoiseModelFactor1<class gtsam::Pose3>(void)" (__imp_??1?$NoiseModelFactor1@VPose3@gtsam@@@gtsam@@UEAA@XZ)
1>    "__declspec(dllimport) public: virtual __cdecl gtsam::NoiseModelFactor1<class gtsam::Rot3>::~NoiseModelFactor1<class gtsam::Rot3>(void)" (__imp_??1?$NoiseModelFactor1@VRot3@gtsam@@@gtsam@@UEAA@XZ)
1>gtsam_wrapper.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: unsigned __int64 __cdecl gtsam::NoiseModelFactor1<class Eigen::Matrix<double,-1,1,0,-1,1> >::key(void)const " (__imp_?key@?$NoiseModelFactor1@V?$Matrix@N$0?0$00$0A@$0?0$00@Eigen@@@gtsam@@QEBA_KXZ) referenced in function "public: virtual void __cdecl gtsam::PriorFactor<class Eigen::Matrix<double,-1,1,0,-1,1> >::print(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::function<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char [... cut ...]
1>gtsam_wrapper.obj : error LNK2001: unresolved external symbol "public: virtual class Eigen::Matrix<double,-1,1,0,-1,1> __cdecl gtsam::NoiseModelFactor1<class gtsam::ParameterMatrix<3> >::unwhitenedError(class gtsam::Values const &,class boost::optional<class std::vector<class Eigen::Matrix<double,-1,-1,0,-1,-1>,class std::allocator<class Eigen::Matrix<double,-1,-1,0,-1,-1> > > &>)const " (?unwhitenedError@?$NoiseModelFactor1@V?$ParameterMatrix@$02@gtsam@@@gtsam@@UEBA?AV?$Matrix@N$0?0$00$0A@$0?0$00@Eigen@@AEBVValues@2@V?$optional@AEAV?$vector@V?$Matrix@N$0?0$0?0$0A@$0?0$0?0@Eigen@@V?$allocator@V?$Matrix@N$0?0$0?0$0A@$0?0$0?0@Eigen@@@std@@@std@@@boos

dellaert, 2022-02-17:
> In plain English, what's the goal here?

Trying to resolve https://github.com/borglab/gtsam/issues/91 and https://github.com/borglab/gtsam/pull/1102, ultimately getting MATLAB and python wrappers working on windows. Borrowing from your little bird analogy, I propose using make check as the canary in the coal mine: this would suss out all compile and linking errors in gtsam proper, make our windows CI better, and move on to any issues in building the MATLAB wrapper. Makes sense?

mattking-smith, 2022-02-23:
> That's weird. I know @mattking-smith uses the MATLAB toolbox, so he must have gotten that to compile. Matt?

@dellaert I am using the MATLAB toolbox in a _Ubuntu 20.04_ environment and have everything successfully compiling, wrapping, and running with no issues. 

So this may be a Windows compilation issue? Are these MATLAB toolbox issue really related to the toolbox itself or an OS compilation problem? 

Gauging from https://github.com/borglab/gtsam/issues/1087#issuecomment-1042953842 it looks like other non-MATLAB toolbox things are going on here.

oicchris, 2023-01-11:
Aloha!  I was attempting to build gtsam as well and ran into this issue.  I believe the issue lies with exporting templates and their specializations.  I'm currently using VS2022 on the v142 toolset.  Here is a tiny example of one of the linkage errors - only because it was the first link error that popped up while compiling.  I will be focusing on the timing/timeShonanFactor application.

```
timeShonanFactor.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) class std::vector<class gtsam::BinaryMeasurement<class gtsam::Rot3>,class std::allocator<class gtsam::BinaryMeasurement<class gtsam::Rot3> > > __cdecl gtsam::parseMeasurements<class gtsam::Rot3>(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class boost::shared_ptr<class gtsam::noiseModel::Diagonal> const &,unsigned __int64)" (__imp_??$parseMeasurements@VRot3@gtsam@@@gtsam@@YA?AV?$vector@V?$BinaryMeasurement@VRot3@gtsam@@@gtsam@@V?$allocator@V?$BinaryMeasurement@VRot3@gtsam@@@gtsam@@@std@@@std@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@2@AEBV?$shared_ptr@VDiagonal@noiseModel@gtsam@@@boost@@_K@Z) referenced in function main
timeShonanFactor.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) class std::map<unsigned __int64,class gtsam::Pose3,struct std::less<unsigned __int64>,class std::allocator<struct std::pair<unsigned __int64 const ,class gtsam::Pose3> > > __cdecl gtsam::parseVariables<class gtsam::Pose3>(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,unsigned __int64)" (__imp_??$parseVariables@VPose3@gtsam@@@gtsam@@YA?AV?$map@_KVPose3@gtsam@@U?$less@_K@std@@V?$allocator@U?$pair@$$CB_KVPose3@gtsam@@@std@@@4@@std@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@2@_K@Z) referenced in function main
```

To fix these two link errors, the template function specializations need to be advertised in gtsam/slam/dataset.cpp by adding GTSAM_EXPORT.
```
template <>
GTSAM_EXPORT std::vector<BinaryMeasurement<Rot3>>
parseMeasurements(const std::string &filename,
                  const noiseModel::Diagonal::shared_ptr &model,
                  size_t maxIndex)

template <>
GTSAM_EXPORT std::map<size_t, Pose3> parseVariables<Pose3>(const std::string &filename,
                                              size_t maxIndex)
```

Alternatively, these type of  unresolved symbol link errors go away if you disable BUILD_SHARED_LIBS in the cmake configuration step.

ProfFan, 2023-01-11:
@oicchris Thank you! Can you put up a small PR for that fix?

oicchris, 2023-01-11:
Oof, I apologize that I do not have time to fix all of the export symbols.  Embarrassingly, I went with the alternate static library route.

varunagrawal, 2023-12-30:
We've massively improved the Windows compilation and a lot of the linker issues have been resolved. I guess we can close this issue unless there is something else that needs to be addressed.

closed 2023-12-30 by varunagrawal