borglab/gtsam #1098: Error with python_wrapper: plot_covariance_ellipse_3d not work as expected

szx0112, 2022-02-14:
## Description

Hi gtsam support team, I tried to plot the marginal covariance matrix using `plot_covariance_ellipse_3d`. The inputs are  `axes`, `3x1 point position` , and `6x6 covariance matrix obtained from gtsam_marginals.marginalCovariance(keys[ii])`. I have checked and ensured these input variables are good.

After running the code, I got the error `shape mismatch: objects cannot be broadcast to a single shape`.
The problem is related to this line:
https://github.com/borglab/gtsam/blob/3e65779421d4218f5bb483c088fac12e1508f680/python/gtsam/utils/plot.py#L102

The `radii` is actually a 6x1 vector instead of 3x1 vector as expected. An example of the `radii` I got is:
` [12.57294157 12.42959364 11.95267523  0.18826816  0.18105469  0.17589153]`.

## Steps to reproduce

1. I modify the code [Pose3SLAMExample_g2o.py] (https://github.com/borglab/gtsam/blob/develop/python/gtsam/examples/Pose3SLAMExample_g2o.py) to read g2o format.
2. test data is downloaded from the [benchmark dataset](https://lucacarlone.mit.edu/datasets/)
3.  I use `marginals = gtsam.Marginals(graph, initial)` to extract `marginals` and iteratively obtain the covariance matrix (6x6 matrix) at each pose based on` initial.keys()[ii]`.
4. I have plotted the pose positions (both before and after optimization) on these dataset without any issue, and got the error (as above) when I try to show the covariance matrix at each pose.

## Expected behavior

The eclipses should be shown at each point

## Environment

Python 3.8 in conda environment. Build from source

Any comment is appreciated.

varunagrawal, 2022-07-14:
Hi @szx0112 does #1193 solve your issue?

varunagrawal, 2023-03-10:
Closing due to lack of a response. Please reopen if this is still an issue.

closed 2023-03-10 by varunagrawal